realsenseai/realsense-ros #2215: camera_imu_optical_frame is not connected with the camera_link

haquebd, 2022-01-08:
I am not receiving any data under "rostopic echo -n1 /tf_static" when I launched "opensource_tracking.launch" file. The "camera_imu_optical_frame" is attached with the "odom" frame and its not moving along with the camera_link (**which is required**).

Also getting error while launching
[ERROR] (2022-01-08 14:52:55.941) Rtabmap.cpp:1249::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 91869 is ignored!

`~/catkin_ws$ rostopic echo -n1 /tf_static
transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715043962
      frame_id: "camera_link"
    child_frame_id: "camera_depth_frame"
    transform: 
      translation: 
        x: 0.0
        y: -0.0
        z: -0.0
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715043962
      frame_id: "camera_depth_frame"
    child_frame_id: "camera_depth_optical_frame"
    transform: 
      translation: 
        x: 0.0
        y: -0.0
        z: -0.0
      rotation: 
        x: -0.5
        y: 0.5
        z: -0.5
        w: 0.5
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715311227
      frame_id: "camera_link"
    child_frame_id: "camera_color_frame"
    transform: 
      translation: 
        x: -0.000278902094578
        y: 0.0143958646804
        z: 0.000307566835545
      rotation: 
        x: 0.00233976123855
        y: -0.0015400829725
        z: -0.000215966429096
        w: 0.999996066093
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715311227
      frame_id: "camera_aligned_depth_to_color_frame"
    child_frame_id: "camera_color_optical_frame"
    transform: 
      translation: 
        x: 0.0
        y: -0.0
        z: -0.0
      rotation: 
        x: -0.5
        y: 0.5
        z: -0.5
        w: 0.5
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715311227
      frame_id: "camera_link"
    child_frame_id: "camera_aligned_depth_to_color_frame"
    transform: 
      translation: 
        x: -0.000278902094578
        y: 0.0143958646804
        z: 0.000307566835545
      rotation: 
        x: 0.00233976123855
        y: -0.0015400829725
        z: -0.000215966429096
        w: 0.999996066093
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715741821
      frame_id: "camera_link"
    child_frame_id: "camera_gyro_frame"
    transform: 
      translation: 
        x: -0.0117399999872
        y: -0.00552000012249
        z: 0.00510000018403
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715741821
      frame_id: "camera_gyro_frame"
    child_frame_id: "camera_imu_optical_frame"
    transform: 
      translation: 
        x: 0.0
        y: -0.0
        z: -0.0
      rotation: 
        x: -0.5
        y: 0.5
        z: -0.5
        w: 0.5
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715882461
      frame_id: "camera_link"
    child_frame_id: "camera_accel_frame"
    transform: 
      translation: 
        x: -0.0117399999872
        y: -0.00552000012249
        z: 0.00510000018403
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1641672194
        nsecs: 715882461
      frame_id: "camera_accel_frame"
    child_frame_id: "camera_accel_optical_frame"
    transform: 
      translation: 
        x: 0.0
        y: -0.0
        z: -0.0
      rotation: 
        x: -0.5
        y: 0.5
        z: -0.5
        w: 0.5
---
`

[frames.pdf](https://github.com/IntelRealSense/realsense-ros/files/7833946/frames.pdf)

MartyG-RealSense, 2022-01-09:
Hi @haquebd  In your ROS launch log, do you have any IMU-related warnings such as the ones described in https://github.com/IntelRealSense/realsense-ros/issues/1224#issue-629530480

haquebd, 2022-01-11:
I am getting the following Warning while running the rs_camera.launch file using initial_reset:=true. I am getting data at rostopic echo /camera/imu but it is still connected to the Odom along with the "camera_imu_optical_frame" frame which we require with the camera_link 

 10/01 22:50:47,684 WARNING [547406999936] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
**[ WARN] [1641873047.693797523]:** 
[document.pdf](https://github.com/IntelRealSense/realsense-ros/files/7843595/document.pdf)

 10/01 22:50:47,693 WARNING [547247604096] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.5/0003:8086:0B3A.0004/HID-SENSOR-200073.1.auto/iio:device1/buffer/enable
[ INFO] [1641873047.755575616]: SELECTED BASE:Depth, 0
[ WARN] [1641873047.851886384]: Publishing dynamic camera transforms (/tf) at 10 Hz
[ INFO] [1641873047.871539195]: RealSense Node Is Up!
**[ WARN] [1641873048.306809757]:** Hardware Notification:Motion Module failure,1.64187e+12,Error,Hardware Error

MartyG-RealSense, 2022-01-11:
There is little documentation available about the **Failed to meet update rate!** error.  When ROS does generate this error though, it sometimes provides this advice: "Try decreasing the rate, limiting sensor output frequency, or limiting the number of sensors".  The IMU configuration settings that are being applied by opensource_tracking.launch do not seem to be very demanding though.

When using rs_camera.launch you should add terms to the roslaunch instruction to enable the gyro and accel IMU topics, as they are disabled by default in that launch file.  For example:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation**

haquebd, 2022-01-12:
Thanks @MartyG-RealSense for your response. Would you let me know which example script is available for a realsense D435i for SLAM as I am trying to implement the camera with a RTab SLAM?

Can you suggest any ideal IMU settings for Opensource_tracking.launch file or shall I follow the rs_camera setting as a base and add 3 more parameters as you told?

MartyG-RealSense, 2022-01-12:
Intel has a D435i SLAM tutorial that makes use of opensource_tracking.launch and rtabmap_ros.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

A RealSense user created an adaptation of this guide that adds localization and tracking of the camera's position.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

If you experience performance problems with the IMU when using opensource_tracking.launch then you may find that changing the **unite_imu_method** setting from 'linear_interpolation' to 'copy' improves stability.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/opensource_tracking.launch#L7

MartyG-RealSense, 2022-01-23:
Case closed due to no further comments received.

closed 2022-01-23 by MartyG-RealSense

cgroves3, 2023-10-21:
I'm experiencing this same issue. Here is my launch script. Please bear with me as I'm a little new to ros2
```
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from launch.substitutions import PathJoinSubstitution

def generate_launch_description():
    # realsense
    rs_camera_launch = LaunchDescription([
        Node(
            package="realsense2_camera",
            executable="realsense2_camera_node",
            name="realsense_camera",
            namespace="camera",
            parameters=[{
                "align_depth.enable": True,
                "linear_accel_cov": 1.0,
                "unite_imu_method": 1,  # linear_interpolation
                "enable_rgbd": True,
                "enable_sync": True,
                "enable_color": True,
                "enable_depth": True,
                "initial_reset": True
            }]
        )
    ])

    rtabmap_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            PathJoinSubstitution([
                FindPackageShare("rtabmap_launch"),
                "launch",
                "rtabmap.launch.py"
            ])
        ),
        launch_arguments={
            "args": "--delete_db_on_start",
            "rgb_topic": "/camera/color/image_raw",
            "depth_topic": "/camera/aligned_depth_to_color/image_raw",
            "camera_info_topic": "/camera/color/camera_info",
            "depth_camera_info_topic": "/camera/depth/camera_info",
            "rtabmapviz": "False",
            "rviz": "False",
            "wait_imu_to_init": "True"
        }.items()
    )

    # imu_filter_madgwick
    imu_launch = LaunchDescription(
        [
            Node(
                package='imu_filter_madgwick',
                executable='imu_filter_madgwick_node',
                name='imu_filter',
                parameters=[{
                    "use_mag": False,
                    "publish_tf": True,
                    "world_frame": "enu"
                }],
                remappings=[
                    ('/imu/data_raw', '/camera/imu'),
                ]
            )
        ]
    )

    # robot_localization
[... 52 lines cut ...]
[realsense2_camera_node-4] [INFO] [1697910508.551350146] [camera.realsense_camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-4] [INFO] [1697910508.567269853] [camera.realsense_camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-4] [INFO] [1697910508.653729059] [camera.realsense_camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.653857093] [camera.realsense_camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.653883589] [camera.realsense_camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.654059240] [camera.realsense_camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-4] [INFO] [1697910508.662696770] [camera.realsense_camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-4] [INFO] [1697910508.676393751] [camera.realsense_camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-4] [INFO] [1697910508.676623995] [camera.realsense_camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-4] [INFO] [1697910508.684884718] [camera.realsense_camera]: Starting Sensor: Motion Module
[realsense2_camera_node-4] [INFO] [1697910508.691061468] [camera.realsense_camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[realsense2_camera_node-4] [INFO] [1697910508.691175166] [camera.realsense_camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-4] [INFO] [1697910508.694376247] [camera.realsense_camera]: RealSense Node Is Up!
[realsense2_camera_node-4] [WARN] [1697910508.709285409] [camera.realsense_camera]: 
[imu_filter_madgwick_node-1] [INFO] [1697910508.869217380] [imu_filter]: First IMU message received.
[rgbd_odometry-2] [ERROR] [1697910509.072118500] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910508.850821
[rgbd_odometry-2] [ERROR] [1697910509.274238295] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910508.856719
[rgbd_odometry-2] [ERROR] [1697910509.477363644] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910508.991390
[rgbd_odometry-2] [ERROR] [1697910509.680161339] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.137486
[rgbd_odometry-2] [ERROR] [1697910509.883610982] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.253501
[... cut ...]
[rgbd_odometry-2] [ERROR] [1697910510.494141992] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910509.929486
[rgbd_odometry-2] [ERROR] [1697910510.697950425] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910510.115256
[rgbd_odometry-2] [ERROR] [1697910510.900790041] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "camera_imu_optical_frame" to frame "base_link", TF not available at time 1697910510.337561