realsenseai/realsense-ros #2216: I run the roslaunch realsense2_camera rs_camera.launch and this error pop up...

LeoWang19, 2022-01-09:
[ERROR] [1641735761.239121470]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /home/wx/livox_ws/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.48: cannot open shared object file: No such file or directory)
[ERROR] [1641735761.239194405]: The error before refreshing the cache was: Failed to load library /home/wx/livox_ws/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.48: cannot open shared object file: No such file or directory)
[FATAL] [1641735761.239454208]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 21570, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/wx/.ros/log/038aa71e-7152-11ec-b7ee-9cb6d0610e2f/camera-realsense2_camera-3.log].
log file: /home/wx/.ros/log/038aa71e-7152-11ec-b7ee-9cb6d0610e2f/camera-realsense2_camera-3*.log

LeoWang19, 2022-01-09:
I am currently use Ubuntu18.04 and ros-melodic, realsense-sdk 2.42.2 and realsense-ros 2.2.22

MartyG-RealSense, 2022-01-09:
Hi @LeoWang19  There was a previous case of this set of errors in https://github.com/IntelRealSense/librealsense/issues/6451 where the errors disappeared after updating the librealsense SDK and RealSense ROS wrapper version.

Which method did you use to install the librealsense SDK and ROS wrapper, please?  Did you build the SDK and librealsense separately, or install them together from packages with the wrapper's **Method 1** installation process, please?

If librealsense and the ROS wrapper are built separately then librealsense must be installed first and the wrapper secondly afterwards.  

If librealsense and the wrappr are built separately, whenever librealsense is updated then the wrapper must be built again afterwards.

LeoWang19, 2022-01-11:
Dear MartyG,

Sorry for being late. I just reviewed all the detailed processes.

1. I installed SDK and ROS wrapper separately.
2. Librealsense is installed first and then wrapper.
3. I do not think I updated Librealsense as it is currently showing version 2.42.0 in the package.xml file.

This is actually a weird issue because my ROS wrapper has been working well since April 2021, and it suddenly failed recently.
I tried to recompile the Realsense-ros package, but the compiling keep failing with the log as follows.

Consolidate compiler generated dependencies of target realsense2_camera
[ 91%] Built target realsense2_camera_generate_messages
[ 93%] Building CXX object realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o
In file included from /opt/ros/melodic/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /opt/ros/melodic/include/librealsense2/hpp/rs_device.hpp:8,
                 from /opt/ros/melodic/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /opt/ros/melodic/include/librealsense2/hpp/rs_context.hpp:8,
                 from /opt/ros/melodic/include/librealsense2/rs.hpp:9,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:12,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/opt/ros/melodic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/melodic/include/librealsense2/hpp/rs_frame.hpp:1047:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/melodic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/melodic/include/librealsense2/hpp/rs_frame.hpp:1069:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/melodic/include/librealsense2/hpp/rs_frame.hpp: In lambda function:
/opt/ros/melodic/include/librealsense2/hpp/rs_frame.hpp:1091:58: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         frm.get_profile().stream_index() == index) f = frm;
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::publishPointCloud(rs2::points, const ros::Time&, const rs2::frameset&)’:
/home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:2174:29: error: ‘find_if’ was not declared in this scope
         texture_frame_itr = find_if(frameset.begin(), frameset.end(), [&texture_source_id, &available_formats] (rs2::frame f)
                             ^~~~~~~
/home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:2174:29: note: suggested alternatives:
In file included from /usr/include/c++/7/algorithm:62:0,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:39,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/melodic/include/class_loader/class_loader.hpp:36,
                 from /opt/ros/melodic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/melodic/include/pluginlib/class_list_macros.h:35,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:6,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/c++/7/bits/stl_algo.h:3923:5: note:   ‘std::find_if’
     find_if(_InputIterator __first, _InputIterator __last,
     ^~~~~~~
In file included from /usr/include/boost/mpl/find.hpp:17:0,
                 from /usr/include/boost/mpl/aux_/contains_impl.hpp:20,
                 from /usr/include/boost/mpl/contains.hpp:20,
                 from /usr/include/boost/math/policies/policy.hpp:10,
                 from /usr/include/boost/math/policies/error_handling.hpp:21,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/melodic/include/ros/time.h:58,
                 from /opt/ros/melodic/include/ros/console.h:39,
                 from /opt/ros/melodic/include/nodelet/nodelet.h:39,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:7,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/wx/ws_livox/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/boost/mpl/find_if.hpp:32:8: note:   ‘boost::mpl::find_if’
 struct find_if
        ^~~~~~~
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:89: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:2511: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

LeoWang19, 2022-01-11:
The possible reason for the recent failure that I can think of is that I modified some settings in **software & updates**. 

This may affect some software sources? I am not sure how would this affect my working Librealsense and Ros wrapper tbh.

LeoWang19, 2022-01-11:
BTW, I can give a shot with the solution that you mentioned in the previous post, which is to update both Realsense SDK and Ros-wrapper. 

I am wondering do I need to do a clean uninstall of Librealsense with:
sudo make uninstall && make clean

Then recompile from source?

Then find and recompile the matched Realsense-ros.

Please let me know your thoughts, really appreciate ur time and efforts.

Cheers

MartyG-RealSense, 2022-01-11:
If you previously built librealsense from source code then yes, go to the **build** folder of librealsense that you created when originally installing the SDK and perform **sudo make uninstall && make clean** to uinstall librealsense and clean the CMake cache.

If librealsense was built from Debian packages instead of built from source code then you can uninstall all librealsense packages with the instruction below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Once librealsense is successfully installed then build the correct match of RealSense ROS wrapper from source code.

*************

You could alternatively build the librealsense SDK and ROS wrapper together at the same time from packages using the ROS wrapper's **Method 1** instructions.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

***************

I am not familiar with Ubuntu's **Software & Updates** interface as I am primarily a Windows user, so I do not have advice regarding how changes in that interface might affect librealsense or the ROS wrapper.  I have not heard it mentioned in the past in relation to RealSense problems.

MartyG-RealSense, 2022-01-17:
Hi @LeoWang19  Do you require further assistance with this case, please?  Thanks!

mm-nasr, 2022-01-20:
I am facing the same issue mentioned above. I am on librealsense 2.48 and I'm trying to build realsense-ros from source release [2.3.1](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.1).

When I tried deleting source code and installing the debian package, it works. However, when I run the cameras, the computer freezes and I have to ultimately power-cycle it.
Same behavior when I build the latest version of realsense-ros with librealsense 2.50

MartyG-RealSense, 2022-01-21:
Hi @mm-nasr  Thanks so much for the update about your solution and sharing it with the RealSense ROS community!

LeoWang19, 2022-01-27:
@MartyG-RealSense @mm-nasr Sorry for late, I will try this solution and close this issue for now. Really appreciate the help!

closed 2022-01-27 by LeoWang19

MartyG-RealSense, 2022-01-27:
You are very welcome, @LeoWang19 - thanks very much for the update!

LeoWang19, 2022-02-16:
just to add on. Worked on this issue yesterday. 

It works by a clear uninstallation of Librealsense, and reinstalling the latest Librealsense V2.50 + Realsense 2.3.2.

The new Librealsense was installed from source code.

MartyG-RealSense, 2022-02-16:
@LeoWang19   I'm very pleased to hear that a clean install resolved your problem.

@mm-nasr  Thanks again for your advice in this case!