realsenseai/realsense-ros #2217: Problems about VINS-FUSION configuration with Realsense D455 under NVIDIA NX

linchen2020, 2022-01-10:
Dear friends: 
    Hello everyone, I have a question to ask you.
     After I install the driver of realsense d455, open terminal and enter "realsense viewer", the phenomenon shown in the following figure will appear. Prompt in terminal: "10 / 01 11:06:29364 warning [548511153792] (DS5 factory. CPP: 1123) DS5 group_devices is empty.", On the graphical interface of the software, "11:27:18 [info] / home / NVIDIA / toollib / librealsense-2.40.0/src/linux/backend-hid.cpp: 1139 - failed to read busnum / devnum. Device path: / sys / bus / IIO / devices / IIO: device0" is constantly prompted。
How to solve this problem？
（Remarks:

System:Ubuntu 18.04.4 (LTS Linux nx 4.9.140-tegra)

Ros: melodic

Hardware platform uses NVIDIA Jetson NX with the specific configuration shown in Fig:

)
[image]

MartyG-RealSense, 2022-01-10:
Hi @linchen2020  The error **busnum/devnum** is a very rare one that has usually occurred in the past on single-board computing devices (like the original Up Board, the Intel Compute Stick and of course Jetson) when the USB port is supplying the camera with insufficient power.  I recall that using the camera in a USB-C micro-sized USB port made it more likely for the error to occur.

May I also ask the following questions please.

1.  Did you install the librealsense SDK first (for example, by using the instructions on Intel's Jetson instructions page) and then build the RealSense ROS wrapper secondly after librealsense installation was completed.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

2.  Are you using a mains electricity powered USB hub or is the camera plugged directly into the Jetson's USB port?  If it is plugged into the Jetson board, is it in one of the full-size USB 3.1 ports?

linchen2020, 2022-01-11:
***@***.***
    For your second question：I use the camera to plug directly into Jetson's USB port, which is usb3 0. It has been verified that images and IMU data can be read through realsense veiwer.
    At present, I reinstall Ubuntu 18 on NX， and I reinstalled 'librealsense SDK' follow the“ https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md ”. But After I put the camera is directly inserted into the USB port of Jetson, when the "realsense viewer" command  run on the terminal, a prompt "[warn] / sandbox / SRC / DS5 / DS5 factory. CPP: 1152 - DS5 group_devices is empty." appears on the graphical interface of the software, and the relevant data of the camera cannot be read.
What's the problem?
I hope you can answer. Thank you very much！

(Remarks:

System: Ubuntu 18.04.4 LTS

Ros:  melodic

kernel：Linux nx 4.9.140-tegra #34 SMP PREEMPT Tue Jul 6 15:43:05 CST 2021 aarch64 aarch64 aarch64 GNU/Linux）

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

MartyG-RealSense, 2022-01-11:
Did you build librealsense from source code using the native backend instructions?  If you did, then did you patch the kernel for Jetson L4T using the script **./scripts/patch-realsense-ubuntu-L4T.sh** before you built librealsense?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

linchen2020, 2022-01-14:
Hello, Marty g-realsense, I reinstalled the system. The system information is shown below：

Linux nx 4.9.201-tegra #210 SMP PREEMPT Wed Apr 28 15:05:40 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

***@***.***:~$ lsb_release  -a

No LSB modules are available.

Distributor ID: Ubuntu

Description:    Ubuntu 18.04.5 LTS

Release:        18.04

Codename:       bionic

After I update the realsense SDK and realsense ROS driver on this system according to the method provided by you, the camera can be used normally, but a warning will be prompted（as shown in the figure below），for this problem, can I delete any rule in these two items？

- Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

MartyG-RealSense, 2022-01-14:
It's excellent news that the Viewer and ROS wrapper are now working for you.

The multiple udev rules warning can occur if librealsense has been built from both packages and source code on the same computer, but it is simple to correct.  You should delete the **/etc** rule.  The warning should then disappear.

linchen2020, 2022-01-14:
The problems have been solved. Thank you very much，MartyG-RealSense

MartyG-RealSense, 2022-01-14:
You are very welcome, @linchen2020   :)

closed 2022-01-20 by MartyG-RealSense