realsenseai/realsense-ros #2218: librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1EiiiRKNS_7Scalar_IdEE

adamanov, 2022-01-10:
I built a realsense_ros as described however whenever I call a `roslaunch realsense2_camera rs_camera.launch` . 
I receive an error `librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1EiiiRKNS_7Scalar_IdEE` 

I do use Ubuntu 18 on VMWare 
`
robolab@ubuntu:~/jackal_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/robolab/.ros/log/dc0157ea-7220-11ec-a483-000c29a5719f/roslaunch-ubuntu-99862.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:37175/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [99882]
process[camera/realsense2_camera-2]: started with pid [99883]
[ INFO] [1641828556.061075394]: Initializing nodelet with 4 worker threads.
[ INFO] [1641828556.166160463]: RealSense ROS v2.3.2
[ INFO] [1641828556.166237654]: Built with LibRealSense v2.50.0
[ INFO] [1641828556.166255049]: Running with LibRealSense v2.50.0
[ INFO] [1641828556.201906443]:  
[ INFO] [1641828556.363191215]: Device with serial number 134722070743 was found.

[ INFO] [1641828556.363283110]: Device with physical ID 4-1-4 was found.
[ INFO] [1641828556.363325405]: Device with name Intel RealSense D435 was found.
[ INFO] [1641828556.363839775]: Device with port number 4-1 was found.
[ INFO] [1641828556.363913452]: Device USB type: 3.2
[ INFO] [1641828556.365612035]: getParameters...
[ INFO] [1641828556.420236126]: setupDevice...
[ INFO] [1641828556.420320048]: JSON file is not provided
[ INFO] [1641828556.420382177]: ROS Node Namespace: camera
[ INFO] [1641828556.420436425]: Device Name: Intel RealSense D435
[ INFO] [1641828556.420454214]: Device Serial No: 134722070743
[ INFO] [1641828556.420468046]: Device physical port: 4-1-4
[ INFO] [1641828556.420481152]: Device FW version: 05.13.00.50
[ INFO] [1641828556.420494434]: Device Product ID: 0x0B07
[ INFO] [1641828556.420508954]: Enable PointCloud: Off
[ INFO] [1641828556.420520956]: Align Depth: Off
[ INFO] [1641828556.420705535]: Sync Mode: Off
[ INFO] [1641828556.420944036]: Device Sensors: 
[ INFO] [1641828556.497502139]: Stereo Module was found.
[ INFO] [1641828556.522739030]: RGB Camera was found.
[ INFO] [1641828556.522920827]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1641828556.523110852]: num_filters: 0
[ INFO] [1641828556.523148500]: Setting Dynamic reconfig parameters.
[ INFO] [1641828559.355400898]: Done Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/robolab/jackal_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1EiiiRKNS_7Scalar_IdEE
[camera/realsense2_camera_manager-1] process has died [pid 99882, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/robolab/.ros/log/dc0157ea-7220-11ec-a483-000c29a5719f/camera-realsense2_camera_manager-1.log].
log file: /home/robolab/.ros/log/dc0157ea-7220-11ec-a483-000c29a5719f/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/robolab/.ros/log/dc0157ea-7220-11ec-a483-000c29a5719f/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

`

adamanov, 2022-01-11:
Hello @MartyG-RealSense, yes the problem was solved, by downloading the packages with `apt-get`. I guess this problem is somehow related to opencv by not sure. Anyway, I close the issue.

closed 2022-01-11 by adamanov

MartyG-RealSense, 2022-01-11:
Thanks very much @adamanov for the confirmation that you achieved a solution   :)