realsenseai/realsense-ros #2219: Jetson problems

sergioar3143, 2022-01-10:
Hello, I was using my camera. A few minutes ago everything was ok. I had a resolution 640x480 but I tried to change the resolution with the command 
roslaunch realsense2_camera rs_rgbd.launch depth_width:=640 depth_height:=360 depth_fps:=30 color_width:=640 color_height:=360 color_fps:=30
This command didn't work so I change to the original resolution but now don't work with any resolution.

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: camera/odom_in
 * /camera/realsense2_camera/unite_imu_method: none
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [8867]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 13fd9eea-723c-11ec-847f-1cbfce542967
process[rosout-1]: started with pid [8883]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [8912]
process[camera/realsense2_camera-3]: started with pid [8918]
process[camera/color_rectify_color-4]: started with pid [8919]
process[camera/points_xyzrgb_hw_registered-5]: started with pid [8920]
[ INFO] [1641836294.546604610]: Initializing nodelet with 4 worker threads.
[ INFO] [1641836296.044884610]: RealSense ROS v2.2.11
[ INFO] [1641836296.045008412]: Running with LibRealSense v2.49.0
[ INFO] [1641836296.081230808]:  
[ INFO] [1641836296.231695183]: Device with serial number 912112073420 was found.

[ INFO] [1641836296.231856068]: Device with physical ID /sys/devices/70090000.xusb/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0 was found.
[ INFO] [1641836296.232164610]: Device with name Intel RealSense D435I was found.
[ INFO] [1641836296.253375808]: Device with port number 2-1.1 was found.
[ INFO] [1641836296.266300807]: getParameters...
[ INFO] [1641836296.527477526]: setupDevice...
[ INFO] [1641836296.527575703]: JSON file is not provided
[ INFO] [1641836296.527648828]: ROS Node Namespace: camera
[ INFO] [1641836296.527715130]: Device Name: Intel RealSense D435I
[ INFO] [1641836296.527777682]: Device Serial No: 912112073420
[ INFO] [1641836296.527842214]: Device physical port: /sys/devices/70090000.xusb/usb2/2-1/2-1.1/2-1.1:1.0/video4linux/video0
[ INFO] [1641836296.527946797]: Device FW version: 05.12.15.50
[ INFO] [1641836296.528007943]: Device Product ID: 0x0B3A
[ INFO] [1641836296.528080339]: Enable PointCloud: Off
[ INFO] [1641836296.528129557]: Align Depth: On
[ INFO] [1641836296.528176589]: Sync Mode: On
[ INFO] [1641836296.530851172]: Device Sensors: 
[ INFO] [1641836296.532623151]: Stereo Module was found.
[ INFO] [1641836296.532987995]: RGB Camera was found.
[ INFO] [1641836296.533081276]: Motion Module was found.
[ INFO] [1641836296.533360287]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1641836296.533437682]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1641836296.533554818]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1641836296.533617214]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1641836296.535207630]: num_filters: 0
[ INFO] [1641836296.535283984]: Setting Dynamic reconfig parameters.
terminate called after throwing an instance of 'rs2::backend_error'
  what():  map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
[camera/realsense2_camera_manager-2] process has died [pid 8912, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/ser/.ros/log/13fd9eea-723c-11ec-847f-1cbfce542967/camera-realsense2_camera_manager-2.log].

What can I do?

MartyG-RealSense, 2022-01-10:
Hi @sergioar3143   Whilst your custom stream configuration in your roslaunch instruction looks correct, there are a couple of problems in your choice of librealsense and RealSense ROS wrapper versions.

1.  It is important to match the wrapper version as closely as possible with a specific librealsense version listed in the release notes of that particular wrapper version.  Wrapper 2.2.11 was designed for the significantly older librealsense **2.31.0**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.11

2.  The second problem is that although you are using librealsense 2.49.0, there is not a matching ROS wrapper for that librealsense version.  The next suitable librealsense and wrapper combination after 2.49.0 is librealsense **2.50.0** and wrapper **2.3.2**.

**librealsense**
https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

**RealSense ROS wrapper**
https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

sergioar3143, 2022-01-10:
How can I install the ROS wrapper?
When I did in the past, I followed this instructions https://github.com/JetsonHacksNano/installLibrealsense

sergioar3143, 2022-01-12:
Thank you, it works.

closed 2022-01-12 by sergioar3143

MartyG-RealSense, 2022-01-12:
That's great to hear, @sergioar3143 - thanks very much for the update!