realsenseai/realsense-ros #2220: Error frame ID for the right infrared image and camera info topic

hanyin-intel, 2022-01-11:
The `frame_id` filed of `/camera/infra2/camera_info` and `/camera/infra2/image_rect_raw` topic should be `camera_infra2_optical_frame` but now is `camera_infra1_optical_frame`. The camera I use is D435 but I think the issue is not related to specific device.

**How to reproduce**
In one terminal, run below command
```
ros2 launch realsense2_camera rs_launch.py device_type:=d435 initial_reset:=true enable_infra:=true infra_width:=1280 infra_height:=800 enable_infra1:=true enable_infra2:=true infra_fps:=15.0 enable_depth:=false enable_color:=false
```
In another terminal, run below command to see the `frame_id` field in `header`
```
ros2 topic list
ros2 topic echo /camera/infra2/camera_info
ros2 topic echo /camera/infra2/image_rect_raw
```

MartyG-RealSense, 2022-01-11:
Hi @hanyin-intel  May I first confirm whether you have the camera plugged into a USB 3 port, please?  Infrared 2 is not available when the camera is on a USB 2.1 connection.

If the camera is plugged into a USB 3 port then you can confirm whether it is being correctly detected as USB 3 by finding the line **Device USB type** near the top of the second half of the launch log in the terminal and seeing if it has '2.1' or '3.2' beside it.  If it is 2.1 then the camera has been detected as being on a USB 2 connection, whilst 3.2 means that it is detected as a USB 3 connection.

[image]

hanyin-intel, 2022-01-11:
Hi @MartyG-RealSense, thanks for your quick reply. My USB cable and port both support 3.0 protocol. The log output is `Device USB type: 3.2` and the `ros2 topic list` is like below.
```
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/theora
/camera/infra1/metadata
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/theora
/camera/infra2/metadata
/parameter_events
/rosout
/tf
/tf_static
```

MartyG-RealSense, 2022-01-11:
Thank you very much for the confirmation.  Next, please try changing the resolution in the roslaunch instruction to 1280x**720**, as 1280x800 is not a supported infrared resolution.  Using 1280x800 in a custom stream configuration in a roslaunch would likely cause the custom configuration to be deemed invalid and the camera's default configuration values would be applied to the launch instead.

hanyin-intel, 2022-01-11:
Thanks for your suggestion. Actually the d435 support both 1280x720 and 1280x800 infrared resolution, I have validate the infrared resolution configuration by `realsense-viewer`. If I select the 1280x800 one, the infrared image format can be Y8 or Y16 but I select he 1280x720, the infrared image format can only be Y8. 

And as your suggestion, I also changed the infrared resolution to 1280x720 by setting `infra_height:=720` but the error frame ID is still there...

MartyG-RealSense, 2022-01-11:
You are corect - though 1280x800 is not listed in the camera data sheet document, it is accessible in the Viewer.  The resolution was not offered on my main computer but was accessible from a D435i on another computer.

In https://github.com/IntelRealSense/realsense-ros/pull/1242 **Doronhi** the RealSense ROS wrapper developer discussed having the right camera frame_id equal to that of left camera, as this was required by some ROS packages such as required by StereoCameraModel

hanyin-intel, 2022-01-12:
> In #1242 **Doronhi** the RealSense ROS wrapper developer discussed having the right camera frame_id equal to that of left camera, as this was required by some ROS packages such as StereoCameraModel
Oh, I don't know such ROS convention previously, actually what I want to do is

closed 2022-01-12 by hanyin-intel

MartyG-RealSense, 2022-01-12:
Does the command below provide the desired result for you?

`rosrun tf tf_echocamera_infra2_optical_frame camera_infra1_optical_frame`

hanyin-intel, 2022-01-12:
Yes, that works. But the problem is if my program doesn't know the frame ID in advance, I have no way to know the infra2 frame ID in TF tree... Actually, I'm not only deal with the realsense input but also rosbag input or any other camera input. Anyway, I can expose this parameter to user or derive it from the infra1 frame ID. Thanks again!