realsenseai/realsense-ros #2224: Camera node do not publish image to topic /camera/color/image_raw when run rs_from_file.launch

ndinhtuan, 2022-01-18:
Here is my environment: 
- Ubuntu 20.04
- ROS Noetic
- RealSense ROS v2.2.13
- LibRealSense v2.50.0

I record bag file from Intel RealSense D435, I open Realsense Viewer normally. But when I use realsense2_camera package to publish image through message, **rs_from_file.launch**, I cannot receive any message at **/camera/color/image_raw**. 
After debugging, I realize that the number of subscribers of info_publisher and image_publisher equal 0 (in file base_realsense_node.cpp, function void BaseRealSenseNode::publishFrame). 
[image]
This is debug image: 
[image]

When I run file launch **rs_aligned_depth.launch**, I still receive message from **/camera/color/image_raw** topic.
This is debug image:
[image]

I'm wondering about what difference between **rs_from_file.launch** and **rs_aligned_depth.launch**. Do anyone see that error ? and why there are no subscriber of info_publisher and image_publisher. Thank you very much.

MartyG-RealSense, 2022-01-18:
Hi @ndinhtuan  Ideally, the RealSense ROS wrapper version should be matched as closely as possible with the librealsense SDK version listed in the wrapper version's release notes.  The correct ROS wrapper to match with SDK 2.50.0 and ROS Noetic is **2.3.2**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

You do not need to install librealsense again to perform the update, just to build the 2.3.2 wrapper from source code.

In regard to the launch, a bag file can only play back the data that was recorded into it as it is not using a live camera.  When recording is activated in the RealSense Viewer, it will record all of the streams that were currently enabled at the time that the record option was clicked.  Did you have a color stream active when recording the bag in the Viewer, please? 

The **rs_from_file.launch** will access the recorded contents of the bag file, **rs_aligned_depth.launch** will use live camera data by default and not the bag and so it would not have a problem with displaying a live stream from the camera's color sensor.

ndinhtuan, 2022-01-18:
@MartyG-RealSense  Thank you for replying.
I will try to build wrapper from source code again with version 2.3.2 and give you feedback soon.
For bagfile 

> Did you have a color stream active when recording the bag in the Viewer, please?

Yes, I do have color stream in bag file, and I also open bag file from Realsense Viewer, it run normally. I also use `rosbag play`, it also works.
Actually, I use `roslaunch`, sometimes it works, particularly I run 10 times, only 2 times I receive message from **/camera/color/image_raw**, remain 8 times, I see nothing happens.

MartyG-RealSense, 2022-01-18:
Thanks very much @ndinhtuan - I look forward to your test results with wrapper 2.3.2.  Good luck!

ndinhtuan, 2022-01-19:
Hi @MartyG-RealSense , I already installed realsense wrapper version 2.3.2. But I still got the same error.
[image]

MartyG-RealSense, 2022-01-19:
Can you share your full roslaunch log from the terminal for **rs_from_file.launch** in a comment below, please?

doronhi, 2022-01-19:
I think it is related to a known issue with playing back rosbag files with the wrapper.
It is mentioned in the known issues part in the release description: "Problems running from a bag file (using the "rosbag_filename" option)."
It is on my to-do list but I still didn't get around to it.

ndinhtuan, 2022-01-20:
@MartyG-RealSense I would like to share you the log of roslaunch from the terminal:
[image]

MartyG-RealSense, 2022-01-20:
@ndinhtuan  I don't see anything wrong in your kindly provided launch log.  As Doronhi said, your problem may be related to the current known issue with bag playback that he referred to.

ndinhtuan, 2022-01-23:
@MartyG-RealSense  Thank you very much for your support. I will find more information about the error that Doronhi mentioned.

MartyG-RealSense, 2022-01-23:
You are very welcome, @ndinhtuan    :)

closed 2022-02-04 by MartyG-RealSense