realsenseai/realsense-ros #2225: Could I download "ros-melodic-realsense2-camera_2.3.1-1bionic.20210921.214447_amd64.deb" file?

samlin0419, 2022-01-18:
Dear Sir,

When I using "ros-melodic-realsense2-camera_2.3.2-1bionic.20211115.142723_amd64.deb" to build Docker image will get error "1 package failed: realsense2_camera", but this error wouldn't appear version 2.3.1, please see the attached log file.

So, Could I download "ros-melodic-realsense2-camera_2.3.1-1bionic.20210921.214447_amd64.deb" file?

Thanks a lot.

Best regards,
Sam
[ROS_v0.9.0_build_log.txt](https://github.com/IntelRealSense/realsense-ros/files/7887317/ROS_v0.9.0_build_log.txt)

samlin0419, 2022-01-18:
Attached another build error log file.
[build_failed_log_for_ros-melodic-realsense2-camera_2.3.2.txt](https://github.com/IntelRealSense/realsense-ros/files/7887447/build_failed_log_for_ros-melodic-realsense2-camera_2.3.2.txt)

MartyG-RealSense, 2022-01-18:
Hi @samlin0419  Your log indicates that you are using librealsense SDK version **2.45.0**.  Is this correct, please?

Ideally, the RealSense ROS wrapper version should be matched as closely as possible with the librealsense SDK version listed in the wrapper version's release notes.  Wrapper version 2.3.2 is designed for use with librealsense 2.50.0.  So if you are using SDK 2.45.0 then the best match with that SDK version would be ROS wrapper **2.3.0**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.0

If you need to build from packages instead of source code then it may be easier to build librealsense 2.50.0 and wrapper 2.3.2 together at the same time using the ROS wrapper's **Method 1** procedure.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

The release of librealsense 2.50.0 was also accompanied by a Docker guide that Intel published.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

samlin0419, 2022-01-18:
> The release of librealsense 2.50.0 was also accompanied by a Docker guide that Intel published.

Dear Sir,

Thanks for your kindly and quickly reply, I am using librealsense SDK version 2.45.0 it's right, I will reference your comment to try it, I will update the test result for you.

Thanks a lot.

Best regards,
Sam

samlin0419, 2022-01-19:
> Hi @samlin0419 Your log indicates that you are using librealsense SDK version **2.45.0**. Is this correct, please?
>
> Ideally, the RealSense ROS wrapper version should be matched as closely as [...]

Dear Sir,

I had already using librealsense SDK version from 2.45.0 to 2.50.0 , but still build failed,  could you help me to solve it?

Please see the attached three build log files.

Thanks a lot.
[build_docker_builder.log](https://github.com/IntelRealSense/realsense-ros/files/7893784/build_docker_builder.log)
[build_docker_prod.log](https://github.com/IntelRealSense/realsense-ros/files/7893785/build_docker_prod.log)
[dependencies.txt](https://github.com/IntelRealSense/realsense-ros/files/7893791/dependencies.txt)

Best regards,
Sam

MartyG-RealSense, 2022-01-19:
My knowledge of Docker is limited unfortunately, so I cannot offer advice on debugging your Docker setup.

There is a relatively recent example of a Dockerfile for RealSense ROS in the link below that was updated 6 months ago at the time of writing this, so that may be a useful reference to compare to your own ROS Dockerfile.

 https://github.com/iory/docker-ros-realsense

This dockerfile references ROS Kinetic, so you may have to edit the contents to Melodic.

Another approach to running RealSense ROS in Docker is here:

https://github.com/ripl/realsense-D400-ros-docker

samlin0419, 2022-01-21:
> My knowledge of Docker is limited unfortunately, so I cannot offer advice on debugging your Docker setup.
>
> There is a relatively recent example of a Dockerfile for RealSense ROS in the link below [...]

Dear Sir,

Thank you very much for your helping and assistance.

Best regards,
Sam

MartyG-RealSense, 2022-01-21:
You are very welcome, @samlin0419  :)

closed 2022-02-02 by MartyG-RealSense