realsenseai/realsense-ros #2226: L515 depth message has no output in ROS

rolandying, 2022-01-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| ROS version                     | Melodic  | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu18.04  | 
| Kernel Version (Linux Only)    | 5.4.0-89-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 
| Language                            |  C++/Python                        | 
|Serial No.			|     f0264437                      | 
|Product ID			|     0B64                  | 

### Issue Description
When I was running the command:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true device_type:=l515 tf_prefix:=measured/camera`
Everything was fine. Both RGB and depth messages were OK, but something went wrong yesterday, I can still receive the depth message, but they are empty. To be more precise, there is no problem at the beginning, but after I run the ROS node, the above-mentioned problems will occur after a minute or so.
**I upload two videos of the whole process on youtube; they can show the problem more clearly**. Here are the links: when I run the `realsense-viewer`, [it works perfectly](https://www.youtube.com/watch?v=aiwKPmcZZ5g). 
Then I started running the ROS node, [something happened](https://www.youtube.com/watch?v=4iKQtj4jTTg). In the second video, before 1:14, the depth message is OK but with some flicker, but after that, it has just gone. 
And when I try to check the message by running the command: `rostopic echo -n1 /camera/aligned_depth_to_color/image_raw`, the output :
```
header: 
  seq: 1844
  stamp: 
    secs: 1642593577
    nsecs: 586819887
  frame_id: "measured/camera_color_optical_frame"
height: 720
width: 1280
encoding: "16UC1"
is_bigendian: 0
step: 2560
data: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 153, 11, 153, 11, 153, 11, 153, 11, 153, 11, 153, 11, 153, 11, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
```
They are all 0, too long to copy them all.

And no specific output information when an error occurs. The terminal is just keeping output the:
e, number: 61
```
 19/01 15:13:28,549 WARNING [139816150759168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/01 15:13:29,799 WARNING [139816150759168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/01 15:13:31,050 WARNING [139816150759168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/01 15:13:32,301 WARNING [139816150759168] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```
According to some replies, they can be ignored.

Every time such an error occurs, I can only temporarily solve the problem by plugging and unplugging the USB cable, because the problem will happen again after a minute.

So does someone know how to handle such things?

MartyG-RealSense, 2022-01-19:
Hi @rolandying   As the L515 camera works fine in the Viewer and works fine in the RealSense ROS wrapper for the first minute before the depth messages disappear after that, it does not seem to be a problem with the camera hardware.  

I see that you are using a PC computer (not a computing device such as Nvidia Jetson or Raspberry Pi) and that your kindly provided video shows that you are using librealsense SDK version 2.48.0 and RealSense ROS wrapper 2.3.1 (which is the correct SDK and wrapper version combination).  Your USB port type is 3.2.  All of those details look fine.

The **/camera/depth/image_rect_raw** topic and the image are publishing correctly at first and then flicker and disappear after the first minute, as you described.  

Does the problem still occur after the first minute if you use a simpler roslaunch instruction?

**roslaunch realsense2_camera rs_camera.launch align_depth:=true**

Also, is the L515 camera being used in a location where there are not infrared light sources such as sunlight presence, as these can interfere with L515 - this camera model is designed for use indoors under controlled lighting conditions.

rolandying, 2022-01-20:
> Hi @rolandying As the L515 camera works fine in the Viewer and works fine in the RealSense ROS wrapper for the first minute before the depth messages disappear after that, it does not seem to be a problem with the camera hardware.
> 
> I see that you are using a PC computer (not a computing device such as Nvidia Jetson or Raspberry Pi) and that your kindly provided video shows that you are using librealsense SDK version 2.48.0 and RealSense ROS wrapper 2.3.1 (which is the correct SDK and wrapper version combination). Your USB port type is 3.2. All of those details look fine.
> 
> The **/camera/depth/image_rect_raw** topic and the image are publishing correctly at first and then flicker and disappear after the first minute, as you described.
> 
> Does the problem still occur after the first minute if you use a simpler roslaunch instruction?
> 
> **roslaunch realsense2_camera rs_camera.launch align_depth:=true**
> 
> Also, is the L515 camera being used in a location where there are not infrared light sources such as sunlight present in the scene, as these can interfere with L515 - this camera model is designed for use indoors under controlled lighting conditions.

Hi, @MartyG-RealSense You can see the lighting condition is very good from the RGB image message, I'm using this camera in the indoor environment, the only light source is the artificial light source.
And even if I run the basic roslaunch instruction:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true`
The depth message would still disappear after serval seconds.

MartyG-RealSense, 2022-01-20:
Thank you very much for the detailed information.  Does the depth disappearance still occur if align_depth=true is not included in the roslaunch instruction?

rolandying, 2022-01-20:
> Thank you very much for the detailed information. Does the depth disappearance still occur if align_depth=true is not included in the roslaunch instruction?

First, I just test the camera in realsense-viewer, it also comes to the problem:
[image]
The output error is:

```
20/01 15:37:37,948 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:37,981 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:38,014 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:38,047 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:38,080 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:38,114 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:38,147 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 20/01 15:37:38,180 ERROR [140143709124352] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
```
Then, I ran the most basic command:
`roslaunch realsense2_camera rs_camera.launch`
The problem also comes.

MartyG-RealSense, 2022-01-20:
Have you tried unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the base of the camera, please (USB-C cables are two-way insertion at the micro-sized end). 
 Does doing so make any difference?

MartyG-RealSense, 2022-01-20:
Bearing in mind the **Out of frame resources** error in the Viewer log earlier, could you use an Ubuntu system monitoring tool such as **htop** to check the CPU usage percentage to see whether it is close to or at 100% when the depth and RGB streams are enabled in the Viewer.  This would indicate whether there is a lack of system resources for processing the frames because the CPU is being over-burdened.

rolandying, 2022-01-21:
> Bearing in mind the **Out of frame resources** error in the Viewer log earlier, could you use an Ubuntu system monitoring tool such as **htop** to check the CPU usage percentage to see whether it is [...]

I don't think it's because of the over-burdened of the CPU, the CPU on my PC is pretty powerful.
[image]

But anyway, this is the screencast of the Htop.
[image]

And today, I connect the camera with another PC, the problem still happens.
So I think it may be a hardware level issue.

MartyG-RealSense, 2022-01-21:
What is the camera's temperature reading in the Viewer after the image disappears?

[image]

rolandying, 2022-01-21:
@MartyG-RealSense I can't catch the perfect moment when the camera died, but the normal running temperature is like:
[image]

Are these acceptable?

MartyG-RealSense, 2022-01-21:
The data sheet document for the L515 camera recommends an operating temperature of 30 degrees C (which I have quoted from the sheet below), so the temperature values on your camera look excessive, especially if they rose to that level within a minute.

[image]

Does the casing of the camera feel hot to the touch after only a minute?

rolandying, 2022-01-21:
@MartyG-RealSense It's not always come to 50+ degrees in just one minute, but it always will come to such a temperature in serval minutes, like 10mins.
And yes, it feels hot when I touch the camera.

MartyG-RealSense, 2022-01-21:
Rapid heating and the outer casing becoming quickly hot to the touch can indicate that there is an overheating problem in the USB system, due to a glitch in the USB port or a bad USB cable.  As you experience the same problem on another computer, this may indicate that the USB cable is the cause.  Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice?

rolandying, 2022-01-21:
> Rapid heating and the outer casing becoming quickly hot to the touch can indicate that there is an overheating problem in the USB system, due to a glitch in the computer's USB port or a bad USB cable. As you experience the same problem on another computer, this may indicate that the USB cable is the cause. Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice?

I'm using a longer cable. And it is installed at the end of a robot arm, the official USB cable is not long enough.

MartyG-RealSense, 2022-01-21:
USB-C cables are known to be susceptible to quality problems, so if your own choice of cable is used and you are extending beyond 1 meter then choosing a high quality cable is important.  Aside from quality issues, if a cable is longer than 2 meters and is not premium grade then you may experience performance degradation and even disconnections.

The company **Newnex** are a supplier of industrial grade USB equipment that has been specifically validated for use with RealSense cameras.

https://www.newnex.com/realsense-3d-camera-connectivity.php

rolandying, 2022-01-21:
> USB-C cables are known to be susceptible to quality problems, so if your own choice of cable is used and you are extending beyond 1 meter then choosing a high quality cable is important. Aside from [...]

But the question is I have been doing this for nearly half a year, nothing happened until serval days ago, so you mean suddenly a problem happened on the cable?
I would have a further test tomorrow with the official cable. Let's see how the temperature goes.

MartyG-RealSense, 2022-01-21:
There is a good article in the link below about how USB cables can wear or break over time, especially if they are made of lower-quality materials.

https://sortatechy.com/why-usb-cables-stop-working/

I look forward to your temperatue test results.  Good luck!

rolandying, 2022-01-22:
Hi, @MartyG-RealSense, One thing I forgot to tell you is during the last temperature test, the camera worked great, even at 57 degrees C. I waited for 40mins and the camera is still working. And today I test the camera for nearly 3 hours. No error had ever occurred. This really makes me confused.
Here are the new testing results.
**First is the starting temperature:**
[image]
 After each test, I will put the camera outdoors for 5 mins, so the cooldown process would be much faster. And sometimes the temperature is even lower than those data in the pic above. So I would run the camera first and wait until the temperature becomes the same value in the picture before I start timing. So I'm trying to make sure the whole process is fair.

**Second, the official cable temperature at 10mins and 20mins:**

[image]

[image]

**Third, the unofficial cable temperature at 10mins and 20mins:**
[image]
[image]

**Fourth, the unofficial cable with an extension cable temperature at 10mins and 20mins:**(The unofficial cable is still not long enough sometimes, so we add an extension USB cable. It is the same situation with our daily work at the hardware level.)
[image]
[image]

**Last, the unofficial cable with an extension cable temperature at 10mins and 20mins when the camera is running on the ROS node:**(It is the same situation with our daily work at both hardware and software levels. One thing needs to be explained, I need to quickly shut down the ROS node and switch to the Realsense-viewer to read the current temperature at the 10 and 20 minute time points. So the data may have some bias.)
[image]
[image]

So after all the tests, I have two points of view:
1. Different cables and software don't have a big influence on the temperature.
2. The USB ports do make a difference when I connect the camera to the strong power supply USB port with a lightning symbol, the temperature of the camera would rise much more rapidly. (The temperature would come to 57 in less then 15mins)
PS: **One more thing, throw all the tests today, the camera works great. I'm got nothing to say, I don't know why this happens. :(**

MartyG-RealSense, 2022-01-22:
Thanks very much for the very detailed feedback and the PS.  Let's keep this case open for a while longer to confirm that the camera remains functional in subsequent tests.

rolandying, 2022-01-22:
> Thanks very much for the very detailed feedback and the PS. Let's keep this case open for a while longer to confirm that the camera remains functional in subsequent tests.

Yes, it's what I want to say. If this error happens again, I'll post it here immediately.

MartyG-RealSense, 2022-01-28:
Hi @rolandying  Has the error occurred again yet or does the problem seem to be resolved?

rolandying, 2022-01-28:
> Hi @rolandying Has the error occurred again yet or does the problem seem to be resolved?

Nope, the error has never happened again. I think it's been solved. Although I don't know what's going on. 😂

MartyG-RealSense, 2022-01-28:
Thanks very much @rolandying for the update!   As the problem appears to have been solved, I will close the case.  Thanks again!  :)

closed 2022-01-28 by MartyG-RealSense

rolandying, 2022-01-28:
> Thanks very much @rolandying for the update! As the problem appears to have been solved, I will close the case. Thanks again! :)

Thanks for your continued help！