realsenseai/realsense-ros #2227: How to change the frame rate of D455 camera?

BSSNBSSN, 2022-01-19:
Hi,
I'm using a D455 camera on ROS and I want to change the frame rate of the RGB camera and depth camera to 60 fps, but I can't save the adjustment in realsense-viewer. So I'd like to ask how to change the frame rate of the D455 camera and save the adjustment.

MartyG-RealSense, 2022-01-19:
Hi @BSSNBSSN  If you are using 1280x720 RGB resolution then the maximum FPS supported is 30.  The lower 848x480 RGB resolution supports 60 FPS.

[image]

Changes to the selected resolution are not saved when the Viewer is closed and will return to the defaults (e.g 1280x720 at 30 FPS for RGB).

If you define a resolution and FPS in a ROS launch then this will not change the resolution and FPS settings in the RealSense Viewer.

If you wish to save a particular resolution / FPS in ROS then you can create a custom version of the launch file and set the color_width, color_height and color_fps values to a configuration that will achieve 60 FPS (color_width = 848, color_height = 480, color_fps = 60).

You can also define a custom stream configuration in the roslaunch instruction without having to edit the launch file by including the three factors of height, width and FPS.  For example:

**roslaunch realsense2_camera rs_camera.launch color_width:=848 color_height:=480 color_fps:=60**

BSSNBSSN, 2022-01-19:
That helps a lot, thank you very much!

closed 2022-01-19 by BSSNBSSN