realsenseai/realsense-ros #2231: What's the recommended way of matching infra1/metadata with its corresponding infra1/image_rect_raw?

martinakos, 2022-01-21:
I want to match infra1/metadata with it's corresponding infra1/image_rect_raw. Both topics are posted at 30Hz. I tried to match them by timestamp. 
I've tried to use a TimeSynchronizer but it can't match any pair of these messages.
Then I tried an ApproximateTimeSynchronizer with a slop parameter of 0.01 (which I think it should be more than high enough) and queue_size of 200. I can match some pairs this way. However, the rate of matching varies over time. To test this I publish a new message each time there is a match and I check the Hz of this new message. In theory this new message should have 30Hz but in practice it reaches 30Hz occasionally but it then slows down. The immediate rate seems to go in bursts. I would assume that a  TimeSynchronizer should be enough to match by timestamp as the metadata belongs to the infra1 frame. What's going on? Any recommendations?

MartyG-RealSense, 2022-01-21:
Hi @martinakos  No metadata topics for infra1 and infra2 are in the topic lists in the ROS and ROS2 wrapper documentation (just ones for depth, color and gyro / accel) and I could find no references about RealSense ROS wrapper infra metadata topics in my research.

**ROS topics**
https://github.com/IntelRealSense/realsense-ros#published-topics

**ROS2 topics**
https://github.com/IntelRealSense/realsense-ros/tree/ros2#published-topics

martinakos, 2022-01-24:
The metadata topics are new to realSense ROS v2.3.2
They look like this:
```
$ rostopic echo /d435i/infra1/metadata
header: 
  seq: 0
  stamp: 
    secs: 1643015327
    nsecs: 306697607
  frame_id: "d435i_infra1_optical_frame"
json_data: "{\"frame_number\":5795,\"clock_domain\":\"global_time\",\"frame_timestamp\":1643015327306.697510,\"\
  frame_counter\":5795,\"hw_timestamp\":204349834,\"sensor_timestamp\":204345584,\"\
  actual_exposure\":8500,\"gain_level\":16,\"auto_exposure\":0,\"time_of_arrival\"\
  :1643015327315,\"backend_timestamp\":0,\"actual_fps\":30,\"frame_laser_power\":150,\"\
  frame_laser_power_mode\":1,\"exposure_priority\":1,\"exposure_roi_left\":0,\"exposure_roi_right\"\
  :847,\"exposure_roi_top\":0,\"exposure_roi_bottom\":479,\"frame_emitter_mode\":1,\"\
  raw_frame_size\":814080,\"gpio_input_data\":248,\"sequence_name\":15,\"sequence_id\"\
  :0,\"sequence_size\":2}"
```
I'm trying to implement the solution that @dorodnic suggested in #1657 . Basically a node that listens to the infra1/infra2 and  depth/color/points and based on the information in their corresponding metadata topics it will publish two sets of topics one with the infra1/infra2 and depth/color/points with the emitter on and another set with the emitter off. However, matching the metadata topics to their corresponding data topics based on timestamp alone is not working very well for me. I thought this was a separate issue to the one in #1657 so I posted a new issue rather than continue there. Do you or @dorodnic  have some suggestion on how to match the metadata messages with their corresponding data messages reliably?

MartyG-RealSense, 2022-01-24:
I will highlight your question to **Doronhi** the RealSense ROS wrapper developer.  Thanks very much for your patience!

doronhi, 2022-01-24:
Hi,
Using `python3 catkin_ws/src/realsense/realsense2_camera/scripts/echo_metadada.py /camera/infra1/metadata` in Terminal1 and `rostopic echo /camera/infra1/image_rect_raw/header` in Terminal2 and then stopping them (first Terminal1 and then, as quickly as possible Terminal2) and scrolling Terminal2 up a little, I can see the following:
The field "frame_timestamp" in the `/camera/infra1/metadata` topic has the same value as the header/stamp in the messages of the `/infra1/image_rect_raw'  topic.

Terminal1:
```
frame_number:6400      
clock_domain:global_time
frame_timestamp:1643027028496.9358
hw_timestamp:247155248 
time_of_arrival:1643027028504
backend_timestamp:1643027028497
```

Terminal 2:
```
---
seq: 1096
stamp: 
  secs: 1643027028
  nsecs: 496935844
frame_id: "d435i_infra1_optical_frame"
---
```
This is consistent with the code as can be seen [here](https://github.com/IntelRealSense/realsense-ros/blob/f400d682beee6c216052a419f419e95b797255ad/realsense2_camera/src/base_realsense_node.cpp#L2455)

I am not familiar with TimeSynchronizer or with ApproximateTimeSynchronizer but the matching code should use these timestamps and not the message timestamp which could be subject to bursts.

Regarding the documentation, The metadata for depth, infra1, and infra2 is the same since they all relate to the same sensor. Still, it is missing from the documentation as well as any explanation regarding the metadata. I'll try to fix that soon.

P.S.
@dorodnic is a great guy but not related to that specific issue :-)

martinakos, 2022-01-24:
Ok. Thanks. I'll try to match these two timestamps. I wonder if you know of any existing TimeSynchronizer extension that can allow for custom timestamp fields? otherwise I'll try to modify the code of TimeSynchronizer to achieve this. 

PS: tagging dorodnic was a mistake I intended to tag you @doronhi

martinakos, 2022-01-28:
So my interest in this data/ metadata topic matching is to implement the interlacing of laser on/off as discussed in https://github.com/IntelRealSense/realsense-ros/issues/1657 So that I can have a point cloud with laser and IR images without laser. 

I found that if the point cloud is set to use texture from the RGB stream (pointcloud_texture_stream=RS2_STREAM_COLOR) is not possible to match the timestamps between the IR image streams and their metadata topics. There is a systematic drift in the timestamps of the messages, even if I match using the frame_timestamp in the metadata. However, if I set the pointcloud_texture_stream=RS2_STREAM_ANY I can match the timestamps without a problem. In this case, the timestamps are the same so I can match them with a TimeSynchronizer without needing any slop. 
Is this a feature or a bug?

doronhi, 2022-01-30:
Sounds like an unintended behavior at the very least. I'll check.

doronhi, 2022-01-30:
When synchronization is on, the header timestamp of all the images streams gets the same timestamp. This is done to allow synchronization between the streams. The chosen timestamp for the set is the one that belongs to the first frame in the set.  When the color stream is enabled, the whole frameset may get the timestamp of the color frame and so the infra1 images are sent with a timestamp that is slightly wrong. The metadata, however, is being sent with the timestamp of the infra1 image.
To fix that we need to give the metadata messages a timestamp that matches the one given to the images. This is done in #2235.

closed 2022-02-01 by doronhi

Pran-Seven, 2022-10-04:
@martinakos were you able to synchronize the topics using the timestamps or were you able to pas custom timestamps to the time synchronizer?

PR #2235 (2022-02-01): Fix Metadata timestamps
Aim to fix #2231, This PR sets metadata messages header time stamp to match images time stamp.
When synchronization is enabled, all the frames in a frameset get the same timestamp, the one that belongs to the first frame in the set. This is done to enable synchronization between them.
The header of the Metadata topics now gets the same timestamp. The individual frame's time is available in the data of the message.