realsenseai/realsense-ros #2232: How to set the depth range of realsense?

YouShaoze, 2022-01-26:
For example, I want to set the effective distance as 0.2m ~ 5.0m，But I can't find the configuration file that can set this value, and there is a point cloud inside the camera in the map

MartyG-RealSense, 2022-01-26:
Hi @YouShaoze   You can define a minimum and maximum depth sensing range in the RealSense Viewer using a post-processing **threshold filter**.  This can be applied in the RealSense Viewer tool by going to the **Post Processing** section of the Stereo Module controls in the Viewer's options side-panel, expanding open the list of filters and expanding open the Threshold Filter.  You can then define the desired minimum and maximum values and enable the filter.

[image]

If you need to integrate the filter into your own script then the links below provide resources for doing so.

**C#**
https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533

**Python**
https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301
https://www.programmersought.com/article/11995485679/

**C++**

https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds

If you are using a D435 or D435i model, these have a minimum depth sensing range of 0.1 m.  If an L515 (0.25 m), D415 (0.3 m) or D455 (0.4 m) is being used though then you would need to increase the value of an option called **Disparity Shift** in order to reduce the minimum depth to 0.2 m.  Increasing the Disparity Shift value to reduce the minimum distance also reduces the maximum observable depth distance.

MartyG-RealSense, 2022-02-03:
Hi @YouShaoze  Do you require further assistance with this case, please?  Thanks!

YouShaoze, 2022-02-07:
> 

Thank you for your reply~~ This is very useful : )

MartyG-RealSense, 2022-02-11:
I'm pleased that I was able to be helpful.  As you did not require further assistance, I will close this case.  Thanks again!

closed 2022-02-11 by MartyG-RealSense

tugbakara, 2023-07-18:
> Hi @YouShaoze   If you are using the RealSense ROS wrapper then you can set a maximum depth sensing distance using the **clip_distance** instruction.  There is not an instruction for setting a minimum distance though.
> 
> **rtabmap_ros** does offer the ability to define a minimum distance with a setting called **RangeMin**
> 
> https://github.com/introlab/rtabmap_ros/issues/413
> 
> *********
> 
> If using the RealSense SDK (librealsense), you can define a minimum and maximum depth sensing range in the RealSense Viewer using a post-processing **threshold filter**.  This can be applied in the RealSense Viewer tool by going to the **Post-Processing** section of the Stereo Module controls in the Viewer's options side-panel, expanding open the list of filters and expanding open the Threshold Filter.  You can then define the desired minimum and maximum values and enable the filter.
> 
> [image]
> 
> If you need to integrate the filter into your own script then the links below provide resources for doing so.
> 
> **C#**
> https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533
> 
> **Python**
> https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301
> https://www.programmersought.com/article/11995485679/
> 
> **C++**
> 
> https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds
> 
> If you are using a D435 or D435i model, these have a minimum depth sensing range of 0.1 m.  If an L515 (0.25 m), D415 (0.3 m) or D455 (0.4 m) is being used though then you would need to increase the value of an option called **Disparity Shift** in order to reduce the minimum depth to 0.2 m.  Increasing the Disparity Shift value to reduce the minimum distance also reduces the maximum observable depth distance.

If I set parameter that you mentioned above, will this affect ROS  side? I mean this feature will be still set on ROS?

MartyG-RealSense, 2023-07-18:
Hi @tugbakara  I do not know whether setting Disparity Shift specifically in the Viewer will affect the ROS wrapper, but in general I have seen reports from RealSense ROs users who stated that a setting they made in the Viewer affected that setting in the ROS wrapper.  So I recommend giving it a try and see what happens.

tugbakara, 2023-07-18:
> Hi @tugbakara  I do not know whether setting Disparity Shift specifically in the Viewer will affect the ROS wrapper, but in general I have seen reports from RealSense ROS wrapper users who stated that a setting they made in the Viewer apparently affected that setting in the ROS wrapper.  So I recommend giving it a try and see what happens.

Actually, I mentioned about threshold filter.
I need more accurate depth data (MORE THAN 2 m) from ROS side, that's why I am asking

MartyG-RealSense, 2023-07-18:
The ROS wrapper does not support the threshold filter (for defining a minimum and maximum depth distance).  You can only set the maximum distance with the wrapper's **clip_distance** parameter.  For example: `clip_distance:=1.5`

The threshold filter will not improve the camera's accuracy over distance though as this depends primarily on the particular RealSense camera model that you are using.  For example, D455 has 2x better accuracy over distance than D435, meaning that D455 has the same accuracy at 6 m that D435 does at 3 m.