realsenseai/realsense-ros #2234: Integrate an external IMU with L515. Failing to cope with aerial indoor changes

hichamhendy, 2022-01-28:
I am trying the camera L515 with rtabmap. However, I think the IMU inside the camera isn't enough the Mapping, especially my platform is aerial and moves sometimes with high acceleration which isn't a problem for the odometry yet the IMU. I decided to use an external IMU with high rate to offset the effect of this problem. Nonetheless, I can't figure the transformation into camera_imu_optical_frame. I managed to get the internal IMU with internal IMU in the camera. But I need to provide a kind of "static_transform" which I really can't figure out! I tried the following but the the cloud is kind of flipped
[image]

IMU rotation is like that 

`  <node pkg="tf" type="static_transform_publisher" name="camera_aligned" args=" 0 0 0 -1.57 0 -1.57 /camera_link /camera_imu_optical_frame  500"/>`

Here (https://github.com/IntelRealSense/librealsense/tree/master/doc) I can only find information regarding D series and T. Yet I ain't sure what from L515.
 of course with internal camera IMU
I get the cloud ordinarily
[image]

My Question: How is the the IMU in l515 is built in? and How are its coordinates transformed into camera transformation. More precisely the transformation between camera_link frame and camera_imu_optical_frame frame.

MartyG-RealSense, 2022-01-28:
Hi @hichamhendy  The L515 uses the **Bosch BMI085** IMU component.  Other IMU-equipped RealSense cameras (D435i, D455) have similarly used Bosch BMI components. The link below leads to technical information about the L515's BMI085.

https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi085/

Although the Bosch data sheet quotes a sensitivity range between -+ 2g and -+ 16g, https://github.com/IntelRealSense/librealsense/issues/3991 disclosed that RealSense limits the supported sensitivity range to -+ 4g in the 400 Series IMU-equipped models.  I would assume that the same is true for the L515.

There is a case about the L515 and the **Overwriting previous data! Make sure IMU is published faster than data rate** error on the RTABMAP forum at https://github.com/introlab/rtabmap/issues/687 that may be a helpful reference.

A RealSense team member advises in https://github.com/IntelRealSense/librealsense/issues/8249 that the coordinate system information for the D435i model applies to the L515 too.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system

hichamhendy, 2022-02-01:
I fixed the IMU in the same way to get the linear acceleration with ENU-coordinates
[image]

I also stream static tf using 
`<node pkg="tf" type="static_transform_publisher" name="camera_aligned" args="0 0 0 -1.571 0 -1.571 /camera_link /camera_imu_optical_frame  500"/>`
, which is identical to the original transform published from the realsense_driver. However, I get the map flipped on left side!!

hichamhendy, 2022-02-01:
I made it to align an external IMU with camera coordinates

`  <node pkg="tf" type="static_transform_publisher" name="camera_aligned" args="0 0 0 -1.571 0 -1.571 /camera_link /camera_imu_optical_frame  500"/>`
[image]

One must reconfigure the IMU to stream on ENU and be aligned with internal IMU. Publish both parallel  till you get the correct alignment.

closed 2022-02-01 by hichamhendy