realsenseai/realsense-ros #2238: Which camera Frame in ROS is used when using align_depth:=True?

joshuawillman, 2022-02-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | v2.50.0 | 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    | 5.11.0-43-generic  | 
| Platform                                 | ROS  |
| SDK Version                            |  2.0                          | 
| Language                            |  Python                          | 
| Segment			|  Robot                          | 

Hi, 

I simply want to know for clarification: which camera or point on the camera is used as the origin point when `align_depth:=True`? The following list contains the available Frames provided by ROS. 
```
Frame: camera_accel_frame
Frame: camera_accel_optical_frame
Frame: camera_bottom_screw_frame
Frame: camera_color_frame
Frame: camera_color_optical_frame
Frame: camera_depth_frame
Frame: camera_depth_optical_frame
Frame: camera_gyro_frame
Frame: camera_gyro_optical_frame
Frame: camera_infra1_frame
Frame: camera_infra1_optical_frame
Frame: camera_infra2_frame
Frame: camera_infra2_optical_frame
Frame: camera_link
```

I tried `camera_color_optical_frame` and `camera_depth_optical_frame`, but I am not sure which is correct. The project I am working on involves moving a robot to the point generated by the camera, and an small error of about 1.5 cm is produced in the x and y directions when actually going to the point. 

I have looked at other issues, such as #1179, but couldn't find a clear answer. 
Thank you so much for your help!

MartyG-RealSense, 2022-02-02:
Hi @joshuawillman  I could not find any information about the origin in regards to the RealSense ROS wrapper.  In librealsense though, the 0,0,0 origin of the coordinate system of the 400 Series cameras is the center of the left infrared imager.

joshuawillman, 2022-02-02:
@MartyG-RealSense 
Thank you so much for your reply!
Out of curiosity, is the left infrared imager labeled as 1 or 2? What I mean is, would that refer to the `camera_infra1_optical_frame`?

MartyG-RealSense, 2022-02-02:
Yes, the left infrared is infra1 and the right infrared is infra2.

The section of ROS wrapper documentation in the link below states that **camera_infra1_optical_frame** refers to the original coordinate system, whilst **camera_infra1_frame** is the coordinate system in the ROS convention.

https://github.com/IntelRealSense/realsense-ros#about-frame-id

joshuawillman, 2022-02-02:
Thank you so much for your help. @MartyG-RealSense!

closed 2022-02-02 by joshuawillman