realsenseai/realsense-ros #2241: How can I get quaternions from RealSense 515

xyxegle, 2022-02-04:
Required Info |  
-- | --
Camera Model | L515
ROS version | Melodic
Firmware Version |  Running with LibRealSense v2.49.0   FW version: 01.05.08.01
Operating System & Version | Ubuntu18.04
Kernel Version (Linux Only) | 5.4.0-89-generic
Platform | PC
Language | C++/Python
Serial No. | f1062522
Product ID | 0B64

I'm going to do an experiment that uses quaternions. I have enabled enable_gyro and enable_accel in the launch file and assigned unite_IMu_method to "copy". Now I have successfully obtained the /camera/ IMU topic, but when I subscribed to the topic, I found that the orientation part of the topic was null, may I ask if I have the wrong configuration there?

This is the topic content THAT I got, and you can see that the orientation section is always empty
[image]

This is the modified content in my launch file. Line 52 is assigned "copy" in addition lines 35 and 36 is assigned to "true".

[image]

xyxegle, 2022-02-04:
I have also assigned 'Unite IMu method' to Linear Interpolation，for further test.What really bothers me is that linee_acceleration's parameters get worse and worse as time goes by.
[image]
Do you have any suggestions？

MartyG-RealSense, 2022-02-04:
Hi @xyxegle   In https://github.com/IntelRealSense/realsense-ros/issues/1449#issuecomment-711675699 **Doronhi** the RealSense ROS wrapper developer suggests displaying the rotation quaternions using the command below:

**rosrun tf tf_echo camera_depth_optical_frame camera_gyro_optical_frame**

xyxegle, 2022-02-04:
[image]
It looks like there's something wrong

MartyG-RealSense, 2022-02-04:
If you are referring to the messages about the IMU not being calibrated and the frame time domain being HARDWARE_CLOCK, this should not affect the retrieval of quaternions as far as I am aware.

xyxegle, 2022-02-04:
However, I did not receive any information about quaternion as you said.

MartyG-RealSense, 2022-02-04:
Did you try an **echo** command of the gyro as suggested above in https://github.com/IntelRealSense/realsense-ros/issues/2241#issuecomment-1029762758 please?

xyxegle, 2022-02-04:
Yes, I tried the method you said, but it doesn't seem to work, the upper left corner of the previous answer is the result of running the command

MartyG-RealSense, 2022-02-04:
Given that you are using **unite_imu_method** to publish the IMU as a single 'imu' topic instead of separate gyro and accel topics, does the echo work if you do it like this:

**rosrun tf tf_echo camera_depth_optical_frame camera_imu_optical_frame**

xyxegle, 2022-02-04:
[image]
It does show some data, but the data doesn't look normal. It only displays data at time 0.000, but does not update over time. Also, I wonder if there is any way to integrate it into the launch file.

MartyG-RealSense, 2022-02-04:
There is a discussion at https://github.com/IntelRealSense/realsense-ros/issues/2097 about 0.000 time when performing a tf echo.

I do not personally know of a method for activating a rosrun tf echo from within a launch file, since rosrun is a process that is performed **after** launch is completed.

xyxegle, 2022-02-04:
After several experiments, I think the solution you give should not be the camera quaternion information. I think it's more like relative position information inside the sensor. I changed the position, direction and Angle of the sensor, but I got the same data after every restart, which I think is abnormal. Do you have any other solutions?

MartyG-RealSense, 2022-02-04:
The extrinsic transforms describe the fixed position and rotation relationship between two sensors.  What **would** in real-time update when a camera that is equipped with an IMU is rotated is the gyro xyz angles (roll, pitch, yaw).

xyxegle, 2022-02-04:
So how do I get the Euler Angle of the IMU output?

The 'unite IMU method' provides both linear and angular velocities. But how do I get euler angles(roll\pitch\yaw)?

xyxegle, 2022-02-04:
[image]

[image]

As you can see, this is gyro and Accel output separately. There is no euler Angle content. How can I get it?

xyxegle, 2022-02-04:
[image]

MartyG-RealSense, 2022-02-04:
In https://github.com/IntelRealSense/realsense-ros/issues/1932#issuecomment-864157132 a RealSense ROS user also obtained the IMU quaternions using a ROS package called **imu_tools**

xyxegle, 2022-02-10:
Thank you very much. I have solved the problem.I have created a new problem, could you help me to solve it?Here's a link to that question：https://github.com/IntelRealSense/realsense-ros/issues/2250

MartyG-RealSense, 2022-02-10:
It's excellent to hear that you found a solution for your quaterions issue, and I have replied at your other question.  I will now close this quaternion issue.   Thanks again!

closed 2022-02-10 by MartyG-RealSense