realsenseai/realsense-ros #2242: Using 5 D450 cameras, libusb "cannot allocate memory"

v-lopez, 2022-02-04:
I am trying to use 5 D450 cameras on the same NUC pc.

- All are connected through USB 3.0, and individually they all run fine. 
- PC has 32GB of memory and a i7-1165G7, and is idle except for this camera experiment.
- I am able to start 4 of them (any 4, not particular ones). But the 5th one has some libusb errors and doesn't publish anything.
- I am subscribing to Pointcloud, RGB, Infra1 and Infra2 at 640x480@30Hz.
- Cameras are not synchronized by hardware
- If I disable color, infra1 or infra2, the  **5 cameras work fine.**
- Lowering resolution to 640x360,  the  **5 cameras work fine.**
- Lowering frequency from 30fps to 15fps, **5th camera still fails**.

I'm attaching the full log of the 5th camera, but I believe the significant part is:
```
[ INFO] [1643974878.788031802]: setupStreams...
 04/02 12:41:18,827 WARNING [140231957280512] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 04/02 12:41:19,100 WARNING [140231328122624] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 130 error: Cannot allocate memory, number: 12
 04/02 12:41:19,100 ERROR [140231328122624] (dispatcher.cpp:35) Dispatcher [0x7f8a442f6678] exception caught: failed to submit UVC request while start streaming
 04/02 12:41:19,173 WARNING [140231957280512] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

[5th_realsense.txt](https://github.com/IntelRealSense/realsense-ros/files/8002974/5th_realsense.txt)

MartyG-RealSense, 2022-02-04:
Hi @v-lopez  You have a powerful 11th generation processor that should be capable of supporting 5 cameras simultaneously (and can do at lower settings or reduced number of topics, as you described above).  

Are your cameras attached to a USB hub?  If they are then you may be using up the available data **bandwith** on the hub when running all 5 cameras with multiple topics or a higher resolution / FPS.  In Intel's multiple camera white-paper document, they published tables estimating the bandwidth usage for up to 6 cameras in different streaming configurations when attached directly to a PC or to a hub. 

 https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-2-multi-camera-considerations

If you are using a hub and the maximum available data bandwidth on the hub is being exceeded, you could consider adding a second hub and putting some of the cameras on it.  The USB standard allows 5 hubs to be linked together.

v-lopez, 2022-02-04:
Hi @MartyG-RealSense , thanks for the quick response.

The PC is this: https://download.asrock.com/IPC/Download/e-catalog/NUC%20BOX-1135G7.pdf

I am not using any Hub, I am using 2 cameras through USB-C to USB-C cables, and 3 cameras through USB-C to USB-A. Using all the front and back USB ports of the NUC.

MartyG-RealSense, 2022-02-04:
If all the cameras are attached directly to the computer's USB ports then they will be drawing all of their power from the computer's power supply unit (PSU), and the stability of power supply to the USB port may be less than that of a mains electricity powered USB hub.   I recommend budgeting up to 2 watts power consumption for each attached camera.

If you are using your own choice of cable instead of the official USB A to C cables then the cable should ideally be a premium quality one due to the high volume of data bandwidth that RealSense cameras can transmit through the cable.  USB-C cables can have quality issues, and a lower quality cable could cause degraded performance or disconnections, especially when the cable is 2 meters or longer.

v-lopez, 2022-02-04:
The cables are USB 3.2, and 0.5m long.
Also, individually the cameras (or even in groups of 4) they don't give any problems.
The power supplies up to 90W. 

And it still would not explain, why disabling infra1 solves the problem. Because both infrared projectors are still running if I request depth information.
Also lowering resolution solves the problem.

I believe the problem might be related to what was commented here https://github.com/jetsonhacks/buildLibrealsense2Xavier/issues/17#issuecomment-578540324

v-lopez, 2022-02-04:
Unfortunately we need all of them, rgb, infra1, infra2 and pointcloud. 
What could be the cause of the out of memory error, and how to solve it?

MartyG-RealSense, 2022-02-04:
You may be reaching the practical limits of the number of RealSense cameras that can be handled on the same computer with multiple streams from each of the 5 cameras.  Even in 2018 Intel were recommending a maximum of 4 cameras on the same computer on an i7 machine.  The more cameras that are added, the more of the computer's resources that are consumed.

If adding a second PC is viable for your project, Intel have a RealSense ROS guide for dividing multiple RealSense cameras across 2 PCs and communicating between the machines with **ROS_MASTER_URI**

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

"If the message **messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11**  is generating continuously multiple times per second then it indicates a serious communication problem between the code and the device.

You could try adding **initial_reset:=true** to your roslaunch instruction to reset the camera at launch to see whether it reduces the occurrence of the control_transfer returned messages.

v-lopez, 2022-02-04:
I tried with **initial_reset:=true** without luck.

I tried with the 5 cameras on separate nodes, as well as all on the same node. The total CPU usage is always below 200% (this machine has 8 cores, so the max would be 800%). The computer should be able to handle much more CPU/memory wise. 

I think it's related to the libusb, but I have no clue as to how it's used.

MartyG-RealSense, 2022-02-04:
If you build the librealsense SDK from source code with CMake and include the build flag **-DBUILD_WITH_OPENMP=true** then YUY to RGB conversion and depth-color alignment operations can take advantage of multiple cores, reducing latency at the expense of greater CPU utilization.  Your log states that you have align_depth = false though, so you may not get a noticable performance advantage from using this method.

If your NUC has Windows installed as the asrock specification suggests then librealsense would be using **Windows Media Foundation** as its backend.  If the CMake build term **-DFORCE_RSUSB_BACKEND=true** is used then the SDK build on Windows should use a backend based on a WinUSB rewrite of the UVC protocol instead.

v-lopez, 2022-02-04:
Thanks I'll try the OPENMP thing.
I'm actually running this on Ubuntu 20.04.

MartyG-RealSense, 2022-02-07:
RealSense 400 Series cameras and the librealsense SDK are designed to run on a very wide range of configurations, from very low spec to high spec.   So unless a command to make use of multiple cores is used (such as using 'make -j8' for faster compilation when building librealsense with CMake or using the flag DBUILD_WITH_OPENMP=true for multi-core support of alignment and RGB color conversion) then it is likely to just utilize a single core by default.

v-lopez, 2022-02-07:
Thanks for those suggestions, we'll take them into account.

Regarding

> "If the message messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11 is generating continuously multiple times per second, as it is in your log, then it indicates a serious communication problem between the code and the device.

> You could try adding initial_reset:=true to your roslaunch instruction to reset the camera at launch to see whether it reduces the occurrence of the control_transfer returned messages.

It doesn't help. And I have lots of these errors even with just one camera being used:

```
[ INFO] [1644253875.986105255]: Initializing nodelet with 8 worker threads.
[ INFO] [1644253876.081249400]: RealSense ROS v2.3.2
[ INFO] [1644253876.081280031]: Built with LibRealSense v2.50.0
[ INFO] [1644253876.081293082]: Running with LibRealSense v2.50.0
[ INFO] [1644253876.120088381]:  
 07/02 18:11:16,196 WARNING [139807225280256] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[ INFO] [1644253876.290269357]: Device with serial number 141322251965 was found.

[ INFO] [1644253876.290298879]: Device with physical ID 2-3-2 was found.
[ INFO] [1644253876.290309469]: Device with name Intel RealSense D455 was found.
[ INFO] [1644253876.290551985]: Device with port number 2-3 was found.
[ INFO] [1644253876.290569784]: Device USB type: 3.2
[ INFO] [1644253876.290589478]: Resetting device...
[ INFO] [1644253882.378880228]:  
[ INFO] [1644253882.482439732]: Device with serial number 141322251965 was found.

[ INFO] [1644253882.482492973]: Device with physical ID 2-3-4 was found.
[ INFO] [1644253882.482504819]: Device with name Intel RealSense D455 was found.
[ INFO] [1644253882.482728249]: Device with port number 2-3 was found.
[ INFO] [1644253882.482739811]: Device USB type: 3.2
[ INFO] [1644253882.484699952]: getParameters...
[ INFO] [1644253882.542020946]: setupDevice...
[ INFO] [1644253882.542061006]: JSON file is not provided
[ INFO] [1644253882.542079919]: ROS Node Namespace: 
[ INFO] [1644253882.542104326]: Device Name: Intel RealSense D455
[ INFO] [1644253882.542131205]: Device Serial No: 141322251965
[ INFO] [1644253882.542163868]: Device physical port: 2-3-4
[ INFO] [1644253882.542198083]: Device FW version: 05.13.00.50
[ INFO] [1644253882.542218029]: Device Product ID: 0x0B5C
[ INFO] [1644253882.542233883]: Enable PointCloud: On
[ INFO] [1644253882.542256002]: Align Depth: Off
[ INFO] [1644253882.542277367]: Sync Mode: On
[ INFO] [1644253882.542354334]: Device Sensors: 
[ INFO] [1644253882.639024266]: Stereo Module was found.
[ INFO] [1644253882.660249801]: RGB Camera was found.
[ INFO] [1644253882.660516893]: Motion Module was found.
[ INFO] [1644253882.660566899]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1644253882.660611553]: Add Filter: pointcloud
[ INFO] [1644253882.661340054]: num_filters: 1
[ INFO] [1644253882.661372527]: Setting Dynamic reconfig parameters.
[ INFO] [1644253885.900500400]: Done Setting Dynamic reconfig parameters.
[ INFO] [1644253885.901126449]: depth stream is enabled - width: 640, height: 360, fps: 15, Format: Z16
[ INFO] [1644253885.901349105]: infra1 stream is enabled - width: 640, height: 360, fps: 15, Format: Y8
[ INFO] [1644253885.901577441]: infra2 stream is enabled - width: 640, height: 360, fps: 15, Format: Y8
[ INFO] [1644253885.902018805]: color stream is enabled - width: 640, height: 360, fps: 15, Format: RGB8
[ INFO] [1644253885.902049355]: setupPublishers...
[ INFO] [1644253885.904621740]: Expected frequency for depth = 15.00000
[ INFO] [1644253885.946353304]: Expected frequency for infra1 = 15.00000
[ INFO] [1644253885.970564747]: Expected frequency for infra2 = 15.00000
[ INFO] [1644253885.994972501]: Expected frequency for color = 15.00000
[ INFO] [1644253886.023514413]: setupStreams...
 07/02 18:11:26,426 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/02 18:11:26,476 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/02 18:11:26,526 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1644253886.577154773]: SELECTED BASE:Depth, 0
 07/02 18:11:26,577 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1644253886.591329055]: RealSense Node Is Up!
[ WARN] [1644253886.810813957]: 
 07/02 18:11:26,811 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... 22 lines cut ...]
 07/02 18:11:59,896 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/02 18:12:00,588 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/02 18:12:01,948 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 07/02 18:13:31,956 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/02 18:13:39,012 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/02 18:13:43,065 WARNING [139806872950528] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2022-02-07:
Are the above errors from a camera on a USB A to C cable or a C to C cable?

MartyG-RealSense, 2022-02-07:
The suggestion is based on a small number of past reports I have seen on this forum of worse performance when using C to C cables compared to A to C.

For example, the RealSense-compatible Dot3D scanning software suggests in their troubleshooting documentation link below to make sure that the C-C cable supports 10 GBps of data transfer bandwidth.

https://desk.zoho.com/portal/dotproduct/en/kb/articles/rsconnectionissues

v-lopez, 2022-02-09:
Hi @MartyG-RealSense,

I can confirm that it happens with each camera, these errors are more common on USB-C to USB-C cables, but they do happen on USB-C to USB-A as  well.

What worries me, is that every now and then, one of the cameras will stop publishing PCL during 3-5 seconds.

For instance, here are the logs of such an occurrence, I am printing every second the messages received in the last second, for each type of image, and the delay of those messages (current timestamp - msg stamp).

This is with the cameras configured at 15Hz, and 640x360 resolution on all topics.
```
[ INFO] [1644398683.880265363]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 68.3645ms
[ INFO] [1644398684.880275275]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 67.4969ms
[ INFO] [1644398685.880255000]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 69.0093ms
[ INFO] [1644398686.880262294]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 67.8849ms
[ INFO] [1644398687.880362929]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 69.2213ms
[ERROR] [1644398688.880237569]: camera3 Hz PCL:12RGB: 15 INFRA1: 16 INFRA2: 16 Delay: 63.7662ms
[ WARN] [1644398688.915350940]: No stream match for pointcloud chosen texture Process - Color
[ERROR] [1644398689.880215493]: camera3 Hz PCL:0RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 38.2141ms
[ERROR] [1644398690.880238386]: camera3 Hz PCL:0RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 39.4432ms
[ERROR] [1644398691.880242730]: camera3 Hz PCL:0RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 41.0831ms
[ INFO] [1644398692.880271039]: camera3 Hz PCL:14RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 68.3124ms
[ INFO] [1644398693.880273894]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 69.9559ms
[ INFO] [1644398694.880268492]: camera3 Hz PCL:15RGB: 15 INFRA1: 15 INFRA2: 15 Delay: 68.5208ms
```

I see the cameras are identified as using 5 GBps of bandwidth, do you have any cable recommendations?
```
$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    |__ Port 1: Dev 2, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 1: Dev 2, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 1: Dev 2, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 3: Dev 3, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 3: Dev 3, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 3: Dev 3, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 3: Dev 3, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 3: Dev 3, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 3: Dev 3, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 4: Dev 4, If 1, Class=Video, Driver=usbfs, 5000M
    |__ Port 4: Dev 4, If 4, Class=Video, Driver=usbfs, 5000M
    |__ Port 4: Dev 4, If 2, Class=Video, Driver=usbfs, 5000M
    |__ Port 4: Dev 4, If 0, Class=Video, Driver=usbfs, 5000M
    |__ Port 4: Dev 4, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
    |__ Port 4: Dev 4, If 3, Class=Video, Driver=usbfs, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 9: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 9: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    |__ Port 4: Dev 8, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 8, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 8, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 8, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 8, If 4, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 4: Dev 8, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
```

MartyG-RealSense, 2022-02-09:
The company **Newnex** are a supplier of industrial grade USB cables and equipment and provide dedicated support for RealSense cameras.

https://www.newnex.com/realsense-3d-camera-connectivity.php

Intel test the cameras for long-running (hours and days).  If the camera remains within recommended operating tmeperature tolerances then they should be able to run indefinitely.  If an error does occur after a long period of time has passed during a long-run session then it is typically due to a non-camera factor such as a glitch in the USB cable / USB port, a computer hardware problem or an operating system issue.

v-lopez, 2022-02-09:
Thanks will take a look at that cable provider.

I was also able to test with V4L, and I am getting better results in all aspects.

I am running 5 cameras simultaneously, with no periods missing data and I am running them at 640x480 (with the other driver I could only reach 640x360). 

After half an hour or so the Projector temperature for all cameras is sitting between 40-45º. 

So for now it seems like switching to V4L has solved all the issues I had.

MartyG-RealSense, 2022-02-09:
That's great to hear that V4L has provided a significant improvement for you, @v-lopez - thanks very much for the update!

MartyG-RealSense, 2022-02-15:
Hi @v-lopez  Do you require further assistance with this case, please?  Thanks!

v-lopez, 2022-02-15:
No, all good!

closed 2022-02-15 by v-lopez