realsenseai/realsense-ros #2245: Not working on ROS2 Galactic/Rolling

yushijinhun, 2022-02-06:
Due to [a breaking change made in ROS2 Galactic](https://github.com/ros2/rclcpp/pull/1522), `realsense2_camera_node` will crash on startup:
```
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
  what():  parameter 'depth_module.auto_gain_limit_toggle' has invalid type: cannot undeclare an statically typed parameter
```
The solution is to set `dynamic_typing` flag in `ParameterDescriptor` (see https://github.com/yushijinhun/realsense-ros/commit/5958859bf2400882b20e0859239c22ec7302545c), but this flag is introduced in Galactic and doesn't work on older ROS2 releases.

MartyG-RealSense, 2022-02-06:
Hi @yushijinhun  I have not seen this error occur previously.  Are you using the most recent ROS2 wrapper version **4.0.1** or the older **3.2.3** that is also compatible with librealsense SDK version 2.50.0?

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.3

yushijinhun, 2022-02-06:
I'm using the latest commit from [`ros2-beta` branch](https://github.com/IntelRealSense/realsense-ros/commit/af449f8563a7eb0c87a42f038d50ede645d6fb80) and ROS2 Rolling. My camera is RealSense D455.

MartyG-RealSense, 2022-02-06:
As this particular error has not been reported before as far as I am aware, it's very useful that you have highlighted it in your report as it may be an issue that was introduced in the recent 4.0.1 rather than the older 3.2.3.  Thanks!

doronhi, 2022-02-07:
Hi @yushijinhun 
In Galactic, The error exists in version 3.2.3 when running with "hdr_merge" filter:
`ros2 launch realsense2_camera rs_launch.py filters:=hdr_merge`

Version 4.0.1 introduced extensive usage of `rclcpp::Node::undeclare_parameter` function which causes this error.
The solution you propose works. I want to make sure that this is the intended approach before applying it. To that end, I posted [question #395760](https://answers.ros.org/question/395760/ros2-galactic-undeclare_parameter-raises-an-exception) in ros_answers.com. I hope to get a clear understanding of the issue soon and fix it accordingly.

yushijinhun, 2022-02-09:
I confirm that it has been fixed in #2247.

closed 2022-02-15 by yushijinhun

doronhi, 2022-02-15:
Thanks for noticing the issue and confirming the correction.

PR #2247 (2022-02-09): ROS2-beta: fix galactic issue with undeclaring parameters
Aim to fix #2245 
Fix dynamic_params syntax.
Fix issue with Galactic parameters set by default to static which prevents them from being undeclared.

This PR needs some more testing before merging. Will appreciate users' reviews.