realsenseai/realsense-ros #2248: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

grafoteka, 2022-02-08:
My hardware and software are：Jetson Nano 4GB; L515; ubuntu18.04; ros-melodic; librealsense v2.50.0.
I installed the packages and libraries with the official ros packages. 
```
$ sudo apt install ros-melodic-realsense2-camera
$ sudo apt install ros-melodic-realsense2-description
```

I can use the realsense-viewer without any problem.

Then I found that when running the ros driver, the following warnings always appear, and the warnings are always generated when the program is running. How can I solve it? What are the consequences of this warning?

It should be noted that this warning will always appear with a certain frequency (~ 1 sec). 

The warning is as following: 
`(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

I have try with 30FPS, 15FPS and 6FPS, but I have the same the error with all the configurations. 

```
$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/robotica/.ros/log/f05ae44c-8911-11ec-a457-6466b3258e07/roslaunch-robotica-jetson-1-15627.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robotica-jetson-1:41971/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
[... 92 lines cut ...]
[ INFO] [1644347072.547114524]: Device Sensors: 
[ INFO] [1644347072.637612288]: L500 Depth Sensor was found.
[ INFO] [1644347072.683273831]: RGB Camera was found.
[ INFO] [1644347072.683886897]: Motion Module was found.
[ INFO] [1644347072.683997374]: num_filters: 0
[ INFO] [1644347072.684063420]: Setting Dynamic reconfig parameters.
[ INFO] [1644347075.542863604]: Done Setting Dynamic reconfig parameters.
[ INFO] [1644347075.543942901]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1644347075.546699403]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1644347075.546855039]: setupPublishers...
[ INFO] [1644347075.555228928]: Expected frequency for depth = 30.00000
[ INFO] [1644347075.666690457]: Expected frequency for color = 30.00000
[ INFO] [1644347075.760143751]: setupStreams...
 08/02 20:04:36,023 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1644347076.060246190]: SELECTED BASE:Depth, 0
 08/02 20:04:36,073 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/02 20:04:36,125 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1644347076.180157141]: RealSense Node Is Up!
[ WARN] [1644347076.209516397]: 
[ WARN] [1644347076.209678804]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 08/02 20:04:36,428 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/02 20:04:36,579 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/02 20:04:36,731 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/02 20:04:37,076 WARNING [547071455616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2022-02-08:
Hi @grafoteka  If the message **messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11**  is generating continuously multiple times per second then it indicates a serious communication problem between the code and the device.

You could try adding **initial_reset:=true** to your roslaunch instruction if you have not done so already in order to reset the camera at launch to see whether it reduces the occurrence of the control_transfer returned messages.  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

Also, may I ask whether you are using the official 1 meter USB cable that was supplied with the camera or a longer cable of your own choice, or a USB **C to C** cable instead of the USB A to C (full sized A, micro-sized C) that the official cable has. 

If you are using a long cable of your own choice (2 meters or more) then it is important that it is a high quality cable due to the high volume of data that a RealSense camera can transmit through the cable.

If it is a C to C type cable of your own choice then it is recommendable that it is one that can support 10Gbps of data transfer bandwidth.

MartyG-RealSense, 2022-02-09:
Thanks very much @grafoteka - I look forward to hearing about the results of the jumper and cable connector reversal checks.

grafoteka, 2022-02-09:
Hi @MartyG-RealSense, I have bad news.

- The jetson was correctly configure to use the barrel jack and the maximum power. I check the jumper and the command.
- Turning the cable didn't solve the problem.

Also, I have notice that with the realsense-viewer, the error is shown.

Finally, I try also installing the realsense2 udev rules, but still having the same error.

MartyG-RealSense, 2022-02-09:
It sounds as though you installed librealsense and the RealSense ROS wrapper together from packages when you used ** sudo apt install ros-melodic-realsense2-camera**

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

Can you confirm please that you did not install librealsense separately by another method before that.  If librealsense was installed separately first and then ' sudo apt install ros-melodic-realsense2-camera' was used, it would install librealsense a second time and may result in a conflict.

grafoteka, 2022-02-09:
I have remove all the related installed to realsense (realsense-camera, realsense-description, librealsense). Then reboot the jetson nano. And after, reinstall with the method #1.

But when launching the realsense (`roslaunch realsense2_camera rs_camera.launch initial_reset:=true`) still having the same error:
`09/02 21:04:19,210 WARNING [546450690432] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

And there is also one warning message: `[ WARN] [1644437055.835520802]: 
[ WARN] [1644437055.835682423]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.`

MartyG-RealSense, 2022-02-09:
Is the camera data able to be viewed in the **RViz** interface despite the warnings?  RViz can be activated with the instruction **rosrun rviz rviz** after launch has completed with the message 'RealSense node is up'

grafoteka, 2022-02-09:
Yes. I can visualize the realsense in Rviz, but the message still being published

MartyG-RealSense, 2022-02-09:
If you can visualize the camera data in RViz without degraded performance then the warnings can be safely ignored.

grafoteka, 2022-02-09:
I can visualize only the depth cloud, not the point clouds 2. 
[image]

MartyG-RealSense, 2022-02-09:
There are known issues with enabling the pointcloud filter on Jetson boards specifically, such as missing color and FPS drops.  These problems typically do not occur when the pointcloud is not enabled.  There is extensive information about the issue at https://github.com/IntelRealSense/realsense-ros/issues/1967

A Nano user recently shared in that discussion a librealsense, ROS wrapper and firmware driver combination that worked for them at https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-1029789663

grafoteka, 2022-02-14:
Hello @MartyG-RealSense, I have continue doing some tests, but I have detected one problem. The RealSense does not show any RGB picture. 

The topic `/camera/color/image_raw` is created, but no image is published, therefore it is not possible to use some packages like RTabMap. 

On the other hand, I have connected directly the RealSense to a desktop computer (intel x64) and it works correctly (even it shows the same error of this topic) the RGB image is shown. 

Do you know what can cause this problem in the RealSense?

Thank you

MartyG-RealSense, 2022-02-14:
As mentioned above in https://github.com/IntelRealSense/realsense-ros/issues/2248#issuecomment-1034222668 - if you have the pointcloud filter enabled on Jetson then color may be missing and should return when the pointcloud filter is disabled.

MartyG-RealSense, 2022-02-20:
Hi @grafoteka  Do you require further assistance with this case, please?  Thanks!

grafoteka, 2022-02-22:
I'm stuck in how to create the GUI but the packages compiles correctly with your instructions.

closed 2022-02-22 by grafoteka