realsenseai/realsense-ros #2249: Machine is frozen when I run and close wrapper many times.

kdhRGT, 2022-02-09:
Hello!

Now I am using a D435 RGB-d camera for developing robot applications.

But I have a problem that sometimes, the machine is frozen when I restart the wrapper.

It looks ok when I just run the wrapper one or two times, but my machine is frozen if I run the wrapper pkg many times.

I am making a service robot, so sometimes it happens to close all of pkg.. 

And I wanna solve it for making a stable system.

Could you give me some advice for solving this problem?

I guess this problem happens because of wrapper problem or intel sdk 

I had experiments in Laptop and lattepanda(embbeded board). But i happened same time.

Thank you so much

Best regards.

MartyG-RealSense, 2022-02-09:
Hi @kdhRGT  May I first ask which kernel version you are using, please?  

There was a recent case at https://github.com/IntelRealSense/librealsense/issues/10207 where a RealSense ROS user was experiencing OS freezes.  Their eventual solution that provided stability was to use kernel version **5.11.0-34-generic**

kdhRGT, 2022-02-09:
Thank you so much for your advice, that is what I was finding.

best regards

MartyG-RealSense, 2022-02-09:
You are very welcome, @kdhRGT   :)

closed 2022-02-09 by kdhRGT