realsenseai/realsense-ros #2251: D435 camera not getting connected to Raspberry Pi 4 - 'No device connected'

roshankaanthmks, 2022-02-11:
| Required Info |
| Camera | Model D435 |
| Firmware Version | 5.13.0.50 |
| Operating System & Version | Ubuntu 20.04 - Noetic |
| Platform | Raspberry Pi 4 |
| Librealsense Version | SDK 2.42 |
| Segment  | Robotics |

Errors: [Untitled document.docx](https://github.com/IntelRealSense/realsense-ros/files/8052175/Untitled.document.docx)

We tried to connect D435 camera with Raspberry Pi 4 and it disconnects frequently throwing an error 'No device connected'. We used the attached python code (pupil_data_zsweep.txt) to connect the camera.  We are getting this error:

Traceback (most recent call last):
File "opencv_viewer_example.py", line 19, in
pipeline.start(config)
RuntimeError: No device connected

Also, we are ending up with errors as shown in the attached docx file when we tried to launch roslaunch realsense2_camera rs_camera.launch

[pupil_data_zsweep.txt](https://github.com/IntelRealSense/realsense-ros/files/8052179/pupil_data_zsweep.txt)

MartyG-RealSense, 2022-02-12:
Hi @roshankaanthmks  May I first confirm that you do not have another program already accessing the camera when running your Python script or launching ROS (for example, running the Python script whilst there is an active ROS session, or roslaunching whilst your Python script is already running).  

Once a program claims access to a particular stream on a particular camera, another program cannot access that stream until the first program releases its claim on it (by stopping streaming or closing the program).  This is because of the librealsense SDK's **Multi-Streaming Model** rules in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

I note that the traceback error relates to the Python example program **opencv_viewer_example.py** and not the script code in your **pupildata_zsweep.txt**.  This makes me wonder whether the 'pupildata_zsweep' was launched first and it claimed access to the camera, and then 'opencv_viewer_example.py' was launched secondly after that and could not start the pipeline because 'pupildata_zsweep' had already placed an access lock on the camera.

********

Regarding the launch log in your kindly provided docx file, the librealsense and ROS wrapper versions (2.42.0 and 2.2.22) are correctly matched, so there should not be a problem related to a version mismatch.  The **control_transfer returned** warnings after launch completes indicates that there may be a communication problem between the wrapper and the camera.  

Sometimes camera data can still be viewed normally in the RViz interface despite continuously generating 'control_transfer returned' warnings.  Are you able to successfully display the depth and color streams if you launch RViz with the command **rosrun rviz rviz** after roslaunch has completed (shown by the message 'RealSense Node is Up!')

MartyG-RealSense, 2022-02-19:
Hi @roshankaanthmks  Do you require further assistance with this case, please?  Thanks!

roshankaanthmks, 2022-02-19:
The issue is resolved. thanks!

closed 2022-02-19 by roshankaanthmks

MartyG-RealSense, 2022-02-20:
That's great to hear that your issue is resolved, @roshankaanthmks - thanks very much for the update!