realsenseai/realsense-ros #2253: pointcloud for an rgb image

Chezhian531, 2022-02-13:
I wish to generate pointcloud for an rgb image just like the example shown in https://github.com/IntelRealSense/realsense-ros using the comment 

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud

Could you please suggest how can i achieve this?

I have ubuntu 18.04, intelrealsense D435i

MartyG-RealSense, 2022-02-14:
Hi @Chezhian531  Once you have successfully launched RealSense ROS with that roslaunch instruction then you should be able to view the pointcloud by launching the RViz interface with the command **rosrun rviz rviz**

https://github.com/IntelRealSense/realsense-ros#point-cloud

[image]

Chezhian531, 2022-02-15:
I dont get that output. I am attaching my rviz output her
[image]
e

Chezhian531, 2022-02-15:
Also I want to generate such an output for the image I have, could you suggest some ways to do it?

MartyG-RealSense, 2022-02-15:
Are you able to generate a point cloud if you use the **Add** button at the bottom of the RViz side-panel to add the topic **depth/color/points/PointCloud2**, as described at https://github.com/IntelRealSense/realsense-ros/issues/1605#issuecomment-754523910

Chezhian531, 2022-02-16:
Yes I do, but is it possible to obtain point cloud for an image?

MartyG-RealSense, 2022-02-16:
Do you mean converting a pre-made file into a pointcloud instead of using live camera data, please?  And if so, what kind of file would you be using (PNG image, bag file, ply pointcloud file, etc).

Chezhian531, 2022-02-16:
yes exactly, a premade file. I want to get point cloud for a jpg image.

MartyG-RealSense, 2022-02-16:
If you need to perform the image file to pointcloud conversion in ROS then I could not find much that offered this feature, though the ROS package in the link below may be a good place to start.

http://wiki.ros.org/pixel_2_3d

closed 2022-02-28 by MartyG-RealSense