realsenseai/realsense-ros #2254: Imu not working--realsense D435i

bhaskar-glitch, 2022-02-15:
Hi,
I'm having **Realsense D435i** camera. But the **imu** of the camera isn't working like on realsense viewer when I'm moving the camera the axis is moving on it but on ros it's not working..also i'm having a JETSON XAVIER NX on which I'm doing the ros stuff.
 
**Result of ROSTOPIC LIST**
/camera/accel/imu_info
/camera/align_to_color/parameter_descriptions
/camera/align_to_color/parameter_updates
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_color/image_raw/theora
/camera/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/theora/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/gyro/imu_info
/camera/imu
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/infra1/metadata
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/image_rect_raw/theora
/camera/infra2/image_rect_raw/theora/parameter_descriptions
[... 170 lines cut ...]
[ INFO] [1644916157.415055170]: Add Filter: pointcloud
[ INFO] [1644916157.417224619]: num_filters: 2
[ INFO] [1644916157.417357415]: Setting Dynamic reconfig parameters.
[ INFO] [1644916161.267245508]: Done Setting Dynamic reconfig parameters.
[ INFO] [1644916161.269063222]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1644916161.269764292]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1644916161.270471858]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1644916161.271966540]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1644916161.277885879]: gyro stream is enabled - fps: 200
[ INFO] [1644916161.278061746]: accel stream is enabled - fps: 250
[ INFO] [1644916161.278157872]: setupPublishers...
[ INFO] [1644916161.290060771]: Expected frequency for depth = 30.00000
[ INFO] [1644916161.659102649]: Expected frequency for infra1 = 30.00000
[ INFO] [1644916161.770629463]: Expected frequency for infra2 = 30.00000
[ INFO] [1644916161.889283938]: Expected frequency for color = 30.00000
[ INFO] [1644916162.009989561]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1644916162.126682265]: Start publisher IMU
[ INFO] [1644916162.138278452]: setupStreams...
[ WARN] [1644916162.663633069]: 
[ WARN] [1644916162.663852168]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

**ROSGRAPH**
[image]

Please help regarding this issue!!!

MartyG-RealSense, 2022-02-15:
Hi @bhaskar-glitch  As you have enabled all stream types simultaneously and are applying the processing burdens of alignment and pointcloud on top of that but are on a limited-bandwidth USB 2.1 connection instead of USB 3.2, let's first try a simple custom roslaunch with depth and IMU only to see whether it makes a difference.

**roslaunch realsense2_camera rs_camera.launch enable_color:=false enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation depth_width:=640 depth_height:=480 depth_fps:=30**

bhaskar-glitch, 2022-02-15:
> roslaunch realsense2_camera rs_camera.launch enable_color:=false enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation depth_width:=640 depth_height:=480 depth_fps:=30

Thanks for the reply Marty!!
As per your given command i tried doing this but nothing happened..

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
[... 98 lines cut ...]
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/gyro/imu_info
/camera/imu
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

bhaskar-glitch, 2022-02-15:
> > roslaunch realsense2_camera rs_camera.launch enable_color:=false enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation depth_width:=640 depth_height:=480 depth_fps:=30
> 
> Thanks for the reply Marty!! As per your given command i tried doing this but nothing happened..and as u said connect it to usb 3.0+ port so jetson xavier have usb port of 3.1 already nd realsense is connected on that port only...i dont know why i'm getting that warning saying connected to usb 2.0 
> # SUMMARY
> PARAMETERS
> 
> * /camera/realsense2_camera/accel_fps: -1
> * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
> * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
> * /camera/realsense2_camera/align_depth: False
> * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/allow_no_texture_points: False
> * /camera/realsense2_camera/base_frame_id: camera_link
> * /camera/realsense2_camera/calib_odom_file:
> * /camera/realsense2_camera/clip_distance: -2.0
> * /camera/realsense2_camera/color_fps: -1
> * /camera/realsense2_camera/color_frame_id: camera_color_frame
> * /camera/realsense2_camera/color_height: -1
> * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
> * /camera/realsense2_camera/color_width: -1
> * /camera/realsense2_camera/confidence_fps: -1
> * /camera/realsense2_camera/confidence_height: -1
> * /camera/realsense2_camera/confidence_width: -1
> * /camera/realsense2_camera/depth_fps: 30
> * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
> * /camera/realsense2_camera/depth_height: 480
> * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
> * /camera/realsense2_camera/depth_width: 640
> * /camera/realsense2_camera/device_type:
> * /camera/realsense2_camera/enable_accel: True
> * /camera/realsense2_camera/enable_color: False
> * /camera/realsense2_camera/enable_confidence: True
> * /camera/realsense2_camera/enable_depth: True
> * /camera/realsense2_camera/enable_fisheye1: False
> * /camera/realsense2_camera/enable_fisheye2: False
> * /camera/realsense2_camera/enable_fisheye: False
> * /camera/realsense2_camera/enable_gyro: True
> * /camera/realsense2_camera/enable_infra1: False
> * /camera/realsense2_camera/enable_infra2: False
> * /camera/realsense2_camera/enable_infra: False
> * /camera/realsense2_camera/enable_pointcloud: False
> * /camera/realsense2_camera/enable_pose: False
> * /camera/realsense2_camera/enable_sync: False
> * /camera/realsense2_camera/filters:
> * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
> * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
> * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
> * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
> * /camera/realsense2_camera/fisheye_fps: -1
> * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
> * /camera/realsense2_camera/fisheye_height: -1
> * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
> * /camera/realsense2_camera/fisheye_width: -1
> * /camera/realsense2_camera/gyro_fps: -1
> * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
> * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
> * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
> * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
> * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
> * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
> * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
> * /camera/realsense2_camera/infra_fps: 30
> * /camera/realsense2_camera/infra_height: 480
> * /camera/realsense2_camera/infra_rgb: False
> * /camera/realsense2_camera/infra_width: 848
[... 14 lines cut ...]
> * /camera/realsense2_camera/stereo_module/exposure/1: 7500
> * /camera/realsense2_camera/stereo_module/exposure/2: 1
> * /camera/realsense2_camera/stereo_module/gain/1: 16
> * /camera/realsense2_camera/stereo_module/gain/2: 16
> * /camera/realsense2_camera/tf_publish_rate: 0.0
> * /camera/realsense2_camera/topic_odom_in: odom_in
> * /camera/realsense2_camera/unite_imu_method: linear_interpolation
> * /camera/realsense2_camera/usb_port_id:
> * /camera/realsense2_camera/wait_for_device_timeout: -1.0
> * /rosdistro: melodic
> * /rosversion: 1.14.12
> 
> NODES /camera/ realsense2_camera (nodelet/nodelet) realsense2_camera_manager (nodelet/nodelet)
> 
> auto-starting new master process[master]: started with pid [18088] ROS_MASTER_URI=http://localhost:11311/
> 
> setting /run_id to f4933fda-8e6d-11ec-bb12-706655b11773 process[rosout-1]: started with pid [18101] started core service [/rosout] process[camera/realsense2_camera_manager-2]: started with pid [18118] process[camera/realsense2_camera-3]: started with pid [18119] [ INFO] [1644936346.892143928]: Initializing nodelet with 6 worker threads. [ INFO] [1644936347.026844535]: RealSense ROS v2.3.2 [ INFO] [1644 [... cut ...] /metadata /camera/gyro/imu_info /camera/imu /camera/motion_module/parameter_descriptions /camera/motion_module/parameter_updates /camera/realsense2_camera_manager/bond /camera/rgb_camera/parameter_descriptions /camera/rgb_camera/parameter_updates /camera/stereo_module/auto_exposure_roi/parameter_descriptions /camera/stereo_module/auto_exposure_roi/parameter_updates /camera/stereo_module/parameter_descriptions /camera/stereo_module/parameter_updates /diagnostics /rosout /rosout_agg /tf /tf_static

bhaskar-glitch, 2022-02-15:
j

> roslaunch realsense2_camera rs_camera.launch enable_color:=false enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation depth_width:=640 depth_height:=480 depth_fps:=30
>
>
> Thanks [...]
ROSTOPIC
========
/camera/accel/imu_info
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/gyro/imu_info
/camera/imu
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

MartyG-RealSense, 2022-02-15:
I see that in your original launch log at the top of this case, **initial_reset** has already been set to True to reset the camera at launch and yet the USB was detected as 2.1, and so adding **initial_reset:=true** to the roslaunch instruction is not likely to make a difference to the incorrectly identified USB status.

As align and pointcloud were not used in my custom roslaunch instruction, it is also likely that your IMU issue is not related to the known problems that can occur on Jetson boards specifically in RealSense ROS when align or pointcloud is enabled.

Is there any change if you change **linear_interpolation** to **copy** as using the Copy setting with unite_imu_method can result in greater stability in IMU data.

Can you also confirm that the 'imu' topic is publishing data by using the command **rostopic echo /camera/imu**

MartyG-RealSense, 2022-02-15:
Is the warning **control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11** generating continuously in the terminal after launch has completed when the message 'RealSense Node is Up!' appears in the log?

bhaskar-glitch, 2022-02-17:
> Is the warning **control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11** generating continuously in the terminal after launch has completed when the message 'RealSense Node is Up!' appears in the log?
> 
> If this warning is generating continuously multiple times per second then it indicates a serious communication problem with the device is occurring.

yes yes this occurs continously...!! even though when my imu was working then also this warning was being printed after launch continously

MartyG-RealSense, 2022-02-17:
Are you able to run the **RealSense Viewer** tool on your Xavier and confirm whether '3.2' is next to the camera name at the top of the Viewer's options side panel, or whether it says '2.1'

If it says 2.1 as the USB connection type in the RealSense Viewer: may I ask if you are using the official 1 meter USB cable supplied with the camera or another cable that you chose yourself, please?  If you are using your own cable and it is a USB 2 cable instead of a USB 3 one then the camera would be detected as USB 2.1 even if plugged into a USB 3 port on your Xavier.  This is because USB 2 cables lack a couple of extra wires that USB 3 cables have that enable a device to be used in USB 3 mode.

bhaskar-glitch, 2022-02-17:
> Are you able to run the **RealSense Viewer** tool on your Xavier and confirm whether '3.2' is next to the camera name at the top of the Viewer's options side panel, or whether it says '2.1'
>
> If it [...]

Everything is perfectly working on Realsense viewer on xavier. nd yes i've confirmed that realsense is plugged in usb 3.2..and also i'm using that cable which comes with realsense itself

MartyG-RealSense, 2022-02-17:
Which installation method did you use to install the RealSense ROS wrapper on your Xavier, please?

bhaskar-glitch, 2022-02-18:
> Which installation method did you use to install the RealSense ROS wrapper on your Xavier, please?

I followed the same steps which are given in the repo for installation in ros. btw i have tried both the 2.3.2 version(earlier) and 4.0.1(latest) of Ros wrapper of realsense but the issue is the same i.e. imu is not working in both.

MartyG-RealSense, 2022-02-18:
2.3.2 is for ROS 1 (Kinetic, Melodic, Noetic) whilst the 4.0 versions are for ROS2 (Dashing, Eloquent, Foxy, Galactic, Rolling).   The launch log in your first message at the start of this case says that you were using Melodic (ROS 1), so should only be using wrapper **2.3.2** if you do not have ROS2 on your computer.

Which method in the ROS wrapper did you build with please - from packages with **Method 1** or from source code with **Method 2**

bhaskar-glitch, 2022-02-18:
> > 2.3.2 is for ROS 1 (Kinetic, Melodic, Noetic) whilst the 4.0 versions are for ROS2 (Dashing, Eloquent, Foxy, Galactic, Rolling). The launch log in your first message at the start of this case says that you were using Melodic (ROS 1), so should only be using wrapper **2.3.2** if you do not have ROS2 on your computer.
> > Which method in the ROS wrapper did you build with please - from packages with **Method 1** or from source code with **Method 2**
> 
> Ohkk okk got it..Btw im using method 2 the source code one for building...also if i want to install 2.3.2 then which version of librealsense should i use??like sometime i use to get **find_if** library while building realsense ros wrappper..what is the solution of that??

MartyG-RealSense, 2022-02-18:
If you are building librealsense and the ROS wrapper separately (which you are doing if you are building from source with Method 2) then you should install librealsense 2.50.0 first.  Once librealsense is successfully installed then you should install the RealSense ROS wrapper **2.3.2** from source code secondly after that.

Wrapper 2.3.2 is the correct version to match with librealsense 2.50.0.  It is important not to install the wrapper first and librealsense second.

Because 2.3.2 is currently the latest version of the ROS 1 wrapper, performng the 'git clone' described in the instructions should install 2.3.2.

MartyG-RealSense, 2022-02-18:
If you are able to use the RealSense Viewer without problems then it sounds as though the JetPack version that you are using is one that is compatible with librealsense, but it is worth checking anyway.  Which JetPack version does your Jetson have?

The next logical step would be to try to solve the problem of the camera being detected in the Xavier's USB port as 2.1 when it is in a USB 3 port and correctly detected as 3.2 in the RealSense Viewer.

bhaskar-glitch, 2022-02-18:
> If you are able to use the RealSense Viewer without problems then it sounds as though the JetPack version that you are using is one that is compatible with librealsense, but it is worth checking [...]

I'm using jetpack 4.6... and actually that USB error was my fault i was using another cable which was not 3.2 so now I changed that USB to 3.2 and that error is no more..but now I'm also getting a new error **[ WARN] [1645198591.852460906]: Hardware Notification: Right MIPI error,1.6452e+12, Error, Hardware Error**..please help regarding this...!!
See around 3-4 weeks earlier everything was working fine with the same jetpack also the same version librealsense(2.50.0),realsense ros(2.3.2)...but recently I re-flashed jetson with 4.6 jetpack and then tried the same thing I was doing earlier...but imu isn't working now!!

MartyG-RealSense, 2022-02-18:
May I confirm that the IMU works normally when using the Motion Module of the RealSense Viewer and it is only in ROS that you are experiencing problems with it.

bhaskar-glitch, 2022-02-18:
> May I confirm that the IMU works normally when using the Motion Module of the RealSense Viewer and it is only in ROS that you are experiencing problems with it.

yes I'm only experiencing this problem with ros!! everything is working fine with realsense viewer

MartyG-RealSense, 2022-02-19:
I notice that your roslaunch doesn't have the 'insert' messages - like the example below - that a typical launch log in the terminal has just before 'RealSense Mode is Up!' appears as the launch completes.  This is likely to be significant to your problem of missing messages.

[ INFO] [2021-12-13 23:40:08]: Start publisher IMU
[ INFO] [2021-12-13 23:40:08]: setupStreams...
[ INFO] [2021-12-13 23:40:09]: insert Gyro to Motion Module
[ INFO] [2021-12-13 23:40:09]: insert Accel to Motion Module

In the first log that you provided in your case, this is the point in the log where the **resource temporarily unavailable** messages started generating instead.

[image]

As you have stated that the camera works perfectly in the RealSense Viewer, there does not seem to be a problem with the camera hardware or its firmware or drivers.  

You also now have the correct USB 3 cable.  You reported that after changing the cable so that the ROS launch correctly identified the camera connection as USB 3.2, you now had a **Right MIPI error**.  This error occurs rarely in RealSense cases.  My understanding is that this error occurs if data is not being reliably received, which does seem to be your main problem in this case.

The point at which your previously working ROS setup broke was, you said, in https://github.com/IntelRealSense/realsense-ros/issues/2254#issuecomment-1044805461 when you re-flashed JetPack 4.6.  You had been using JetPack 4.6 without problems before the re-flash.  You could try using the Nvidia SDK Manager to flash JetPack 4.6 again in case there was a problem with the previous flashing.

https://developer.nvidia.com/embedded/downloads

bhaskar-glitch, 2022-02-19:
> I notice that your roslaunch doesn't have the 'insert' messages - like the example below - that a typical launch log in the terminal has just before 'RealSense Mode is Up!' appears as the launch completes. This is likely to be significant to your problem of missing messages.
> 
> [ INFO] [2021-12-13 23:40:08]: Start publisher IMU [ INFO] [2021-12-13 23:40:08]: setupStreams... [ INFO] [2021-12-13 23:40:09]: insert Gyro to Motion Module [ INFO] [2021-12-13 23:40:09]: insert Accel to Motion Module
> 
> In the first log that you provided in your case, this is the point in the log where the **resource temporarily unavailable** messages started generating instead.
> 
> [image]
> 
> As you have stated that the camera works perfectly in the RealSense Viewer, there does not seem to be a problem with the camera hardware or its firmware or drivers.
> 
> You also now have the correct USB 3 cable. You reported that after changing the cable so that the ROS launch correctly identified the camera connection as USB 3.2, you now had a **Right MIPI error**. This error occurs rarely in RealSense cases. My understanding is that this error occurs if data is not being reliably received, which does seem to be your main problem in this case.
> 
> The point at which your previously working ROS setup broke was, you said, in [#2254 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2254#issuecomment-1044805461) when you re-flashed JetPack 4.6. You had been using JetPack 4.6 without problems before the re-flash, and support for 4.6 (L4T 32.6.1) was added in librealsense 2.50.0.
> 
> You could try using the Nvidia SDK Manager to flash JetPack 4.6 again in case there was a problem with the previous flashing.
> 
> https://developer.nvidia.com/embedded/downloads

I have tried these stuff my friend!! i have tried flashing jetson again with SDK manager too also I tried reinstalling realsense ros ,librealsense many times but it's not working..!! :(

MartyG-RealSense, 2022-02-19:
Sometimes when you have tried everything that you can think of, the solution that succeeds (with both librealsense and the ROS wrapper) is just to wipe the storage completely clean and install everything again from the beginning, including the Ubuntu OS.  Doing this can clear hidden conflicts / breakages that are preventing the camera from working correctly.  Have you tried a complete wipe and reinstall yet, please?

bhaskar-glitch, 2022-02-19:
> Sometimes when you have tried everything that you can think of, the solution that succeeds (with both librealsense and ROS wrapper problems) is just to wipe the storage completely clean and install everything again from the beginning, including the Ubuntu OS.  Doing this can clear hidden conflicts / breakages that are preventing the camera from working correctly.  Have you tried a complete wipe and reinstall yet, please?

Yess!! I've tried this all nothing helped
:(

MartyG-RealSense, 2022-02-19:
You have told me that you install the ROS wrapper from source code.  Which instructions do you use to install **librealsense** on your Jetson Xavier?

bhaskar-glitch, 2022-02-19:
> You have told me that you install the ROS wrapper from source code.  Which instructions do you use to install **librealsense** on your Jetson Xavier?

[image]

I followed this!!

MartyG-RealSense, 2022-02-19:
It looks like you are building from source using a Jetson hacks script - **buildLibrealsense.sh** - but the instructions may be out of date.  A more modern Jetson hacks instruction page is at the link below.

https://github.com/jetsonhacks/installRealSenseSDK#installlibrealsensesh

The instructions that you were using are a bit confusing in the way that they are written.  JetsonHacks give you the choice of building from packages with the script **installLibrealsense.sh** or building from source code with the **buildLibrealsense.sh** script.  

You should only use **one** of these scripts, depending on whether you want to build from packages (which JetsonHacks recommends) or building from source.  The instructions that you were using could give the impression that **both** scripts have to be run though.  This could create problems.  If you build librealsense from both packages and source on the same computer then you can get a conflict in the **udev rules** that control camera device handling.

closed 2022-03-03 by MartyG-RealSense