realsenseai/realsense-ros #2260: Implementation issue of the opensource_tracking.launch

SahanGura, 2022-02-22:
In the opensource_tracking.launch file, filtered odometry given by the ukf of robot localization package has not fed into the rtabmap node. Which means rtabmap node subscribes odometry from rgbd_odometry node thorugh rtabmap/odom topic. So filtered odometry from ukf is not used in rtabmap not if I am not mistaken?

Correct implementation should be something like following node graph.
[image]

MartyG-RealSense, 2022-02-22:
Hi @SahanGura  I must emphasize in advance that this particular topic is outside of my knowledge, but I will do my best to provide some useful research leads.  Thanks in advance for your patience and understanding!

The link below features a case involving RealSense D435i, RTABMAP, robot_localization and opensource_tracking.launch that has graphs to compare to your own. 

https://answers.ros.org/question/317419/realsense-d435-rtabmap-pixhawk-imu-robot_localization/

Easier to read versions of the link's graphs that can be viewed in full size when clicked on are below.

[image]

[image]

SahanGura, 2022-02-23:
Thank you for the links you provided. According to the conclusion of the [matlabbe](https://answers.ros.org/users/20914/matlabbe/), my implementation is correct. Which means the open_source_tracking.launch file which has provided for D435i cameras is not correct. Because the filtered odometry, after sensor fusion, does not take into rtabmap node to create the map. In that case, having an IMU in realsense D435i is not useful. 
However the current issue I am dealing with is, when I am running all these nodes in raspberry pi 4 with ubuntu 18.04 and ros melodic, the generated map is distorted. I doubt this might be an issue with the performance of the rpi. Because when I am running all these nodes in the computer, generated map is not distorted.
Do you have any links related to similar kind of issue?

MartyG-RealSense, 2022-03-01:
I apologize that you did not receive a further response to your follow-up question.  Do you still require links to similar issues, please?

closed 2022-03-07 by MartyG-RealSense