realsenseai/realsense-ros #2262: Camera position

MelanieFo94, 2022-02-24:
Hi, I would like to know how and which file I have to adapted to indicate my D455 or D435i is 0.7 meters from the ground ?

I would like to do a map with the following command : `roslaunch realsense2_camera opensource_tracking.launch` 
I tested it several time but as I didn't obtain a very good map, I need to put the position of the camera but I don't know where I can do it.

MartyG-RealSense, 2022-02-24:
Hi @MelanieFo94  The RealSense ROS wrapper has a Python script called **show_center_depth.py** that demonstrates subscribing to depth image and depth camera info topics and extracting depth and 3D position in the camera's coordinates.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py

The link below describes how to run the 'show_center_depth.py' script.

https://github.com/IntelRealSense/realsense-ros/commit/62d36696999b2fbe100e6f92158dc5746a8d5efa#commitcomment-45621558

**********

If the camera will be facing downwards towards the ground from above then another approach could be to set the ROS wrapper's **clip_distance** parameter to '0.7' (meters).  This would set the camera's maximum observable distance to 0.7 meters, so that if the camera is raised up higher from the ground then the ground no longer appears on the depth image because it is further away from the camera than 0.7 m.

MelanieFo94, 2022-02-25:
Thank for your answer but it's not the answer I wanted. 

I'm going to try to be clearer. When I launch RVIZ with the following command : `roslaunch realsense2_camera opensource_tracking.launch` and check the map and camera link frames, I would to see in the camera_link frames the value "0.7" for z position.
The map frame should be at 0, at the origin of the RVIZ map. 

And as It's impossible to modify it on RVIZ, how I should proceed ?

MartyG-RealSense, 2022-02-25:
I am not absolutely certain what you are describing, but if you want to change the origin position then a **static transform** may enable this, as discussed in https://github.com/IntelRealSense/realsense-ros/issues/1837

MartyG-RealSense, 2022-03-03:
Hi @MelanieFo94  Do you require further assistance with this case, please?  Thanks!

MelanieFo94, 2022-03-03:
Hi @MartyG-RealSense , your second response helped me to change the camera position on RVIZ. Thank you !

MartyG-RealSense, 2022-03-03:
That's great to hear, @MelanieFo94  :)  If your problem is solved, please feel free to close the case using the **Close Issue** button beneath the comment writing box if you do not require further assistance.  Thanks!

closed 2022-03-03 by MelanieFo94