realsenseai/realsense-ros #2263: CMake Error after switching branches

CWAndersn, 2022-02-24:
I previously successfully built and ran this package using the realsense distribution(method 2 of the README) with no issues. However, I switched to the occupancy_mapping branch, and when I made the switch back I was unable to make the files using catkin. I receive the error below.

```
CMake Error at /home/acl/code/vision_ws/devel/share/ddynamic_reconfigure/cmake/ddynamic_reconfigureConfig.cmake:113 (message):
  Project 'ddynamic_reconfigure' specifies
  '/home/acl/code/vision_ws/src/realsense-ros/ddynamic_reconfigure/include'
  as an include dir, which is not found.  It does neither exist as an
  absolute directory nor in
  '/home/acl/code/vision_ws/src/realsense-ros/ddynamic_reconfigure//home/acl/code/vision_ws/src/realsense-ros/ddynamic_reconfigure/include'.
  Report the problem to the maintainer 'Noam Dori <theNODO55@gmail.com>' and
  request to fix the problem.
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  realsense-ros/realsense2_camera/CMakeLists.txt:11 (find_package)
```

I tried completely removing the package and running catkin_make, which succeeded. However, when I then follow the instructions again from the beginning, I run into the same error. I'm not sure what's going on here, but it seems that running catkin_make under a different branch has messed up the configuration of my workspace. Any ideas on how to fix this error?

MartyG-RealSense, 2022-02-25:
Hi @CWAndersn  A past case of this error can be found at https://github.com/IntelRealSense/realsense-ros/issues/812#issuecomment-502644678 and the advice provided by a RealSense team member was that ddynamic_reconfigure had been cloned from the wrong source and should be cloned from the link below instead:

https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel

CWAndersn, 2022-02-25:
Thank you @MartyG-RealSense. After cloning the package into my workspace catkin_make runs without issue. I was initially concerned that I was running ros-noetic, yet the ddynamic_reconfigure branch you linked specifies ros kinetic. However, this did not end up causing any issue. 

I believe my original mistake was attempting to clone ddynamic_reconfigure using apt with the following command.

```
apt-get install ros-noetic-ddynamic-reconfigure
```

As you point out, this is the incorrect source and the one from pal-robotics works just fine. Thank you!

closed 2022-02-25 by CWAndersn

MartyG-RealSense, 2022-02-25:
It's great to hear that you were successful, @CWAndersn - thanks very much for the update   :)