realsenseai/realsense-ros #2264: realsense d435i roslaunch realsense2_camera rs_camera.launch error

hyeonminseok, 2022-02-27:
i have a relsense d435i camera. i want to work it Although realsense-viewer worked well but roslaunch realsense2_camera rs_camera.launch had error it is that..

seok@seok-ThinkPad-E15-Gen-3:~/catkin_ws/src$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/seok/.ros/log/45e29308-97a2-11ec-b860-84a9386b66ed/roslaunch-seok-ThinkPad-E15-Gen-3-16646.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://seok-ThinkPad-E15-Gen-3:42815/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 75 lines cut ...]
[ INFO] [1645948377.091700755]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1645948377.091752192]: num_filters: 0
[ INFO] [1645948377.091769919]: Setting Dynamic reconfig parameters.
[ INFO] [1645948380.181096910]: Done Setting Dynamic reconfig parameters.
[ INFO] [1645948380.182297166]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1645948380.184352599]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1645948380.184483395]: setupPublishers...
[ INFO] [1645948380.187876196]: Expected frequency for depth = 30.00000
[ INFO] [1645948380.222650971]: Expected frequency for color = 30.00000
[ INFO] [1645948380.243257179]: setupStreams...
 27/02 16:53:00,435 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1645948380.486490117]: SELECTED BASE:Depth, 0
[ INFO] [1645948380.508800009]: RealSense Node Is Up!
[ WARN] [1645948380.555121079]: 
 27/02 16:53:00,556 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:00,606 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:00,658 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:01,011 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:01,164 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:01,214 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:01,366 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:01,417 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 16:53:01,569 WARNING [140571228698368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

help me please i don't know what i should do

MartyG-RealSense, 2022-02-27:
Hi @hyeonminseok   The **control_transfer returned** warnings in the log can be safely ignored if they are only appearing at the start of launch or appearing about once a minute instead of generating continuously.  If these warnings are generated continuously though then this may indicate a serious communication problem with the device.

Sometimes the topics may still be able to be correctly published in the **RViz** interface even if these messages appear.  If the camera data is being displayed correctly in RViz with no sign of degraded performance then again the warnings can be ignored.

Could you test please whether adding **initial_reset:=true** to your roslaunchh instruction to reset the camera at launch makes a positive difference, please?  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

hyeonminseok, 2022-02-28:
thank to your rapid reply
this is resolved

MartyG-RealSense, 2022-02-28:
You are very welcome, @hyeonminseok - I'm pleased to hear that the problem was resolved.  I will therefore close this case.  Thanks very much!

closed 2022-02-28 by MartyG-RealSense

jrlearncode05, 2024-07-02:
Hello, I am also having this problem. I tried the "roslaunch realsense2_camera rs_camera.launch initial_reset:=true" command and it still results i the error "RLException: [rs_camera.launch] is neither a launch file in package [realsense2_camera] nor is [realsense2_camera] a launch file name
The traceback for the exception was written to the log file". May you tell me how to resolve this issue please?

MartyG-RealSense, 2024-07-03:
Hi @jrlearncode05  Have you installed the RealSense SDK (librealsense2) and then the RealSense ROS1 wrapper, please?

The link below has a previous example of this error.

https://answers.ros.org/question/304600/realsense-ros-wont-start-with-roslaunch-realsense2_camera-rs_cameralaunch/

rutledgejaniah, 2024-07-03:
Hi, yes I have installed both the librealsense2 and the RealSense ROS1 wrapper. I am also having trouble with "pyrealsense2" I tried pip installing it and building it from source.

MartyG-RealSense, 2024-07-03:
@rutledgejaniah  Which method did you use to install librealsense2 and the RealSense ROS wrapper?

For example, if you installed librealsense2 first and then installed the wrapper with 'Method 1' ( the `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` command) then this command would install both librealsense2 and the wrapper at the same time.  If librealsense2 had already been installed then you could end up with two conflicting installations of librealsense2 on the same computer.