realsenseai/realsense-ros #2265: Jetson nano wrong measures

sergioar3143, 2022-03-01:
Hello I am using Jetson nano and I am trying to segment the color of a ball, because I want to estimate the trajectory. The problem is that I have wrong measures. I have calibrated the camera many times and I still having the same problem. What can I do?

MartyG-RealSense, 2022-03-01:
Hi @sergioar3143   The RealSense-compatible ball color segmentation project for ROS at the link below may be a helpful reference.

https://github.com/ctu-mrs/object_detect

If you used the **On-Chip** calibration tool then that will calibrate depth.  The RealSense SDK's **Dynamic Calibration** tool can perform a robust calibration of both depth and RGB.  It can be installed on Linux by following the instructions on page 14 onwards of the tool's user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

sergioar3143, 2022-03-02:
Thank you so much. I have one more question.
I calibrated the camera on windows  with the Dynamic calibrator at 0 grades like the first image, but I used the camera at 30 grades on the jetson nano like the second one, this could be a problem with the measures, couldn't it

[image]
[image]

MartyG-RealSense, 2022-03-02:
You should be fine.  30 degrees is the maximum that I would recommend tilting the camera.  You can go beyond 30 degrees but would risk introducing problems into the image.

sergioar3143, 2022-03-02:
Okay, thank you so much.

MartyG-RealSense, 2022-03-08:
Hi @sergioar3143  Do you require further assistance with this case, please?  Thanks!

sergioar3143, 2022-03-09:
No, thank you

closed 2022-03-09 by sergioar3143