realsenseai/realsense-ros #2268: Is there a parameter to remove the acceleration of gravity from the accelerometer?

j-corvo, 2022-03-02:
Hello, 

I'm using the IMU of both d455 and d435i realsense cameras with ROS, but I want to remove the gravity's acceleration. Is there any way I can do that?

Thanks for your help!

MartyG-RealSense, 2022-03-02:
Hi @j-corvo  If you only wanted the gyro information then you could add **enable_gyro:=true** to your roslaunch instruction or set enable_gyro to True inside the launch file, but not enable accel or define the **unite_imu** parameter.  

If unite_imu is not set to 'linear_interpolation' or 'copy' then the **imu** topic will not be published.  Instead, the gyro should be published on its own as the topics **/camera/gyro/imu_info** and **/camera/gyro/sample**

j-corvo, 2022-03-03:
Ok, so I guess there's no option in realsense-ros to remove the default gravity's acceleration of the accelerometer, right?

MartyG-RealSense, 2022-03-03:
That is correct, there is not a control in the RealSense ROS wrapper for adjusting the accelerometer gravity.

Some RealSense ROS users with a D435i have made use of **VINS-FUSION** though, which has a parameter called **g_norm** relating to gravity magnitude (set to '9.805' by default).

https://github.com/introlab/rtabmap_ros/issues/517
https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/6
https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/36

[image]

Also, the librealsense SDK's **rs-motion** C++ example demonstrates the principle of increasing an 'alpha' value to adjust calculations to give more weight to the gyro instead of the sensitive acceleremoter.  The cost of doing so though is possible drift.

https://github.com/IntelRealSense/librealsense/blob/master/examples/motion/rs-motion.cpp#L117-L118

MartyG-RealSense, 2022-03-09:
Hi @j-corvo  Do you require further assistance with this case, please?  Thanks!

j-corvo, 2022-03-09:
Hi @MartyG-RealSense sorry for the delay in my response. No, I don't, it was very helpful!

Thank you for your availability.

closed 2022-03-09 by j-corvo

MartyG-RealSense, 2022-03-09:
You are very welcome, @j-corvo - I'm pleased that I could help.  Thanks for the update!