realsenseai/realsense-ros #2269: Enable decimation filter even when "align_depth = true "

mszuyx, 2022-03-03:
Hi intel RS team,

This is a revisit / feature suggestion following the issue: #1834 

Basically, I would like the point cloud / depth image's resolution to be different than the RGB image (as robot navigation usually don't require high point cloud resolution, but a clear RGB image is always welcome). Meanwhile, I need the depth image and RGB image to have similar FOV. (I am using D455, the FOV difference exist but relatively small)

I have been setting the **align_depth = false** and use the **decimation filter** to achieve this. The  **align_depth** option allows me to decouple the resolution setting between RGB image and depth image. And **decimation filter** allows me to further reduce the resolution.

However, as I need to preform mask mapping between RGB image and depth image, I need to set **align_depth = true**. But doing this effectively overwrite the depth image resolution setting and  decimation filter (no matter what you set the resolution to / or resolution the terminal reports, the cloud resolution is ALWAYS the same as the RGB image).

I wonder:

1. Is there a way to let **decimation filter** and **align_depth** compatible with each other?
2. If not, can you provide some inside on how to post-process it in a custom node? (for now I am just cropping the depth image, is it the equivalent of **align_depth**? Or the **align_depth** is more complicated than that?)

Thanks!
Yu

MartyG-RealSense, 2022-03-03:
Hi @mszuyx  I believe that this question is a follow-up to your align / decimation question at https://github.com/IntelRealSense/realsense-ros/issues/1924 in 2021.  Doronhi suggested a 'bypass' method at https://github.com/IntelRealSense/realsense-ros/issues/1924#issuecomment-864510866 for enabling align_depth and decimation to work together, which I believe you tried implementing but it did not solve your particular problem.  Is that correct, please?

At the end of that case, Doronhi also suggested at https://github.com/IntelRealSense/realsense-ros/issues/1924#issuecomment-904354287 that modifying the **frame_callback()** function may be a solution for what you were requesting.

Instead of using decimation for the purposes of reducing resolution, defining a custom stream configuration with the roslaunch instruction and setting your own choice of stream resolutions may be an easier alternative.  Below is an example of doing this for the depth and color streams.

`roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30`

In a custom stream configuration, three factors must be specified for each stream type defined - width, height and FPS.  If these 3 factors are not provided then the RealSense ROS wrapper deems the configuration invalid and automatically uses the default configuration of the particular RealSense camera model instead.

In answer to your two questions:

1.  I don't have any advice that was not already covered in https://github.com/IntelRealSense/realsense-ros/issues/1924

2.  I could not find any resources regarding accessing librealsense's post-processing functions from a ROS node script instead of using the wrapper's filters.

mszuyx, 2022-03-03:
@MartyG-RealSense You are absolutely correct! I forgot we have went that route already! (great catch! )
I think because I recently git from the repo again and that undone my previous effort, causing me to think the method didn't work.

So the take away is, since Doronhi's 'bypass' method does allow align_depth and different resolution settings to co-exist, lets use resolution setting (instead of decimation filter) to reduce resolution? 

I will try it today if that solve the problem. And by "post-processing", I was thinking about something like [depth_image_proc/register](http://wiki.ros.org/depth_image_proc?distro=noetic). If I can't resolve the issue with launch file and realsense_node, I will try that and let you know how it goes!

Thanks again!
Yu

MartyG-RealSense, 2022-03-04:
On D455, 256x144 depth is supported at **90** FPS.  The processing burden of streaming 90 FPS at that minimum resolution should not be very high though.

mszuyx, 2022-03-08:
@MartyG-RealSense Hi, 

I did some tests today to compare the **256x144@90FPS** vs. **424x240@30FPS (with decimation@2)** configs.
At first I was trying to compare the CPU usage in each case. But I found out that the depth image and point cloud creating with 256x144 option seems cut-off:
[image]

And here is the same camera pointing at the same position (scene) using the 424x240 option:
[image]

Could you help me understand this?

Here is my launch file that enables 424x240 setup if you are interested:
[rs_d455_camera_pco.zip](https://github.com/IntelRealSense/realsense-ros/files/8209571/rs_d455_camera_pco.zip)
To enable 256x144 I just change the depth resolution accordingly and FPS to 90 and set decimation to 1 at the end.

I also notice when the align_depth is enabled, ros-realsense will publish a topic that describes the extrinsic matrix between RGB and depth frame. I am thinking maybe I can do the registering/alignment myself with the 424x240 option using reasonable resources. But meanwhile, if realsense can allow me to use decimation with align_depth, that will be very useful!

Thanks

mszuyx, 2022-03-08:
@MartyG-RealSense Is it possible for you to provide the homography transformation between the rgb and depth frame? Knowing this transformation matrix, I assume the alignment can be done with _cv2.warpPerspective()_.

MartyG-RealSense, 2022-03-09:
Are you asking why 256x144 without decimation does not look the same as when 424x240 is decimated, please?  If so, I am not certain about the precise mechanics of the process.  A decimated image is different from setting the resolution with the direct selection of a supported RealSense resolution mode though, since the decimation process is taking a higher resolution and 'downsampling' the pixels, scaling the image down proportionally in both dimensions to preserve the aspect ratio.

In regard to comparing the transform between two coordinate systems in ROS, my understanding is that this can be done with **rosrun tf tf_echo**.  For example, if comparing the transform between gyro and depth:

**rosrun tf tf_echo camera_depth_optical_frame camera_gyro_optical_frame**

You can also retrieve the extrinsic transformations between two sensor types on a particular camera using the librealsense tool **rs-enumerate-devices** by launching the tool in calibration information mode:

`rs-enumerate-devices - c`

mszuyx, 2022-03-10:
@MartyG-RealSense Yes. Ideally a change in resolution setting shouldn't affect the FOV of the view. And the fact that this resolution setting only work with 90 FPS leads me to think this 256x144 setting is subscribing to a sub-system in the camera. (i.e., the IR projector/receiver. And not even using the stereo features. -- that explains the rigid constraint on FPS and the change in FOV)

For now I am trying to create a homography matrix based on camera models and tf transformation. I will keep you posted.

mszuyx, 2022-03-10:
@MartyG-RealSense Interesting!

Btw, I got the homography transform to work (actually the affine transform works just fine). I guess for future reference, if people want to create their own alignment solutions. They can check out these two links:
[geometric transformation](https://towardsdatascience.com/image-geometric-transformation-in-numpy-and-opencv-936f5cd1d315)
[register depth image](https://github.com/ros-perception/image_pipeline/blob/noetic/depth_image_proc/src/nodelets/register.cpp)

closed 2022-03-10 by mszuyx

MartyG-RealSense, 2022-03-10:
That's excellent news that you achieved a solution!  Thanks very much for the update and for sharing details of your method with the RealSense community  :)