realsenseai/realsense-ros #2271: aligned depth to colour values units

rakshith95, 2022-03-04:
Hello, 
I'm using the D435i, and I run the roslaunch command with the align_depth=true argument: `roslaunch realsense2_camera rs_camera.launch align_depth:=true`

Now, when I display the values published to the aligned_depth topic with `rostopic echo /camera/aligned_depth_to_color/image_raw`, I get the following: 

`...0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 204, 0, 205, 0, 205, 0, 205, 0, 205, 0, 205, 0, 205, 0, 205, 0, 205, 0, 62, 7, 57, 7, 57, 7, 51, 7, 45, 7, 45, 7, 40, 7, 34, 7, 34, 7, 29, 7, 29, 7, 24, 7, 18, 7, 18, 7, 13, 7, 7, 7, 7, 7, 2, 7, 2, 7, 253, 6, 248, 6, 248, 6, 242, 6, 237, 6, 237, 6, 232, 6, 232, 6, 227, 6, 227, 6, 227, 6, 227, 6, 222, 6, 222, 6, 222, 6, 227, 6, 227, 6, 227, 6, 227, 6, 227, 6, 227, 6, 227, 6, 222, 6, 222, 6, 222, 6, 222, 6, 222, 6, 217, 6, 217, 6, 217, 6, 217, 6, 217, 6, 217, 6, 212, 6, 212, 6, 212, 6, 207, 6]` and so on. 

How do I get the true depth values in 'm' from these? I've seen in some other repositories where they divide these values by 256, but I'm not sure. Please advise.

MartyG-RealSense, 2022-03-04:
Hi @rakshith95  https://github.com/IntelRealSense/realsense-ros/issues/714 should be helpful in answering your question.

rakshith95, 2022-03-04:
Thank you! I guess the values are in 'mm' then. Closing this issue

closed 2022-03-04 by rakshith95