realsenseai/realsense-ros #2276: rs_d400_and_t265.launch and rs_aligned_depth.launch fail

MTDzi, 2022-03-08:
I have a freshly flashed Jetson Nano 4GB (L4T 32.6.1, Jetpack 4.6) with the D435i and T265 cameras plugged in, with a freshly installed librealsense2 (v2.50.0), and I managed to catkin_build the package, but when I run
```
roslaunch realsense2_camera rs_d400_and_t265.launch
```
I get the following errors (I sent a ^C because otherwise my device stops responding):
```
... logging to /home/md/.ros/log/61893682-9ec5-11ec-95b2-94b86db8791e/roslaunch-nano-20934.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nano:34853/

SUMMARY
========

PARAMETERS
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/confidence_fps: 30
 * /d400/realsense2_camera/confidence_height: 480
 * /d400/realsense2_camera/confidence_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: False
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_confidence: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: False
 * /d400/realsense2_camera/enable_fisheye2: False
 * /d400/realsense2_camera/enable_fisheye: False
 * /d400/realsense2_camera/enable_gyro: False
 * /d400/realsense2_camera/enable_infra1: False
 * /d400/realsense2_camera/enable_infra2: False
 * /d400/realsense2_camera/enable_infra: False
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_pose: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
 * /d400/realsense2_camera/gyro_fps: 0
[... 477 lines cut ...]
 08/03 10:58:51,632 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,031 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,063 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,097 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,130 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,164 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,199 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,565 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,597 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,631 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,665 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:52,698 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
 08/03 10:58:53,066 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:53,097 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 08/03 10:58:53,132 ERROR [546232578432] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done 
```

Any ideas / suggestions as to what I should do?

RealSenseSupport, 2022-03-10:
@MTDzi Can you launch single camera(either D435i or T265) successfully? Did you install RealSense ROS wrapper by debian package or build from source code? Please note that if you install by debian package, librealsense2 is not built to use native v4l2 driver but the less stable RS-USB protocol. Could you please try to build from source code according to https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources ?

RealSenseSupport, 2022-03-17:
@MTDzi Any update from your side? Looking forward to your reply. Thanks!

MTDzi, 2022-03-17:
@RealSenseSupport  I'm terribly sorry, I missed the first notification.

I'll get back to you later today (in 12h or so). 

My librealsense2 comes from a debian package, of that I'm sure. And I think (although I'm not 100% certain)  I followed exactly the instructions you linked (https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources).

What I'm hesitant about is: whether I first tried to install the RealSense ROS wrapper by a debian package and when that didn't work, I built it with catkin (following the above instructions). If that was the case, are there any instructions on how to correctly uninstall the RealSense ROS wrapper before trying to build them? Or should this not be an issue?

MTDzi, 2022-03-21:
OK,
I tried running the `scripts/patch-realsense-ubuntu-L4T.sh` but since I'm using JetPack 4.6, that didn't end well.
I'm going to re-flash it with 4.5.1 (for which it should work out of the box), will keep you posted.

RealSenseSupport, 2022-03-22:
@MTDzi Please note that Jetpack 4.6/4.5.1 has included the kernel patch. So you don't need run "scripts/patch-realsense-ubuntu-L4T.sh".

If you installed RealSense ROS wrapper by Debian package, I would suggest you uninstall it by "sudo apt purge ros-$ROS_DISTRO-realsense2-camera" and then follow https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources to build from ROS wrapper source code. And also please make sure you've installed librealsense before that. If not installed yet, then please follow https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages to install librealsense first.

RealSenseSupport, 2022-04-07:
@MTDzi Could you please update? Thanks!

RealSenseSupport, 2022-04-21:
@MTDzi Any progress from your side? Please note that this ticket will be closed if we don't get update from you for another 7 days.

RealSenseSupport, 2022-05-05:
@MTDzi Sorry we don't hear from you for weeks. The ticket will be closed now. Please feel free to create another ticket if you still have other questions or issues. Thanks!

closed 2022-05-05 by RealSenseSupport