realsenseai/realsense-ros #2278: Not able to Catkin_make / Catkin_Build Realsense2_camera

varungoyal87, 2022-03-09:
Here are the steps which I have followed:

1. Installed librealsense v2.50 and get able to see the camera working fine in realsense-viewer.
2.  Later on , I git clone the realsense-ros (v4.0.2 compatile with librealsense v2.50) and tried 
     - to compile with catkin_make  , it failed .
     - to compile with catkin build, it skipped the realsense package and it succeed but not able to launch the rs_camera.launch file.
     - to compile with own ros packages, get the error "This workspace non-catkin packages in it and catkin can not build a non- 
        homogenous workspace without isolation"  . I observed "plain cmake " mentioned in front of realsense2_Camera _msgs, 
        realsense2_desription.

I used following url :
https://github.com/IntelRealSense/realsense-ros : 
- used method 2 (realsense distribution) followed 
step1 [Linux Debian Installation Guide] 
 (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)  - installing the packages 

step 2 ubuntu , getting an error in catkin_make clean .
Note: I tried to make separate catkin_ws and included with own ROS package as well, both time it failed
 Note: I tried this work around https://github.com/IntelRealSense/realsense-ros/issues/1432 but still not working

Framework : Virtual Box VM : ubuntu 18.04
                     ROS release : melodic

Can you please advise for this to make it work .

MartyG-RealSense, 2022-03-09:
Hi @varungoyal87  If you are using ROS **Melodic** then the 4.02 version of the RealSense ROS wrapper would not work for you, as it is for use with **ROS2** (Dashing, Eloquent, Foxy, Galactic and Rolling).  If you have librealsense SDK 2.50.0 and Melodic then the correct ROS wrapper version to match to the SDK is **2.3.2**

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

The easiest installation method for you if you want a package-based install may be to use ROS **Method 1**, which installs correctly matched librealsense and ROS wrapper versions together from packages.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

Method 1 for ROS should currently install librealsense 2.50.0 and ROS wrapper 2.3.2.

Please also bear in mind that Intel recommend using **VMWare Workstation Player** (Workstation Pro should work too) as a VM tool instead of VirtualBox, as VMWare provides emulation of the USB 3 controller.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

varungoyal87, 2022-03-10:
Thanks alot for the information and it fixed my issue.

closed 2022-03-10 by varungoyal87

MartyG-RealSense, 2022-03-10:
That's great news, @varungoyal87 - thanks very much for the update!