realsenseai/realsense-ros #2280: Unable to install on Windows 10

stevkr, 2022-03-10:
Hello everyone,

I would like to get a D435 running via ROS on a Windows 10 machine.
I have followed the instructions:
There are no problems with the installation of ROS noetic.
However, I have problems following Step 2: Install Intel RealSense ROS from Sources:
The commands 
```
mkdir c:\catkin_ws\src
cd c:\catkin_ws\src

git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
```
are working fine.
The first problem occurs in this line: 
`git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1` `
because the grep command is not recognised. I guess this is just a version indicator or something and not important for my runnability.
The much bigger problem arises when I run 
`catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release`
in. A larger list of errors appears there:

```
C:\catkin_ws>catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Rel
Base path: C:\catkin_ws
Source space: C:\catkin_ws\src
Build space: C:\catkin_ws\build
Devel space: C:\catkin_ws\devel
Install space: C:\catkin_ws\install
####
#### Running command: "cmake C:\catkin_ws\src -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Rel -DCATKIN_DEVEL_PREFIX=C:\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=C:\catkin_ws\install -G Ninja" in "C:\catkin_ws\build"
####
-- Using CATKIN_DEVEL_PREFIX: C:/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;c:\opt\ros\noetic\x64
-- This workspace overlays: c:/opt/ros/noetic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~  - realsense2_description
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Create Debug Build.
-- realsense2_camera: 3 messages, 1 services
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/catkin_ws/build
####
#### Running command: "ninja -j4 -l4" in "C:\catkin_ws\build"
####
[8/39] Building CXX object ddynamic_reconfigure\CMakeFiles...rver.dir\test\test_bool_dynamic_reconfigure_server.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[10/39] Building CXX object ddynamic_reconfigure\CMakeFile...re_server.dir\test\fake_dynamic_reconfigure_server.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[21/39] Building CXX object ddynamic_reconfigure\CMakeFiles\ddynamic_reconfigure.dir\src\ddynamic_reconfigure.cpp.obj
[... 46 lines cut ...]
C:\opt\ros\noetic\x64\include\cv_bridge/cv_bridge.h(396): warning C4267: 'return': conversion from 'size_t' to 'uint32_t', possible loss of data
C:\catkin_ws\src\realsense-ros\realsense2_camera\include\realsense2_camera/constants.h(98): warning C4305: 'initializing': truncation from 'double' to 'float'
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(122): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_width_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(123): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_row_step_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(126): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_height_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(129): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_width_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(130): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_row_step_type', possible loss of data
[38/39] Building CXX object realsense-ros\realsense2_camer...s\realsense2_camera.dir\src\realsense_node_factory.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++11'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
C:\opt\ros\noetic\x64\include\cv_bridge/cv_bridge.h(396): warning C4267: 'return': conversion from 'size_t' to 'uint32_t', possible loss of data
C:\catkin_ws\src\realsense-ros\realsense2_camera\include\realsense2_camera/constants.h(98): warning C4305: 'initializing': truncation from 'double' to 'float'
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(122): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_width_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(123): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_row_step_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(126): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_height_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(129): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_width_type', possible loss of data
C:\opt\ros\noetic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(130): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_row_step_type', possible loss of data
C:\catkin_ws\src\realsense-ros\realsense2_camera\src\realsense_node_factory.cpp(104): warning C4267: 'argument': conversion from 'size_t' to 'uint32_t', possible loss of data
ninja: build stopped: subcommand failed.
Invoking "ninja -j4 -l4" failed

```
Can anyone help me?

Many greetings

MartyG-RealSense, 2022-03-10:
Hi @stevkr  The problems that you experienced have been encountered by others in the past, such as the case at https://github.com/IntelRealSense/realsense-ros/issues/1797

You can get around the git checkout issue on Windows by specifying the exact version number instead of searching for it with a tag, as described in the above case at https://github.com/IntelRealSense/realsense-ros/issues/1797#issuecomment-817095558

Even after getting past this stage though, the **-DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release** can act as a block to progressing further.

If setting DCATKIN_ENABLE_TESTING to True instead of False does not work for you, my understanding is that the rosdep command below can be used instead.  Navigate to the catkin workspace folder with **cd ~/catkin_ws/src/** and then run the rosdep command from that folder:

**rosdep install --from-paths src --ignore-src -r -y**

stevkr, 2022-03-11:
Thanks for your answer @MartyG-RealSense 

Unfortunately, the commands do not work as desired.
After replacing 
`catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release`  
with 
`catkin_make -DCATKIN_ENABLE_TESTING=True -DCMAKE_BUILD_TYPE=Release` I get the message:

```
c:\catkin_ws>catkin_make -DCATKIN_ENABLE_TESTING=True -DCMAKE_BUILD_TYPE=Release
Base path: c:\catkin_ws
Source space: c:\catkin_ws\src
Build space: c:\catkin_ws\build
Devel space: c:\catkin_ws\devel
Install space: c:\catkin_ws\install
####
#### Running command: "cmake c:\catkin_ws\src -DCATKIN_ENABLE_TESTING=True -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=c:\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=c:\catkin_ws\install -G Ninja" in "c:\catkin_ws\build"
####
-- Using CATKIN_DEVEL_PREFIX: C:/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;c:\opt\ros\noetic\x64
-- This workspace overlays: c:/opt/ros/noetic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: True
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/noetic/x64/Scripts/nosetests.exe
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Create Debug Build.
-- realsense2_camera: 2 messages, 0 services
-- Configuring done
-- Generating done
-- Build files have been written to: C:/catkin_ws/build
####
#### Running command: "ninja -j6 -l6" in "c:\catkin_ws\build"
####
[8/28] Building CXX object realsense-ros\realsense2_camera...s\realsense2_camera.dir\src\realsense_node_factory.cpp.obj
FAILED: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.obj
C:\PROGRA~2\MIB055~1\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\cl.exe  /nologo /TP -DNOMINMAX -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"realsense2_camera\" -DWIN32_LEAN_AND_MEAN -D_USE_MATH_DEFINES -Drealsense2_camera_EXPORTS -IC:\catkin_ws\devel\include -IC:\catkin_ws\src\realsense-ros\realsense2_camera\include -IC:\catkin_ws\src\ddynamic_reconfigure\include -IC:\opt\ros\noetic\x64\include -IC:\opt\ros\noetic\x64\share\xmlrpcpp\cmake\..\..\..\include\xmlrpcpp -IC:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows\include -fPIE -fPIC -std=c++11 -D_FORTIFY_SOURCE=2 -fstack-protector -Wformat -Wformat-security -Wall /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG   /D _VARIADIC_MAX=10 /Zc:__cplusplus -std=c++11 /showIncludes /Forealsense-ros\realsense2_camera\CMakeFiles\realsense2_camera.dir\src\realsense_node_factory.cpp.obj /Fdrealsense-ros\realsense2_camera\CMakeFiles\realsense2_camera.dir\ /FS -c C:\catkin_ws\src\realsense-ros\realsense2_camera\src\realsense_node_factory.cpp
cl : Command line error D8021 : invalid numeric argument '/Wformat'
[10/28] Building CXX object ddynamic_reconfigure\CMakeFile...est.dir\test\ddynamic_reconfigure_auto_update_test.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[11/28] Building CXX object ddynamic_reconfigure\CMakeFile...rver.dir\test\test_bool_dynamic_reconfigure_server.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[12/28] Building CXX object ddynamic_reconfigure\CMakeFile...re_server.dir\test\fake_dynamic_reconfigure_server.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[13/28] Building CXX object ddynamic_reconfigure\CMakeFiles\ddynamic_reconfigure.dir\src\ddynamic_reconfigure.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
[... cut ...]
Invoking "ninja -j6 -l6" failed
```
Moreover I cannot use the rosdep install command in the folder "src":

```
c:\catkin_ws\src>rosdep install --from-paths src --ignore-src -r -y
given path 'src' does not exist

```
but one level above:

```
c:\catkin_ws>rosdep install --from-paths src --ignore-src -r -y
#All required rosdeps installed successfully
```
But even after entering the command, the next command in the instructions (catkin_make install) gives still error messages, do you have an idea what it could be?

MartyG-RealSense, 2022-03-11:
What method did you use to install the RealSense SDK, please - with the automated installer file **Intel.RealSense.SDK-WIN10** or with vcpkg.  If you installed with cpkg then I would recommend using the automated installer file instead if possible.

stevkr, 2022-03-11:
Hi, I removed all programs again (ROS & Intel SDK) hoping to get it error free.
I have now installed ROS Melodic. When I install Intel SDK with the automated installer file, after the command `catkin_make clean` appears the following error message : 

```
c:\catkin_ws>catkin_make clean
Base path: c:\catkin_ws
Source space: c:\catkin_ws\src
Build space: c:\catkin_ws\build
Devel space: c:\catkin_ws\devel
Install space: c:\catkin_ws\install
####
#### Running command: "cmake c:\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=c:\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=c:\catkin_ws\install -G NMake Makefiles" in "c:\catkin_ws\build"
####
-- Using CATKIN_DEVEL_PREFIX: C:/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\melodic\x64\tools\vcpkg\installed\x64-windows;C:/opt/ros/melodic/x64
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/ros/melodic/x64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/ros/melodic/x64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/melodic/x64/Scripts/nosetests-2.7.exe
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:42 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
So probably the automatic installer based SDK is not found.
Only after I install the SDK via vcpkg the command "catkin_make clean" works but then I get all the error messages in the following commands...

MartyG-RealSense, 2022-03-11:
2.34.0 is a very old version dating from March 2020, yes.  It is also a version that I do not recommend using because it had an issue with continuously generating timing errors for some RealSense users.

I would also recommend not using the vcpkg method to install librealsense if possible due to its complexity compared to the automated installer file and because Microsoft rather than Intel are responsible for vcpkg's upkeep.

More importantly, support for the L515 camera model was not introduced until 2.35.2, ta couple of releases after 2.34,0.

stevkr, 2022-03-11:
Yes I can confirm, first I installed RealSense SDK and then I try RealSense ROS wrapper.
After entering catkin_make clean command the following happens:

```
c:\catkin_ws>catkin_make clean
Base path: c:\catkin_ws
Source space: c:\catkin_ws\src
Build space: c:\catkin_ws\build
Devel space: c:\catkin_ws\devel
Install space: c:\catkin_ws\install
####
#### Running command: "cmake c:\catkin_ws\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=c:\catkin_ws\devel -DCMAKE_INSTALL_PREFIX=c:\catkin_ws\install -G NMake Makefiles" in "c:\catkin_ws\build"
####
-- The C compiler identification is MSVC 19.29.30141.0
-- The CXX compiler identification is MSVC 19.29.30141.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: C:\opt\ros\melodic\x64\tools\vcpkg\installed\x64-windows;C:/opt/ros/melodic/x64
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Found PythonInterp: C:/opt/ros/melodic/x64/python.exe (found suitable version "2.7.15", minimum required is "2")
-- Using PYTHON_EXECUTABLE: C:/opt/ros/melodic/x64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\ros\melodic\x64\lib\site-packages\em.pyc
-- Using empy: C:/opt/ros/melodic/x64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/melodic/x64/Scripts/nosetests-2.7.exe
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense2_description'
-- ==> add_subdirectory(realsense-ros/realsense2_description)
-- +++ processing catkin package: 'ddynamic_reconfigure'
-- ==> add_subdirectory(ddynamic_reconfigure)
-- +++ processing catkin package: 'realsense2_camera'
-- ==> add_subdirectory(realsense-ros/realsense2_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:42 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
So i guess, the installation of the SDK via .exe is not recognized

MartyG-RealSense, 2022-03-11:
I researched the issue further but are aware that in past cases involving this problem, if the advice provided above does not solve the problem with installing the ROS wrapper on Windows then a solution is usually not achieved.  The last resort would be to check the advice given in https://github.com/IntelRealSense/realsense-ros/issues/280 which provides suggestions such as deleting the build folder in catkin_ws and then re-running catkn make.  I do apologize.

MartyG-RealSense, 2022-03-15:
The vast majority of RealSense ROS setups are on Linux operating systems.

MartyG-RealSense, 2022-03-21:
I'm pleased to hear that you found a solution, @stevkr - thanks very much for the update!

wolegequwangge, 2023-02-01:
version 2.2.22 successful

MartyG-RealSense, 2023-02-01:
Thanks very much @wolegequwangge for sharing your experience with the RealSense ROS community of what works for you!