realsenseai/realsense-ros #2282: xxxxxx_multiple_cameras.urdf.xacro doesn't support different variant cameras

HappySamuel, 2022-03-14:
Hi

Using test_d435_multiple_cameras.urdf.xacro can properly load d435 cameras. However when adding <xacro:include > any other variant of cameras, it will throw error ` Error:   material 'aluminum' is not unique`. Is there any solution for this? Because i wanna build a robot model that using 2 / more different type of cameras.

Best,
Samuel

MartyG-RealSense, 2022-03-14:
Hi @HappySamuel  I have not encountered this particular error before in relation to librealsense.  My research of your problem indicates that it has also occurred in non-RealSense cases where multiple robots or multiple different components on the same robot have been used.  A ROS forum discussion at the link below suggests a couple of workarounds.

https://answers.ros.org/question/304797/pragma-once-equivalent-for-urdfxacro-files/

HappySamuel, 2022-03-15:
Hi

I tried `xacro:include` different type of cameras.urdf in both ROS1 and ROS2, both ROSversions will throw same error `Error: material 'aluminum' is not unique`

[image]

Best,
Samuel

MartyG-RealSense, 2022-03-15:
What happens if you edit the D455 urdf file to change the material name from 'aluminum' to **aluminum2** to distinguish it from the definition in the D435 urdf file?

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_description/urdf/_d455.urdf.xacro#L70

HappySamuel, 2022-03-15:
Hi 

> What happens if you edit the D455 urdf file to change the material name from 'aluminum' to **aluminum2** to distinguish it from the definition in the D435 urdf file?
> 
> https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_description/urdf/_d455.urdf.xacro#L70

This doesn't solve the issue. The issue is due to both` _d435.urdf.xacro` and `_d455.urdf.xacro` are including same `_materials.urdf.xacro`, which resulting duplication. And i found a solution is not to `<xacro:include filename="$(find realsense2_description)/urdf/_materials.urdf.xacro" />` in any cameras' urdf, while for those need material color can redefine the material in each cameras' urdf.xacro, like below:
```
<material name="aluminum">
      <color rgba="0.5 0.5 0.5 1"/>
</material>
```

Above suggested solution has been tested with 3 different type of cameras and there's no error thrown out. Below is the multiple_cameras.urdf
[image]

Well, the best approach is still using .dae file which contain color, therefore can drop the usage of material color.

Best,
Samuel

MartyG-RealSense, 2022-03-15:
Thanks so much @HappySamuel for sharing your solution with the RealSense ROS community!

HappySamuel, 2022-03-15:
Hi

You're welcome. Hope the modification will be applied into debian version too, then everyone can install it via apt directly.

Best,
Samuel

MartyG-RealSense, 2022-03-15:
If you wish, you can submit a Pull Request (PR) for your change so that it can be considered by the RealSense team for official inclusion in the RealSense ROS wrapper.  Thanks again!

https://github.com/IntelRealSense/realsense-ros/pulls

HappySamuel, 2022-03-15:
Hi

How shall i submit a PR for the change? I have never done it before. Any tutorials?

Best,
Samuel

MartyG-RealSense, 2022-03-15:
Pull submission nstructions for the main librealsense GitHub should be applicable to the RealSense ROS one too.

1.  Go to the ROS wrapper's pull request page and click the **New pull request** button

2.  For **base** combo-box select **development**, since you want to submit a PR to that branch

3.  For **compare** combo-box select **name_of_your_contribution** with your commit

4.  Review your changes and click **Create pull request**

5.  Wait for all automated checks to pass

6.  The PR will be approved / rejected after review from the team and the community

caranx-medical, 2022-04-20:
Has been this merged?

MartyG-RealSense, 2022-04-20:
Hi @caranx-medical  It has not yet been confirmed by @HappySamuel that a pull was submitted by them.

caranx-medical, 2022-04-20:
Created a PR: https://github.com/IntelRealSense/realsense-ros/pull/2325

MartyG-RealSense, 2022-04-20:
Thanks very much, @caranx-medical - I have added an **Enhancement** tag to this case and re-opened it.  The case should be kept open for the durartion that the PR is active.

srirsank, 2022-07-19:
Is there a time-frame for approving this PR for merge?  Seems pretty straight-forward and useful for any and all projects working with multiple cameras.

MartyG-RealSense, 2022-07-19:
Hi @srirsank  The PR should be merged at a future date but there is not a specific time estimate currently available.

luis-camero, 2022-09-09:
Being able to use two Realsense cameras of different models without having to create our own URDF's is a fairly reasonable ask. 
Has anyone reviewed this PR yet? And if so, is there any reason this implementation (with separate definitions of materials) undesirable?

MartyG-RealSense, 2022-09-10:
Hi @luis-camero  The PR is awaiting review at a future date.   There is not a time estimate available, unfortunately.

HappySamuel, 2022-09-12:
Hi @MartyG-RealSense 

There's someone submitted PR that use the same solution i suggested, but still pending to be approved.
[[Bugfix] Commented materials urdf and defined each material aluminum … #2325](https://github.com/IntelRealSense/realsense-ros/pull/2325)
Need your help to review it and approve, so the community will now have this issue resolved.

Best,
Samuel

MartyG-RealSense, 2022-09-12:
Hi @HappySamuel  I will pass the request from yourself and others who have recently asked about it to the ROS wrapper development team.  Thanks!

MartyG-RealSense, 2022-09-23:
Hi @HappySamuel  The ROS wrapper development team was pleased to receive your enquiry about your PR.  They created their own official internal report so that they can explore the subject of multiple URDF.

SamerKhshiboun, 2023-04-11:
Hi all,

As I see there are several PRs solving this issue:
- #2325 (This PR)
- #1637 
- #2674 

I think that all of them solve the problem, but as I see #1637 is the one with minimal changes.
Do you agree with me ?
If so, I can close other PRs and approve that PR.

Thanks,
Samer

HappySamuel, 2023-04-12:
OK, as long as this issue resolved in Gazebo simulation.

Best,
Samuel

closed 2023-04-12 by MartyG-RealSense