realsenseai/realsense-ros #2283: [ROS2] Fail to spawn cameras' urdf into Gazebo

HappySamuel, 2022-03-15:
Hi

I faced issue for importing any of the cameras' urdf (`d415`, `d435`, `d455`, `l515`) into `Gazebo`. There's no issue for loading their urdf into rviz2. As you can see from below picture, this one indicate the successful loading of cameras' urdf into rviz2
[image]

While another one shows the error of couldn't find the location of the cameras' mesh files.
[image]

Any solution for this? I'm trying to load them into Gazebo.

Best,
Samuel

MartyG-RealSense, 2022-03-15:
Hi @HappySamuel  My research found a similar situation in a non-RealSense case at https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1272 where the urdf could be imported successfully into Gazebo in ROS but not in ROS2.  

A commenter in that discussion at https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1272#issuecomment-858407996 suggested that the spawning method had changed in ROS2 and it was only able to find the meshes if they were in the default Gazebo models directory instead of the workspace.

HappySamuel, 2022-03-15:
Hi

I found some reference from here [https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1284#issuecomment-1028393652](url)
For ROS2 usage, by changing the following
`<mesh filename="package://realsense2_description/meshes/d415.stl" />`
into
`<mesh filename="file://$(find realsense2_description/meshes/d415.stl" />`
will resolve the issue, as it will evaluate to the full path when xacro generates the URDF.

Right after i found the suggestion from the link, i tried out on all 4 type cameras (`d415`, `d435`, `d455`, `l515`) that `ROS2 realsense2_description` provided, and all of them work like charm.

You may try out the same, strongly believe this will help all realsense users on gazebo simulation.

Best,
Samuel

closed 2022-03-15 by HappySamuel

MartyG-RealSense, 2022-03-15:
Thanks @HappySamuel   :)