realsenseai/realsense-ros #2286: D435 optical frame orientation

antoineTK, 2022-03-15:
Hi guys

I have noticed there are slight angular values in tranforms between camera_link and optical_frame, is this correct? (got a D435_camera_FWD prefix in screenshot below)

[image]

All other frames seem to be aligned or have 90° rotations on some axis. Only optical frame has those "weird" angles.

Would anyone know if that's how it's supposed to be?

antoineTK, 2022-03-17:
got it, I guess it's normal then, had to check

thanks a lot @MartyG-RealSense !

MartyG-RealSense, 2022-03-23:
Hi @antoineTK  Do you require further assistance with this case, please?  Thanks!

antoineTK, 2022-03-24:
all good, thanks @MartyG-RealSense

MartyG-RealSense, 2022-03-24:
Great, thanks very much @antoineTK for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-03-24 by MartyG-RealSense