realsenseai/realsense-ros #2287: Unable to locate pacakge ros-galactic-realsense2-camera

maximereloxrob, 2022-03-16:
Hi, I have been using the package realsense2-camera (galactic) in a ROS2 container for a while now. A couple of days ago, when rebuilding the container I got the following error:

MartyG-RealSense, 2022-03-16:
Hi @maximereloxrob   I am not involved in ROS wrapper development activities and so do not have information about the status of the packages, though I note that the ROS2 wrapper page for Galactic is available.

https://index.ros.org/p/realsense2_camera/#galactic

If you need to install the wrapper as soon as possible and are using librealsense 2.50.0 then I would recommend building wrapper version 3.2.3 from source code.

maximereloxrob, 2022-03-16:
Hi @MartyG-RealSense, thanks for the answer. I will try what you suggested. Cheers!

Nir-Az, 2022-03-17:
We are trying to investigate what went wrong and working on a fix for the public rosdistro release, I will update once it will be fixed.

MartyG-RealSense, 2022-03-20:
Hi @maximereloxrob  There has been a new ROS2 wrapper release today (version 4.0.4).

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.0.4

maximereloxrob, 2022-03-21:
Thanks @MartyG-RealSense, will look into that :+1:

MartyG-RealSense, 2022-03-27:
Hi @maximereloxrob  Do you require further assistance with this case, please?  Thanks!

maximereloxrob, 2022-03-28:
@MartyG-RealSense I tested what you recommended and it seems to work fine for me.

I tried again today with `apt-get install -y ros-$ROS_DISTRO-realsense2-camera`. Still not working unfortunately. 

I will let you judge if you want to keep this issue open or not.

Thanks for the help!

MartyG-RealSense, 2022-03-28:
@maximereloxrob  You are very welcome!

The RealSense ROS wrapper developers advised that at the time of writing this, if ROS2 Foxy or Galactic is being used then the 4.x wrapper has to be built from source code or from the ubuntu/testing server until the next ROS sync is completed.

As your source-code build is working, I will close this issue.  Thanks again!

closed 2022-03-28 by MartyG-RealSense

maximereloxrob, 2022-04-01:
The issue was fixed in the latest ROS2 galactic sync (https://discourse.ros.org/t/preparing-for-galactic-sync-2022-03-31/24888).

Tested with `apt-get install -y ros-galactic-realsense2-camera` and working fine.

MartyG-RealSense, 2022-04-01:
Great, thanks very much @maximereloxrob for the confirmation that the sync resolved the issue for you!