realsenseai/realsense-ros #2289: Pluging and Unpluging a d435i camera while using ROS components

imstevenpmwyca, 2022-03-16:
We are running into the following issue when we load the camera ROS component and we physically plug/unplug the camera.

We load the camera ROS component with the camera plugged. Everything works fine. Then we unplug the camera and get the usual info/error messages. Then, if we plug back the camera again, the system works fine but we still get the following errors spamming:
`16/03 13:18:54,397 ERROR [140270528091904] (librealsense-exception.h:52) acquire: Cannot open '/dev/video6 Last Error: No such file or directory
 16/03 13:18:54,397 ERROR [140270528091904] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 16/03 13:18:54,397 ERROR [140270528091904] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 16/03 13:18:54,397 ERROR [140270528091904] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video6 Last Error: No such file or directory
 16/03 13:18:54,506 ERROR [140270649964288] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 16/03 13:18:54,506 ERROR [140270649964288] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 16/03 13:18:54,506 ERROR [140270649964288] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 16/03 13:18:54,506 ERROR [140270649964288] (error-handling.cpp:78) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device`

You can easily replicate these behaviors by running:
`ros2 run rclcpp_components component_container` and
`ros2 component load /ComponentManager realsense2_camera realsense2_camera::RealSenseNodeFactory -p usb_port_id:=X.Y`
And proceeding to unplug or plug the camera as described.

We are using multiple cameras at the same time, so that why we need to set a specific usb_port when loading the ROS component. We are using the latest ros2 branch.

Thank you very much for your time and help,
Kind regards!

MartyG-RealSense, 2022-03-16:
Hi @imstevenpmwyca  Which method are you using to roslaunch the multiple cameras, please?  Are you launching each camera individually in its own separate ROS terminal with **rs_launch.py** or are you launching them all with a single instruction using **rs_multi_camera_launch.py**

https://github.com/IntelRealSense/realsense-ros/tree/ros2#work-with-multiple-cameras

Detecting the cameras via **usb_port_id** to find a camera on a specific USB port is typically done when a camera cannot be reliably detected by providing its serial number (in which case it does not matter which USB port a particular camera is plugged into).

MartyG-RealSense, 2022-03-16:
There are not many references about accessing the camera with **ros2 run** instead of roslaunch.  It does seem though that the problem is occurring because forcibly disconnecting the camera is causing a conflict with the kernel, resulting in the **xioctl(VIDIOC_REQBUFS)** errors.  It may therefore be worth testing whether building the librealsense SDK with the **RSUSB backend** method resolves the errors.

An RSUSB build of librealsense bypasses the Linux kernel.  It is not dependent on Linux versions or kernel versions and does not require patching.

When building librealsense and the RealSense ROS wrapper separately, librealsense should be built first and then the ROS wrapper built from source code secondly afterwards.

MartyG-RealSense, 2022-03-22:
Hi @imstevenpmwyca  Do you require further assistance with this case, please?  Thanks!

imstevenpmwyca, 2022-03-22:
Dear @MartyG-RealSense,

> There are not many references about accessing the camera with ros2 run instead of roslaunch.

I am sorry if I didn't explain myself properly, but we are not `ros2 run`-ing the camera nodes, we are `ros2 component load`-ing. This is similar to what a nodelet was in ROS1, [see more here](https://docs.ros.org/en/foxy/Concepts/About-Composition.html). However, it seems you managed to replicate the error messages, so that means the issue isn't related to how we start the camera node and hence, it isn't relevant either.

I still have to test whether building the librealsense SDK with the RSUSB backend method works, as you proposed. I'll update as soon as I do, or maybe you already did so?

Thank you very much for your time and consideration,
Kind regards!

MartyG-RealSense, 2022-03-22:
Please perform a build of the librealsense SDK with the RSUSB backend method when you have the opportunity to do so and update with the results here.  Thanks!

MartyG-RealSense, 2022-03-28:
Hi @imstevenpmwyca  Do you have an update about this case that you are able to provide, please?  Thanks!

MartyG-RealSense, 2022-04-10:
Case closed due to no further comments received.

closed 2022-04-10 by MartyG-RealSense

imstevenpmwork, 2022-09-16:
Dear @MartyG-RealSense,
Is there any guide for the RSUSB backend built method?
Thanks for your support, greetings!

MartyG-RealSense, 2022-09-16:
Hi @imstevenpmwork  RSUSB installation instructions can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

Alternatively, there is also a simple installation procedure at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md