realsenseai/realsense-ros #2290: D405 ROS Wrapper

fiorano10, 2022-03-18:
Currently, only ROS2 supports the new D405 camera, is there a timeline when it will be available for ROS 1? If yes, what is it?

MartyG-RealSense, 2022-03-18:
Hi @fiorano10   I will make enquiries about your question.  Thanks in advance for your patience.

closed 2022-03-23 by fiorano10

eric-schleicher, 2022-04-27:
I'm a bit unclear on the final disposition of this issue.   Is there an accepted pattern for using ROS2 wrappers to provide compatibility for non-EOL distributions of ROS1 (noetic, melodic)?   Or is this a quixotic recommendation stipulating that customers of with a *new* 405 camera MUST upgrade their ROS platform system to ROS2?  I hope not, because that sounds insane.

[image]

MartyG-RealSense, 2022-04-28:
The new D405 model is only supported by the 4.x ROS2 wrapper series on the **ros2_beta** branch.  There are no plans to add support for D405 to the 2.x ROS1 wrapper or the previous 3.x ROS2 wrapper as the focus of the RealSense ROS wrapper's developer team is now on the 4.x ROS2 wrapper series.

eric-schleicher, 2022-04-30:
Without prejudice; it's unfortunate to see this get passed over.  Hopefully this will change in the future.