realsenseai/realsense-ros #2294: hardware failure issue!

Karishinamdar, 2022-03-22:
I am trying to launch realsense D435i but it's showing following error. The camera is working fine with inlet-realsense viewer. Could it be camera communication issue? Do I need to replace my camera? 

[ WARN] [1647982691.323889089]: Hardware Notification:Depth stream start failure,1.64798e+12,Error,Hardware Error

MartyG-RealSense, 2022-03-22:
Hi @Karishinamdar  If your camera is working correctly with the RealSense Viewer program then there is likely not a problem with the camera, the firmware driver or the USB cable / port.

Is the **Depth stream start failure** warning occurring when launching the RealSense ROS wrapper?  If it is then please first try adding **initial_reset:=true** to your launch instruction to reset the camera at launch if you are not using the reset instruction already..  

For example for ROS1:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

Or for ROS2:

**ros2 launch realsense2_camera rs_launch.py initial_reset:=true**

Karishinamdar, 2022-03-22:
Hi @MartyG-RealSense, I have tried that as well. When I run with rtab map I get above problems and independently I am noticing the some new problems such as:
 22/03 15:02:51,445 WARNING [140012578404096] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-03-22:
The **Resource temporarily unavailable** type of control_transfer returned warning may suggest that the processor of your computer / computing device is having difficulty keeping up with processing the incoming frames, at least with the ROS wrapper (since the Viewer is performing normally).

What computer are you using the camera with, please (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc)?

Karishinamdar, 2022-03-22:
I am using my msi laptop (intel 10th generation core i7) with Ubuntu 20.04.4 LTS on it. I run htop while using ROS wrapper, it didn't show CPU % high or memory leak.

MartyG-RealSense, 2022-03-23:
What instruction are you using to launch the camera, please?

Karishinamdar, 2022-03-23:
I am trying to use rtab map for localization and mapping. First I am using this for RGBD-sensor:
$ roslaunch realsense2_camera rs_camera.launch \
    align_depth:=true \
    unite_imu_method:="linear_interpolation" \
    enable_gyro:=true \
     enable_accel:=true

$ rosrun imu_filter_madgwick imu_filter_node \
    _use_mag:=false \
    _publish_tf:=false \
    _world_frame:="enu" \
    /imu/data_raw:=/camera/imu \
    /imu/data:=/rtabmap/imu

and then for mapping I am using this:
 roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false \
    wait_imu_to_init:=true \
    imu_topic:=/rtabmap/imu

MartyG-RealSense, 2022-03-23:
Have you tried Intel's D435i SLAM guide for ROS, which also uses rtabmap_ros and imu_filter_madgwick, but uses the launch file **opensource_tracking.launch** instead of rs_camera.launch

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

Karishinamdar, 2022-03-24:
This actually worked! Thanks a lot.

MartyG-RealSense, 2022-03-24:
You are very welcome.  I'm pleased to hear that you were successful, @Karishinamdar  :)

closed 2022-04-04 by MartyG-RealSense