realsenseai/realsense-ros #2295: pointcloud not published in ROS2 Foxy

Yoshihito-M, 2022-03-23:
- Infos
LibRealSense v2.50.0
ROS2 Foxy
ROS2 Foxy Wrapper build 4.0.4
Ubuntu 20.04　LTS
D455

- Ploblem
After launching
`ros2 launch realsense2_camera rs_launch.py filters:=pointcloud align_depth:=true`,

using ros2 topic list gives me the topics as follows:
`/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/parameter_events
/rosout
/tf_static`.

As shown above, the topics /camera/depth/color/points is not listed. 
In addition, when launching the launch file, the concerned WARN message is shown as follows:
`[WARN] could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: 
parameter 'pointcloud.stream_index filter' could not be set: Parameter {} doesn't comply with integer range`.

I guess that that WARN message affect failure of the topics /camera/depth/color/points.
I would like to know how to fix 'pointcloud.stream_index filter'.

MartyG-RealSense, 2022-03-23:
Hi @Yoshihito-M  Do you receive the same warnings in ROS2 wrapper 4.04 when enabling the pointcloud using the launch instruction below?

**ros2 launch realsense2_camera rs_launch.py align_depth:=true pointcloud.enable:=true**

Yoshihito-M, 2022-03-24:
Hi @MartyG-RealSense 

Thank you for helpful advice.
/camera/depth/color/points could be listed by using the launch instruction .
However, another error message is shown as follows when displaying /camera/depth/color/points in Rviz2:
`No stream match for pointcloud chosen texture process - Color`.

MartyG-RealSense, 2022-03-24:
The warning **No stream match for pointcloud chosen texture process - Color** can appear every time that there is an unavailable texture, likely because of a frame drop.  My understanding is that it is an annoyance rather than something that prevents the point cloud from streaming correctly.  Is the point cloud able to be generated in RViz2 even though the warning is appearing, please?

Yoshihito-M, 2022-03-30:
@MartyG-RealSense 
The issue I mentioned above has been cleared up.
The important point to display /camera/depth/color/points in Rviz2 is plugging in PC into the power outlet while using Rviz2.
I guess that Rviz2 should not be used on battery-powered mode of PC.

As a result, I could solve my issue by your advices.
Thanks!!

closed 2022-03-30 by Yoshihito-M

MartyG-RealSense, 2022-03-30:
I'm pleased to hear that you achieved a solution, @Yoshihito-M - thanks very much for the update!

JMadhup, 2023-07-04:
hi @Yoshihito-M and @MartyG-RealSense,

I am using the D445 camera on the humble version of ROS2. The problem I am facing is that when I do a ros2 topic list, the /camera/depth/color/points topic is present but it seems no data is being published over it. I tried ros2 topic echo and also tried to visualize it in rviz2 but couldn't visualize the data or echo it in the terminal.

MartyG-RealSense, 2023-07-04:
Hi @JMadhup  Are you including **pointcloud.enable:=true** in your launch instruction, please?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true`

JMadhup, 2023-07-04:
@MartyG-RealSense yes that is exactly the same command I am using. 
I have also tried using ros2 launch realsense2_camera rs_launch.py align_depth:=true pointcloud.enable:=true but the issue persists.

MartyG-RealSense, 2023-07-04:
Which RealSense ROS wrapper versions and librealsense SDK versions are you using?

JMadhup, 2023-07-05:
@MartyG-RealSense I am not sure how to check which version I am using. Could you assist me with how to check the version?

JMadhup, 2023-07-05:
@MartyG-RealSense I am using librealsense version 2.51.1 and realsense ROS version 4.51.1

MartyG-RealSense, 2023-07-05:
In the ROS2 wrapper, the align parameter is `align_depth.enable ` rather than align_depth.

I note that the `/camera/depth/color/points `topic is published in the list of topics but there are no messages.  This at least indicates that the topic is being created by pointcloud.enable. 

Are there still no messages if the launch is defined with a low depth and color resolution and FPS?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.profile:=640x480x15 pointcloud.enable:=true`

JMadhup, 2023-07-05:
@MartyG-RealSense No. Chaning launching with low FPS does not help

MartyG-RealSense, 2023-07-05:
Which messages, if any, are published in the ROS launch log after **RealSense Node is Up** (the point in the log where launch is completed and the sesssion is now running).

JMadhup, 2023-07-05:
@MartyG-RealSense 
[realsense2_camera_node-1] [WARN] [1688555081.354793351] [camera.camera]:
 is the only message after that

MartyG-RealSense, 2023-07-05:
What method did you use to install librealsense and the ROS wrapper?  Did you install librealsense first and test that it worked successfully, and then install the wrapper secondly after that from source code with **colcon build**

JMadhup, 2023-07-05:
@MartyG-RealSense 
I first built librealsense from source and tested that it worked successfully and then installed the ROS wrapper using apt

MartyG-RealSense, 2023-07-05:
Let's check if the align function is able to publish aligned topics and messages are being generated.  If align also does not publish messages then it could suggest a problem with the ROS wrapper or the particular computer that the wrapper is being run on. 

`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true`

JMadhup, 2023-07-27:
Hello @MartyG-RealSense,

Sorry for the late response, I got otherwise engaged.

I tried using the command that you specified and the align function **does** infact publish the aligned topic and messages are also being generated. I was able to visualize the image_raw message using the image_view package but the image was same as it is on the /camera/depth/image_rect_raw topic

MartyG-RealSense, 2023-07-27:
What do the aligned topics look like in **RViz** instead of image_view?

MartyG-RealSense, 2023-07-27:
@JMadhup  Which computer you are using.  Is it a PC such as a laptop or desktop, or a computing board such as Raspberry Pi or Nvidia Jetson?

JMadhup, 2023-07-28:
@MartyG-RealSense 
I am using a Raspberry Pi 4 running an installation of ubuntu desktop 22.04

MartyG-RealSense, 2023-07-28:
Is the pointcloud published if you leave out the `align_depth` parameter from the launch, as it should be generated even if align_depth is not set to True.

` ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true `

ashBabu, 2023-08-31:
Raspbery Pi 4 (8 GB), Ubuntu 22.04, ROS2 humble, d435i

Launching realsense camera with the launch_arguments `"pointcloud.enable": "True"` on raspberry pi 4 will not publish pointclouds. This argument in fact slows down the publishing rate of depth images from the specified rate of 30 to around 8 Hz. However if the `pointcloud.enable` argument is removed, then the rate of publishing of depth images is at the specified 30 Hz itself and by using either the `plugin='rtabmap_util::PointCloudXYZ'` or [depth_image_proc](http://wiki.ros.org/depth_image_proc), the pointclouds could be published at around the same 30 Hz rate. I installed everything using apt. `sudo apt-get install ros-humble-realsense2*`. More details on how to launch  (for anyone interested) is available at [my blog](https://ashbabu.github.io/blog/2023/Obstacle-Detection-from-Pointclouds/)

MartyG-RealSense, 2023-08-31:
Thanks so much @ashBabu for sharing your Pi pointcloud method with the RealSense community!

anishk85, 2026-02-08:
Hi @MartyG-RealSense,
I’m using an Intel RealSense D415 on ROS 2 Humble running on a Jetson Nano. I launched rs_launch.py with the argument enable_pointcloud:=true, but I still don’t see the point cloud topic being published.

Could you please help me understand why this might be happening and what I should check to get the topic published correctly?