realsenseai/realsense-ros #2299: /camera/aligned_depth_to_color/image_raw not publish on Raspi 4B

howtoloveyou, 2022-03-24:
- Infos
    Raspi 4B
    LibRealSense v2.50.0
    ROS2 Galactic
    ROS2 Galactic Wrapper build 4.0.4
    Ubuntu 20.04
    D455
-  Ploblem
    After launching
   `ros2 launch realsense2_camera rs_launch.py align_depth:=true depth_module.profile:=1280x720x30`
   I can get results from `ros2 topic echo /camera/depth/image_rect_raw`
   but I can't see any results from `ros2 topic echo /camera/aligned_depth_to_color/image_raw`
Is anything wrong here? Thank you!

MartyG-RealSense, 2022-03-24:
Hi @howtoloveyou  Have you confirmed that the topic **/camera/aligned_depth_to_color/image_raw** is being published by using the command **ros2 topic list** to list all the published topics, please?

May I also ask which method you used to build the ROS wrapper 4.0.4 (packages or source code).  At the time of writing this, when using ROS2 Foxy or Galactic the 4.x wrapper has to be built from **source code** or from the **ubuntu/testing server** until the next sync is completed.

howtoloveyou, 2022-03-25:
@MartyG-RealSense
1. I'm sure i can see the topic **/camera/aligned_depth_to_color/image_raw** by using the command `ros2 topic list`, 
2. And I was build the ROS wrapper 4.0.4 from source code by using this command `git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-beta`
3. By the way, I can't locate the **ros-galactic-realsense2-camera** by using 'sudo apt install ros-galactic-realsense2-*', is there any wrong with Raspi 4B or **ARM64**？As I known, ARM should download software from ubuntu-ports rather than ubuntu. ^_^|||
---
Thank you!

howtoloveyou, 2022-03-25:
@MartyG-RealSense 
1. I have solved my problem by using **--branch ros2**, and I can get the information from topic **/camera/aligned_depth_to_color/image_raw** now. (Raspi4B ubuntu20.04 Galactic)
2. Maybe **ROS wrapper 4.0.4** doesn't support Raspi(or ARM) ? -_-|||
3. I will close this issue soon, thank you very much!

closed 2022-03-25 by howtoloveyou

MartyG-RealSense, 2022-03-25:
It's great to hear that you achieved a solution!   :)

At the time of writing this, Foxy and Galactic users should build the ros2_beta branch's 4.x wrapper from source code or by using the 'ubuntu/testing' server instead of building from packages.  Package **apt** install should become possible once the wrapper becomes available on the apt server via a ROS 'sync' that makes the packages accessible to the ROS end-user.

I am not aware of any problems with the 4.x wrapper regarding Arm compatibility.