realsenseai/realsense-ros #2301: RS2_USB_STATUS_ACCESS error code

dyhe-2000, 2022-03-26:
[INFO] [1648269669.766889818] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1648269669.768663231] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1648269669.769975543] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [WARN] [1648269669.850828398] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1648269669.851126319] [camera.camera]: The requested device with  is NOT found. Will Try again.
[realsense2_camera_node-1]  26/03 05:41:09,848 ERROR [547272780192] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
[realsense2_camera_node-1]  26/03 05:41:09,849 ERROR [547768654240] (sensor.cpp:572) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  26/03 05:41:09,850 WARNING [547768654240] (rs.cpp:310) null pointer passed for argument "device"

MartyG-RealSense, 2022-03-26:
Hi @dyhe-2000  Do you still experience the error if you include **initial_reset:=true** in the launch instruction to reset the camera at launch, please?  For example:

**ros2 launch realsense2_camera rs_launch.py initial_reset:=true**

dyhe-2000, 2022-03-26:
Yes the issue persisted after using initial_reset:=true. I am using https://github.com/IntelRealSense/realsense-ros/releases/tag/4.0.4 for ros2 galactic.
I first downloaded the package into my workspace src file. Then I installed all rosdep successfully, and concon builded.

dyhe-2000, 2022-03-26:
@MartyG-RealSense I use D455

MartyG-RealSense, 2022-03-26:
Thank you for the information.  Does the camera work correctly when used with the RealSense Viewer program?

Can you also confirm please that you installed librealsense first and then installed the ROS wrapper secondly after the librealsense installation had been successfully completed?

dyhe-2000, 2022-03-26:
@MartyG-RealSense Hi, can we have a zoom call? I am available 03/27/2022 12pm-11pm PST, 03/28/2022 12pm-11pm PST or a later time as your convenience.

MartyG-RealSense, 2022-03-26:
Support can only be provided through this forum.  Please post images into the comments to visually illustrate your problem if it is possible for you to do so.

closed 2022-04-08 by MartyG-RealSense