realsenseai/realsense-ros #2304: Enabling disparity_filter and subscribing to depth image's camera_info topic throws an error

natali-maidbot, 2022-03-29:
device info:
```
Device Name: Intel RealSense D435
Device FW version: 05.13.00.50
```

configuration:
- enable_depth : true
- alight_depth.enable : false
- decimation_filter.enable : true
- disparity_filter.enable : true
- disparity_to_depth.enable : false
- temporal_filter.enable : true

What:

After enabling the disparity filter on the depth image, when subscribing to the depth image's camera_info topic this error is thrown:
`[ERROR] [1648582895.516828389] [devices.d435.camera]: An error has occurred during frame callback: map::at
`
This error seems to be thrown because float32 encoding is not supported? see the `_encoding` map in `BaseRealSenseNode` class.

Expected behavior: 
- disparity_filter is enabled 
- filtered Z16 depth image and it's camera_info are published

MartyG-RealSense, 2022-03-30:
Hi @natali-maidbot  When the error **An error has occurred during frame callback: map::at** has occurred in the past, the first suggested action to try as a workaround has often been to set **enable_infra** to True, as described by a RealSense team member at https://github.com/IntelRealSense/realsense-ros/issues/1348#issuecomment-684818853

john-maidbot, 2022-04-02:
Following up on @natali-maidbot 's post. Thank you for the quick reply @MartyG-RealSense ! We tried setting `enable_infra` to true, but that did not resolve the problem for our case. I think what we are seeing is actually an unexpected / undocumented new feature. I have seen very similar behavior on the realsense-viewer.
In the realsense-viewer, if you go into depth module post-processing and enable the disparity filter, the depth image is convert to disparity before filtering. and the output of the filtering is a float32 disparity image. And if you want to view the 16-bit depth image, you have to also enable disparity_to_depth.
In the old realsense-ros, if you set `disparity` in the `filters` comma-separated string parameter, realsense-ros seemed to interpret this as, convert the depth image to disparity --> do the filtering in disparity space --> and then convert it back to depth before publishing.
In the new version of realsense-ros we are using (v4.0.4), it seems that the interface got closer to the realsense-viewer interface I described. So if you set disparity_filter.enable to true, the output of the filtering is a float32 disparity image (and since realsense-ros does not support float32 encoding, the node throws that map::at error). If you want the 16-bit depth output, it looks like you now have to also set `disparity_to_depth` to true. Doing this appears to tell librealsense to convert the disparity image back to depth after filtering and before handing it off to realsense-ros. 
`TL;DR` It appears that setting disparity_filter.enable to true AND disparity_to_depth to true is required in realsense-ros v4.0.4 and removes the map::at error (Because then realsense-ros will get the 16-bit image from librealsense as expected).

MartyG-RealSense, 2022-04-03:
Thanks so much @john-maidbot for your very detailed feedback about setting the disparity filter in the new 4.x ROS2 wrapper.  :)

MartyG-RealSense, 2022-04-10:
Hi @natali-maidbot and @john-maidbot  Do you require further assistance with this case, please?  Thanks!

john-maidbot, 2022-04-14:
No, this case is resolved for us. Thank you! 
For posterity, do you think the `disparity_to_depth` parameter should be added to the ros2-beta readme?

MartyG-RealSense, 2022-04-14:
Thanks very much for the update!

I will forward your documentation suggestion to the RealSense ROS team.

closed 2022-04-15 by natali-maidbot