realsenseai/realsense-ros #2305: I can't see the infrared camera

alvaropascuaI, 2022-03-30:
When I launch the following command ros2 launch realsense2_camera rs_launch.py it tells me that the stream depth and color are enabled but about the infrared it doesn't tell me anything, I have also launched the /camera/realsense2_camera/enable_infra1:=True commands and I get The topic about the infrared camera still does not come out

Device Name: Intel RealSense D435
Ros Version: Ros foxy

MartyG-RealSense, 2022-03-30:
Hi @alvaropascuaI  In the RealSense ROS wrapper, the Infra1 and Infra2 topics are disabled by default and have to be enabled in order to be published.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/launch/rs_launch.py#L38-L39

Is the Infra1 topic published if you use the ROS2 launch command below, please?

**ros2 launch realsense2_camera rs_launch.py enable_infra1:=true**

alvaropascuaI, 2022-03-31:
I have both set to true, but the topic about the infra still does not come out, I do not know if I will have to enable the stream over the infrared anyway

MartyG-RealSense, 2022-03-31:
Would it be possible to post your ros launch log from the terminal in a comment below, please?

alvaropascuaI, 2022-03-31:
[INFO] [launch]: All log files can be found below /home/panoimagen/.ros/log/2022-03-31-09-54-04-975959-panoimagen-27516
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [27518]
[realsense2_camera_node-1] [INFO] [1648720445.493761824] [camera.camera]: RealSense ROS v3.2.3
[realsense2_camera_node-1] [INFO] [1648720445.494025228] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1648720445.494079956] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1648720445.768036832] [camera.camera]: Device with serial number 950122070567 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1648720445.768210247] [camera.camera]: Device with physical ID 2-4-4 was found.
[realsense2_camera_node-1] [INFO] [1648720445.768252095] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1648720445.768862429] [camera.camera]: Device with port number 2-4 was found.
[realsense2_camera_node-1] [INFO] [1648720445.768913067] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1648720445.774758432] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1648720445.786362293] [camera.camera]: setupDevice...
[realsense2_camera_node-1] [INFO] [1648720445.786462724] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1648720445.786510124] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1648720445.786571551] [camera.camera]: Device physical port: 2-4-4
[realsense2_camera_node-1] [INFO] [1648720445.786609566] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1648720445.786644916] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1648720445.786682309] [camera.camera]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1648720445.786720704] [camera.camera]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1648720445.786756696] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1648720445.786827296] [camera.camera]: Device Sensors:
[realsense2_camera_node-1] [INFO] [1648720445.897900574] [camera.camera]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1648720445.921506269] [camera.camera]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1648720445.921963349] [camera.camera]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1648720445.922193673] [camera.camera]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1648720445.922266017] [camera.camera]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1648720445.922304183] [camera.camera]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1648720445.922340981] [camera.camera]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1648720445.922378352] [camera.camera]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1648720445.922416142] [camera.camera]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1648720445.922462260] [camera.camera]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1648720445.922497856] [camera.camera]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1648720450.454908192] [camera.camera]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1648720450.457736263] [camera.camera]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1648720450.466454023] [camera.camera]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1648720450.468547482] [camera.camera]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1648720450.485548684] [camera.camera]: setupStreams...
[realsense2_camera_node-1]  31/03 09:54:10,704 WARNING [139822209009408] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1648720450.755573629] [camera.camera]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1648720450.757533051] [camera.camera]: Device Serial No: 950122070567
[realsense2_camera_node-1] [INFO] [1648720450.757644011] [camera.camera]: RealSense Node Is Up!

MartyG-RealSense, 2022-03-31:
Near the bottom of the log, just above the **RealSense Node is Up!** message that indicates completion of the launch process, the details of the streams that are being published are provided.  In your launch, it confirms that depth and color are published but infrared is not.

[image]

There is not anything obvious in the log to suggest a problem as the details look fine.

Let's try a launch with a custom stream configuration that will override the stream configuration in the launch file to see whether it makes a difference.  This configuration should stream depth and infra1 at 848x480 at 30 FPS and color at 1280x720 and 30 FPS.

**ros2 launch realsense2_camera rs_launch.py enable_infra1:=true depth_width:=848 depth_height:=480 depth_fps:=30.0 infra1_width:=848 infra1_height:=480 infra1_fps:=30.0 color_width:=1280 color_height:=720 color_fps:=30.0**

alvaropascuaI, 2022-03-31:
I haven't launched anything, everything is the same without the topic and without the infrared stream enabled

MartyG-RealSense, 2022-03-31:
In your opening message you stated that you launched with **ros2 launch realsense2_camera rs_launch.py**

alvaropascuaI, 2022-03-31:
Yes

MartyG-RealSense, 2022-03-31:
Could you tell me please whether infra1 is published if you use the custom launch instruction that I provided above, please?

alvaropascuaI, 2022-03-31:
Whether I launch the ros2 launch realsense2_camera rs_launch.py enable_infra1:=true depth_width:=848 depth_height:=480 depth_fps:=30.0 infra1_width:=848 infra1_height:=480 infra1_fps:=30.0 color_width:=1280 color_height:=720 color_fps:=30.0 as ros2 launch realsense2_camera rs_launch.py I still get the same

MartyG-RealSense, 2022-03-31:
Let's confirm whether Infrared1 is accessible in the RealSense Viewer tool.  If you launch the Viewer and expand open the Stereo Module options, are **Infrared** and **Infrared 2** listed or is only Depth listed with no Infrared options?

[image]

alvaropascuaI, 2022-03-31:
If I do it from the viewer it works for me but I want to launch it with the ros modules

MartyG-RealSense, 2022-03-31:
I asked because there is a problem that can cause the infrared streams to become inaccessible in RealSense applications  and so I wanted to confirm that you did not have that problem, which can be indicated by missing infrared options in the Viewer.  Thanks very much for the confirmation that Infrared is available to you in the Viewer.

Instead of using roslaunch, what happens if you use this alternative launch method:

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p filters:=colorizer**

alvaropascuaI, 2022-03-31:
[image]

alvaropascuaI, 2022-03-31:
[image]
All except the infrared

MartyG-RealSense, 2022-03-31:
Thanks very much for your patience.  I researched the warning **Given stream configuration is not supported by the device** and found several other past cases where the Infrared streams had been declared unsupported by this message.  RealSense team members on those cases advised to check whether the camera has its **Advanced Mode** enabled.

This can be performed in the RealSense Viewer.  

1.  Left-click on the **More** option near the top of the Viewer's options side-panel to reveal its drop-down menu and look at the**Advanced Mode** menu option. 
 
2.  If it has a tick-mark beside it then this means that Advanced Mode is enabled.  If there is not a tick-mark, left-click on Advanced Mode to place a tick.  Then please test whether the Infrared stream can now be found by the ROS wrapper.

[image]

alvaropascuaI, 2022-04-01:
[image]
Nothing, I already have it activated and just in case I launched the command again but even so I still can't see the infrared topic

MartyG-RealSense, 2022-04-01:
Let's go back to basics to get some information about your computer configuration.

-  Which computer or computing device are you using (Raspberry Pi, Nvidia Jetson, laptop / desktop PC, etc)
-  Which methods did you use to install the librealsense SDK 2.50.0 and the RealSense ROS2 Foxy wrapper 3.2.3

alvaropascuaI, 2022-04-04:
- Intel up Squared, it has 64gb, but i don't know if the atom or pentium version
- Ubuntu 20.04 Server
- for the sdk copy the following commands into a .sh file and run it and for the ros wrapper I followed this tutorial https://github.com/IntelRealSense/realsense-ros

# preparations
export REALSENSE_SOURCE_DIR=$HOME/projects/librealsense/
sudo apt-get install guvcview git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
git clone https://github.com/IntelRealSense/librealsense.git $REALSENSE_SOURCE_DIR
mkdir $REALSENSE_SOURCE_DIR/build
cd $REALSENSE_SOURCE_DIR/build

# build
export REALSENSE_INSTALL_PREFIX=/opt/realsense
sudo mkdir -p $REALSENSE_INSTALL_PREFIX;
sudo chown $USER:$USER -R $REALSENSE_INSTALL_PREFIX # not relay needed -> you could run _make_ followed by _sudo make install_
cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=false -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_INSTALL_PREFIX=$REALSENSE_INSTALL_PREFIX
make install 

# extend the ld search path
sudo sh -c "echo $REALSENSE_INSTALL_PREFIX/lib > /etc/ld.so.conf.d/realsense.conf"
sudo ldconfig
 
# udev rules
cd ~/projects/librealsense/
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger

# make cmake config avaliable
echo "export realsense2_DIR=/opt/realsense/lib/cmake/realsense2" >> ~/.bashrc 

# reboot

MartyG-RealSense, 2022-04-04:
400 Series cameras can work with any Intel or Arm processor and so there should not be a compatibility issue with any of the CPUs offered in the Up Squared board series, though performance may of course differ as the Atom has 1.84 Ghz clock speed compared to 2.4 / 2.5 Ghz on the other CPU options.

Putting installation commands into an .sh installation script file is a well-proven method for installing the librealsense SDK.  So there should be no problem there.  

It looks as though you are using the install script from the link below.

https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/

I would recommend trying to use Intel's official 'libuvc_installation.sh' script in the link below as your .sh script code instead, with the single edit of changing its short CMake build instruction for a longer one based on the one in your current script:

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

Then build the ROS wrapper afterwards once the librealsense installation has been confirmed as successful.

As librealsense is being built with RSUSB set to True, it should not matter that you are using Ubuntu Server instead of normal Ubuntu because an RSUSB build is not dependent on Linux versions or kernel versions and does not require patching.  That raises a good point though.  Can you confirm that you have **not** patched the kernel before running your .sh script, since patching is not necessary with an RSUSB=true build and may indeed cause problems with the RSUSB build of librealsense if done.

alvaropascuaI, 2022-04-04:
Ok, I'm going to try to launch that file with the commands and the change that you told me and about the kernel I don't understand the question very well

alvaropascuaI, 2022-04-04:
It gives me the following error
[image]

MartyG-RealSense, 2022-04-04:
The CMake instruction is building librealsense from source code instead of from Debian packages.  Normally it is required that a patch is applied to the Linux kernel before performing a source code build of librealsense.  When RSUSB backend is set to True in the CMake build instruction though, the build bypasses the kernel and so it is not necessary to kernel-patch.  

Normally when the **libuvc_installation.sh** script is run, a chmod is initiated before the script is run.

```
chmod +x ./libuvc_installation.sh
./libuvc_installation.sh
```

alvaropascuaI, 2022-04-04:
Yes, I have launched the chmod +x but it still gives me the error

MartyG-RealSense, 2022-04-04:
I suspect that the problem may be that you are using Ubuntu 20.04 (Focal), as this script was designed for earlier Ubuntu versions such as 16.04 and 18.04.  I do apologize.  It is attempting to read amazonaws package addresses that are now obsolete and replaced by different addresses, which is why the install process cannot find them.

Okay, let's backtrack and look at your original build process using the script from answers.ros.org.  The librealsense installation with that script seems to work and you can access the infrared streams correctly in the Viewer, but not in the ROS wrapper.

So after librealsense has been successfully installed, what are the next steps that you perform afterwards to install the ROS wrapper?

alvaropascuaI, 2022-04-04:
I followed this other tutorial https://github.com/ros2/rosbag2 and I'm on the foxy - eloquent branch

alvaropascuaI, 2022-04-04:
I am in the sudo rosdep init step and it fails me and asks for the following, then I try to launch it again after launching rosdep update, I continue with the following command?

MartyG-RealSense, 2022-04-04:
Okay, so you have performed **sudo apt-get install python3-rosdep -y** and then inputted **sudo rosdep init** without the # sign on the end, yes?

alvaropascuaI, 2022-04-04:
Sorry my problem I thought it was failing but everything was fine, but the last command tells me the following

[image]

MartyG-RealSense, 2022-04-04:
These messages can apparently occur if the installation process believes that the librealsense SDK has not been installed first before the RealSense ROS wrapper was built secondly after that.

alvaropascuaI, 2022-04-04:
could it be that I delete the realsense-ros folder that I had before, do I launch the command again for the file about installing the realsense sdk?, because the realsense viewer continues to work normally

MartyG-RealSense, 2022-04-04:
As you built the RealSense  from source code, you should be able to delete a wrapper installation by de;leting the entire **/ros2_ws/src** folder and then building the wrapper again.

alvaropascuaI, 2022-04-04:
do you recommend me to launch sudo apt-get install ros-$ROS_DISTRO-realsense2-camera with my version of foxy?

MartyG-RealSense, 2022-04-04:
If you are willing to do a fresh install of librealsense then sudo apt-get is certainly an easy way to install both librealsense and the wrapper at the same time and get a perfect version match between SDK and wrapper.

I would recommend removing the existing librealsense installation first though, so that it does not conflict with the librealsense installed from packages by sudo apt-get

alvaropascuaI, 2022-04-04:
I have deleted the librealsense2 folder that I had, I have launched the following command **sudo apt-get install ros-foxy-realsense2-camera** has not given me an error, I have re-launched this command **ros2 launch realsense2_camera rs_launch.py ​​enable_infra1:=true depth_width: =848 depth_height:=480 depth_fps:=30.0 infra1_width:=848 infra1_height:=480 infra1_fps:=30.0 color_width:=1280 color_height:=720 color_fps:=30.0** that you told me but everything is still the same, I don't know if I have to delete something else apart from the librealsense

MartyG-RealSense, 2022-04-04:
What happens when you launch this way:

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_infra1:=true**

alvaropascuaI, 2022-04-04:
oh I think it works for me, really thank you very much, I'm sorry for the inconvenience, I don't know what happened from everything we've done

[image]

alvaropascuaI, 2022-04-04:
one last thing, I am trying to see the cameras with the following command ros2 run image_view image_view image:=/camera/color/image_raw, and nothing comes out but if I echo the topic I do get data

MartyG-RealSense, 2022-04-04:
It's no trouble at all.  Switching from the /rosbag2 wrapper that you were originally using to the official Intel RealSense ROs2 wrapper helped, of course.  

In regard to the **ros2 run** instruction working, the current RealSense ROS2 wrapper is a new system with some differences to the previous ROS2 wrapper.  So we have both learned new things from this case  :)

Regarding changing the color of the depth camera, do you mean changing how the depth data is colorized?  If so, you should be able to enable the **colorizer** when launching.  I believe that when using the 'ros2 run' launch type, it would be something like this:

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_infra1:=true -p colorizer.enable:=true**

In regard to image_view, does this work:

**ros2 run image_view image_view --ros-args --remap /image:=/camera/color/image_raw**

alvaropascuaI, 2022-04-04:
If I run the ros2 command run realsense2_camera realsense2_camera_node --ros-args -p enable_infra1:=true -p colorizer.enable:=true I get [INFO] [1649085058.096946863] [camera]: Setting Dynamic reconfig parameters. and it does not advance me, what I mean by color is that by default it looks like the first one and I want to see it the second way

[image]

about the image was that the name of the topic was misspelled

MartyG-RealSense, 2022-04-04:
The colorizer is what converts a monochrome depth image into color by color-shading the depth points according to how near or far they are from the camera.  The shading is determined by the colorizer **color scheme** that is enabled.  In the RealSense Viewer, it uses the **Jet** color scheme by default to shade the depth from blue (near distance) to red (far distance), like in your color bar above.

[image]

I do not currently have a reference though for how the colorizer is set in the new 4.x ROS2 wrapper.  I will seek advice from the ROS wrapper developers.  Thanks very much for your patience!

alvaropascuaI, 2022-04-04:
Ok, thank you very much and for the last one (I think), I don't know if you know how to make the pieces when I save a topic from a python node, it would be like by command line with -d but from a node

MartyG-RealSense, 2022-04-04:
Could you provide more information about what 'make the pieces' means, please?  Do you mean launching a Python node script from the ROS wrapper?

MartyG-RealSense, 2022-04-04:
I'm not familiar with that particular approach of splitting recordings.  There is **ros2 bag record** though that offers a bag splitting function with -d (similar to the -split command in rosbag record in ROS1).

https://github.com/ros2/rosbag2#splitting-recorded-bag-files

Please try this command which I have adapted from the above documentation link:

**ros2 bag record /aaa -d 10**

If you wanted to record all topics, try this:

ros2 bag record -a -b 10**

alvaropascuaI, 2022-04-05:
By commands, yes, I know how to do it, in the documentation that you gave me, what I told you is not there, it is something basic but well, thank you, have you discovered the colors of the realsense camera?

MartyG-RealSense, 2022-04-05:
I am awaiting a response to my query to Intel about the colorizer and will pass it on to you as soon as I have it.  Thanks very much for your patience.

alvaropascuaI, 2022-04-05:
Aaa okay okay, don't worry, thanks to you too, I'll be waiting, and if not I'll look

alvaropascuaI, 2022-04-07:
Good, I'm having problems with the accelerometer and the gyroscope because the topic does come out but I try to see what data passes through the topic and I don't get any data

MartyG-RealSense, 2022-04-07:
Intel are still investigating the colorizer issue.  I will update you about that as soon as I have more information that I can pass on.

IMU topics are usually disabled by default in the RealSense ROS wrapper.  Do you receive the data if you enable them in your launch by setting enable_accel and enable_gyro to True?

**ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_infra1:=true -p enable_accel:=true -p enable_gyro:=true**

alvaropascuaI, 2022-04-07:
Sometimes, it happens to me that I launch the camera node then I can't stop it, look what I get when I do the ctrl c

[image]

MartyG-RealSense, 2022-04-07:
The **[rclcpp]: signal_handler (signal_value=2)** error appears in ROS2 in general and not just in the RealSense ROS2 wrapper.  An example where it is occurring when CTRL-C is used is discussed in the link below.

https://answers.ros.org/question/317630/ros2crystal-when-using-rclcppexecutorsmultithreadedexecutorspin-if-i-use-ctrlc-it-will-not-shutdown-the-script/

The commenters in that case thought that it might be being caused by a **segmentation fault** generated after CTRL-C is used.

In a RealSense ROS case in https://github.com/IntelRealSense/realsense-ros/issues/2107 that experienced the error, the RealSense ROS user thought that it might be related to RViz2.  Have you launched RViz2 before you use CTRL-C please?

MartyG-RealSense, 2022-04-07:
That's great to hear  :)

alvaropascuaI, 2022-04-11:
Hello, lately I'm having problems with some camera, either the gyroscope or the accelerometer or both or the depth camera fails and when I solve one the other one fails and when I launch a command it stays in Setting Dynamic Reconfig parameters and I can't exit with the ctrl + C, so I have to reboot the intel

MartyG-RealSense, 2022-04-11:
In your launch instruction, you have not included the commands to enable the gyro and accel topics.

**-p enable_accel:=true -p enable_gyro:=true**

alvaropascuaI, 2022-04-11:
If I do **ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true -p enable_infra1:=true -p depth_width:=640 -p depth_height:=480 -p depth_fps:=6.0 -p infra1_width:=640 -p infra1_height :=480 -p infra1_fps:=6.0 -p color_width:=640 -p color_height:=480 -p color_fps:=6.0 -p enable_accel:=true -p enable_gyro:=true -p gyro_fps:=400.0 -p accel_fps:= 250.0** everything works for me at least for now, would you know how to launch this module with so many things with a ros launch?

MartyG-RealSense, 2022-04-11:
The 4.x wrappers introduced a shorthand method of configuring depth, color and infrared with the instruction **depth_module.profile** when the resolution / FPS for the depth, color and infra stream types is the same.  For example:

` depth_module.profile:=640x480x30`

ros2 launch realsense2_camera rs_launch.py initial_reset:=true depth_module.profile:=640x480x6 enable_infra1:=true enable_accel:=true enable_gyro:=true gyro_fps:=400.0 accel_fps:= 250.0 

Bearing in mind that you had problems before with using the rs_camera launch file instead of ros2 run, does the above command work for you?

alvaropascuaI, 2022-04-11:
It gives me problems because the infrared camera is not activated because, for example, I launched this and **ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_infra1:=true depth_width:=640 depth_height:=480 depth_fps:=6.0 infra1_width:=6.0 was not activated =640 infra1_height:=480 infra1_fps:=6.0 color_width:=640 color_height:=480 color_fps:=6.0 enable_accel:=true enable_gyro:=true gyro_fps:=400.0 accel_fps:=250.0**

MartyG-RealSense, 2022-04-11:
I thought that using rs_launch.py might still give you problems but it was worth trying.

I note that in the wrapper documentation's example of using depth_module.profile they are using infra1 but not including infra1_enable.  Do you still not receive the infra1 topic if you remove infra1_enable from the instruction, please?

**ros2 launch realsense2_camera rs_launch.py initial_reset:=true depth_module.profile:=640x480x6 enable_accel:=true enable_gyro:=true gyro_fps:=400.0 accel_fps:= 250.0**

Does the launch still fail if you remove the IMU topics?

**ros2 launch realsense2_camera rs_launch.py initial_reset:=true depth_module.profile:=640x480x6**

alvaropascuaI, 2022-04-11:
It doesn't matter which of the two I launch because the stream of the infrared camera is not enabled and in the first one the gyroscope and the accelerometer are enabled but in the second only the depth camera and the color camera

MartyG-RealSense, 2022-04-11:
If it is not possible for you to use a launch file such as **rs_launch.py** then using a lengthy launch command with ros2 run may be the only option available to you.

A previous launch-file approach that has been used with ROS2 (the older 3.2.2. wrapper, not the current 4.0.4 one) is to put the paramters in a **yaml** configuration file instead of listing them in the rs_launch.py launch instruction.  https://github.com/IntelRealSense/realsense-ros/issues/2043 provides information about this.  This approach may not work though, given your previous problems with using a launch file instead of ros2 run.

[66 more replies]

closed 2023-02-23 by alvaropascuaI