realsenseai/realsense-ros #2307: Question regarding high quality depth image and infra without dots

Kaijun101, 2022-04-01:
Hi @MartyG-RealSense, can I check with you if its possible to enable to emitter to on and off so that it can generate high quality depth image while not having infra emitter on for the infra stream?

MartyG-RealSense, 2022-04-01:
Hi @Kaijun101  Emitter on-off will continuously alternate the emitter on and off.  If you wish to completely turn off the emitter then you can either set the Laser Power option to '0' with **RS2_OPTION_LASER_POWER** or disable the emitter with **RS2_OPTION_EMITTER_ENABLED**

The link below provides example C++ code under the **librealsense2** heading for controlling the laser power and emitter on / off status.  If you require Python code for doing so, please let me know.

https://dev.intelrealsense.com/docs/api-how-to#section-controlling-the-laser

Kaijun101, 2022-04-01:
Yes, I would like to know how to control it in ROS. Like how can I off the emitter when publishing infra image and enable it when publishing depth image. So that, I can achieve both high quality depth image and infra without the emitter dots.

MartyG-RealSense, 2022-04-01:
If your scene is well lit then a 400 Series camera can alternatively use the ambient light in a scene to analyze surfaces for depth detail instead of having the IR dot pattern projection enabled.

If you are using ROS1 (Kinetic, Melodic, Noetic), a RealSense ROS user shares their launch file instructions in https://github.com/IntelRealSense/realsense-ros/issues/1379#issuecomment-934502369 for disabling the emitter with a rosparam command.

```
<rosparam> 
/camera/stereo_module/emitter_enabled: 2
</rosparam>
```

Again for ROS1, https://github.com/IntelRealSense/realsense-ros/issues/1379 provides an example of controlling the laser power and emitter on / off status during runtime after launch with dynamic reconfigure.

Kaijun101, 2022-04-06:
Hi @MartyG-RealSense, I notice that on my realsense-ros i state it was bulit with librealsense v2.50.0, even though I checkout realsense-ros to 2.3.0. But on my com, I built only built only v2.4.50 and when I run realsense-viewer, it shows as v2.45.0. Can u assist me on this?

MartyG-RealSense, 2022-04-06:
If the log states that it was built with 2.50.0 but running with 2.45.0, that would indicate that originally librealsense 2.50.0 was installed at the time that the ROS wrapper was built but then the librealsense version was changed to 2.45.0 at some time **after** the ROS wrapper was built.

Whenever the librealsense version is changed, the ROS wrapper must be built again or the different built with / running with version numbers occur.

Kaijun101, 2022-04-07:
I have uninstall both and reinstall but it still state as built with v2.50.0. Can u advise me on how to resolve this or is there any recommended steps to do so?

MartyG-RealSense, 2022-04-07:
Do you mean that the log still says 'built with 2.50.0, running with 2.45.0' ?

Kaijun101, 2022-04-07:
Yes, it says built with 2.50.0

MartyG-RealSense, 2022-04-07:
If you built the ROS wrapper with **git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`** then the latest wrapper version will be installed.  That would currently be 2.50.0.

Please try this command to checkout version 2.45.0 specifically if you are not doing so already.  I see that you used this command a couple of months ago at https://githubhot.com/repo/engcang/vins-application/issues/9

**git checkout v2.45.0**

Kaijun101, 2022-04-07:
Thanks for your advice. I managed to resolve this issue. But now when I run the ros-wrapper it shows my d435i as connected to usb 2.1 even though I am using a usb3 port.

MartyG-RealSense, 2022-04-07:
It's excellent to hear that you made progress.

Does the USB still show as 2.1 in the log if you add a camera reset instruction to your roslaunch to reset the camera at launch?

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

Kaijun101, 2022-04-07:
Yes the issue presists. I tried the same wire with another d435i i got usb3.2.

MartyG-RealSense, 2022-04-07:
Does the Motion Module Failure error occur on the camera detected as USB 2.1 or on the other camera that you tried?

Which camera firmware driver version are you using with the cameras, please?

Kaijun101, 2022-04-07:
The error occurs on the camera detected as usb 2.1. I tried both 5.13.0.50 and 5.12.14.50 but both also detect as usb 2.1

MartyG-RealSense, 2022-04-07:
So the USB cable is okay as your other D435i camera was detected as USB 3.2 when using it with that camera.

Have you tried unplugging the micro end of the USB cable from the side of the camera, turning it around the other way and re-inserting it (as USB-C cables are two-way insertion on the micro sized end).  Whilst doing this reversal usually only makes a difference on the L515 camera model, it is worth trying in order to eliminate it as a possible solution.

Kaijun101, 2022-04-08:
Yes, I have tried inserting it both ways and it still detect as usb 2.1. Not sure what's wrong with the D435i, but the cable is ok as the other camera is detected as usb 3.2. I changed another cable and its also usb 2.1.

MartyG-RealSense, 2022-04-08:
Does the affected camera have 2.1 beside its name in the options side-panel of the Viewer, or is it only the ROS wrapper that the USB mis-detection is occurring in?

[image]

Kaijun101, 2022-04-08:
Yes, realsense-viewer also shows usb2.1.

MartyG-RealSense, 2022-04-08:
Is the camera still detected as USB 2.1 if you reset the camera to its factory-new configuration in the Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

Kaijun101, 2022-04-09:
Yes, the camera is still detected as USB 2.1.

MartyG-RealSense, 2022-04-09:
Having looked at all the information in this case, there seems to be a high probability that the problem is with the particular computer rather than the camera hardware.  Sometimes a particular camera may stop working correctly on a certain PC whilst other cameras work normally, whilst the affected camera works correctly when tried on a different computer.  

I have a D435i that exhibits this behaviour (works fine on one Windows laptop but does not detect correctly on another Windows laptop, whilst other cameras work normally on the affected laptop).  As this is apparently a Windows related issue, a solution has not yet been found.

Kaijun101, 2022-04-10:
I am currently using ubuntu 18.04.

MartyG-RealSense, 2022-04-10:
Thanks for the confirmation of your OS.  Is it still detected as 2.1 if you reload the Ubuntu **udev rules**

`sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger`

Kaijun101, 2022-04-11:
Yes, it still detected as usb2.1. The fiollowing errors appears as well I noticed.

 11/04 23:25:35,846 ERROR [547406999328] (error-handling.cpp:78) Error during polling error handler: usb device disconnected
 11/04 23:25:36,847 ERROR [547113385760] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 11/04 23:25:36,847 ERROR [547406999328] (sensor.cpp:526) acquire_power failed: failed to set power state
 11/04 23:25:36,847 ERROR [547406999328] (error-handling.cpp:78) Error during polling error handler: failed to set power state
 11/04 23:25:37,848 ERROR [547113385760] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 11/04 23:25:37,848 ERROR [547406999328] (sensor.cpp:526) acquire_power failed: failed to set power state
 11/04 23:25:37,849 ERROR [547406999328] (error-handling.cpp:78) Error during polling error handler: failed to set power state
 11/04 23:25:38,849 ERROR [547113385760] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 11/04 23:25:38,849 ERROR [547406999328] (sensor.cpp:526) acquire_power failed: failed to set power state
 11/04 23:25:38,849 ERROR [547406999328] (error-handling.cpp:78) Error during polling error handler: failed to set power state

MartyG-RealSense, 2022-04-11:
Are you using two D435i at the same time or just one at a time, please?

MartyG-RealSense, 2022-04-12:
Which firmware version is the other D435i that works normally using?

closed 2022-04-24 by MartyG-RealSense