realsenseai/realsense-ros #2309: rosdep install error: `No definition of [xacro] for OS version [bionic]`

GuillaumeLaine, 2022-04-01:
I have successfully installed the Realsense SDK following the Jetson Installation guide (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) on a Jetson AGX + realsense-viewer is working successfully. 

However, when trying to install the ROS2 wrapper from sources, step 4 with:
`rosdep install -i --from-path src --rosdistro $_ros_dist --skip-keys=librealsense2 -y`
gives same errors as mentioned in #2173  which are also fixed by replacing `$_ros_dist` by `$ROS_DISTRO` (curious that the issue has been closed but the readme wasn't updated? At least on the ros2 branch).
Then running:
`rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y`
gives me the error:
`No definition of [xacro] for OS version [bionic]`
I've tried to skip that key:
`rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys "librealsense2 xacro" -y`
 and everything seemed to work afterwards. However, I'm not sure if this is normal and if this will cause issues in the future?

I'm trying to install the ROS2 wrapper within the container provided by isaac_ros_common (https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common)

MartyG-RealSense, 2022-04-01:
Hi @GuillaumeLaine  My understanding is that xacro would be important if you were going to be performing robot **simulation** with Gazebo.

GuillaumeLaine, 2022-04-06:
Hi @MartyG-RealSense thanks for your reply, good to know! 

I'll let you decide whether this issue should now be closed. I'm not planning on using Gazebo at the moment and this answers my question perfectly, however I imagine the missing xacro key might still be a problem to others; the rosdep `-r` flag seems to work by just ignoring installation errors and might not be a sufficient solution to everyone.

At any rate thanks again, cheers!

MartyG-RealSense, 2022-04-06:
You are very welcome   :)

As this is a rarely occurring issue, I will close the case.  Thanks again!

closed 2022-04-06 by MartyG-RealSense