realsenseai/realsense-ros #2311: How to convert 2D coordinates to 3D coordinates ?

BnHassan1998, 2022-04-05:
I am using yolov3 with darknet to train and test my project.   how I can  convert 2D coordinates to 3D coordinates  to know the Z
 (depth )  . because I have to do localization for the objects

MartyG-RealSense, 2022-04-05:
Hi @BnHassan1998  As you are using datknet, the **darknet_ros_3d** package that can calculate a 3D bounding box with a 3D position for each detected object may be an appropriate solution.

https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d

https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/issues/15 discusses using a Real sense D435 camera with darknet_ros_3d, with the comment https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/issues/15#issuecomment-620276472 in that discussion being particularly useful.

BnHassan1998, 2022-04-05:
I have changed to Yolo v4. but is it ok if I use the normal images for my dataset, I am using only a software project so I don't have images taken by realsense camera so is it possible to do the conversion? to get the Z (depth) to do the localization

BnHassan1998, 2022-04-05:
> Hi @BnHassan1998 As you are using darknet, the **darknet_ros_3d** package that can calculate a 3D bounding box with a 3D position for each detected object may be an appropriate solution.
> 
> https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d
> 
> [IntelligentRoboticsLabs/gb_visual_detection_3d#15](https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/issues/15) discusses using a RealSense D435 camera with darknet_ros_3d, with the comment [IntelligentRoboticsLabs/gb_visual_detection_3d#15 (comment)](https://github.com/IntelligentRoboticsLabs/gb_visual_detection_3d/issues/15#issuecomment-620276472) in that discussion being particularly useful.

conversion

MartyG-RealSense, 2022-04-05:
It sounds as though the task that you are trying to achieve with yolov4 and darknet (and ordinary RGB image files?) is similar to the one that somebody was asking in the link below about how to do (but had not yet achieved).

https://stackoverflow.com/questions/67640416/how-to-calculate-length-breadth-2d-dimensions-of-an-objet-using-darknet-and-r

Does the above case accirately describe your own project goals, please?

closed 2022-04-20 by MartyG-RealSense