realsenseai/realsense-ros #2313: [ROS2] L515 has no default profile

dmark27, 2022-04-07:
I am in the process of upgrading from ROS 1 to ROS 2 and have run into an issue that throws an error when trying to launch rs_camera when connected to my L515. I've checked in realsense-viewer and the hardware still works, and I also have a d435 that launches without issue and I can see the point cloud in rviz2.

Ubuntu: 20.04
ROS distro: Galactic

`$ ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true inital_reset:=true
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [49110]
[realsense2_camera_node-1] [INFO] [1649352820.974361687] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1649352820.974421892] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1649352820.974437813] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1]  07/04 13:33:41,191 WARNING [140613857040128] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[realsense2_camera_node-1]  07/04 13:33:41,328 WARNING [140613857040128] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[realsense2_camera_node-1]  07/04 13:33:41,381 WARNING [140613857040128] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[realsense2_camera_node-1]  07/04 13:33:41,435 WARNING [140613857040128] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[realsense2_camera_node-1] [INFO] [1649352821.521864431] [camera.camera]: Device with serial number f0171990 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1649352821.521949478] [camera.camera]: Device with physical ID 3-1-7 was found.
[realsense2_camera_node-1] [INFO] [1649352821.521968150] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1649352821.522237385] [camera.camera]: Device with port number 3-1 was found.
[realsense2_camera_node-1] [INFO] [1649352821.522264778] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1649352821.522281242] [camera.camera]: Device f0171990 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1649352821.523846587] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1649352821.524123884] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1649352821.524149800] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1649352821.524163625] [camera.camera]: Device Serial No: f0171990
[realsense2_camera_node-1] [INFO] [1649352821.524175870] [camera.camera]: Device physical port: 3-1-7
[realsense2_camera_node-1] [INFO] [1649352821.524187204] [camera.camera]: Device FW version: 01.05.08.01
[realsense2_camera_node-1] [INFO] [1649352821.524198890] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1649352821.524210063] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1649352822.722891956] [camera.camera]: Set ROS param depth_module.profile to default: 320x240x30
[realsense2_camera_node-1] [ERROR] [1649352822.866776369] [camera.camera]: Error starting device: No default profile found
`

MartyG-RealSense, 2022-04-07:
Hi @dmark27  I note in your log that the depth default is being set as 320x240 and the USB mode is detected as 2.1 instead of 3.2.  This suggests that unplugging the micro-sized end of the USB cable from the base of your L515, turning the connector around the opposite way and re-inserting it into the base of the camera may correct the problem.  

This is because the L515 cameras model has a known hardware issue where this incorrect depth and USB profile may occur when the micro-sized end of the cable is plugged into the camera in one of the two possible orientations.

dmark27, 2022-04-07:
This fixed the issue thanks!

closed 2022-04-07 by dmark27

MartyG-RealSense, 2022-04-07:
Great to hear, @dmark27 - thanks very much for the update!