realsenseai/realsense-ros #2314: foxy D435 /camera/depth/image_rect_raw data is 0

miku54, 2022-04-08:
When I turn on D435, I can clearly see the /camera/depth/image_rect_raw depth image, but when I use ros2 topic echo /camera/depth/image_rect_raw to view the topic data is all 0, why is this?   My system is Ubuntu20.04 ros2 foxy version
[image]
[image]

miku54, 2022-04-08:
[image]

MartyG-RealSense, 2022-04-08:
Hi @miku54  The multiple **control_transfer returned** warnings suggest that there is difficulty in the communications with the device.  The **resource temporarily unavailable** type of this message suggests that the computer may be having difficulty with processing incoming new camera frames.  This could be because the computer's CPU is over-burdened by processing or because of an issue with the camera's USB connection.

For example, your USB connection's speed is being detected in the log as USB 2.1 instead of the much faster USB 3.2.  Is your camera plugged into a USB 3 port, please?

Could you also provide the ROS2 launch instruction that you are using, please?  The new ROS2 4.x series of wrappers uses some different parameters compared to the previous ROS2 wrapper (up til 3.2.3) and some frame IDs such as **_optical_frame_id** have been removed.  Details are at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#please-notice-if-you-are-moving-from-realsense-ros2-branch-to-ros2-beta

miku54, 2022-04-12:
Sorry for not being able to reply to you in time due to the recent move.
My camera is plugged into the Jetson nano's USB port, and its USB is all 3.0 ports, so I have no reason to connect to a usb2 port.
My ROS2 startup command is:
roslaunch realsense2_camera rs_camera.launch
Because I have set the device_type to D435 in rs_camera.launch, I did not add parameters to the command.
I am using exactly the ROS2-beta branch.
BTW, when opening realsense-viewer everything is fine.

MartyG-RealSense, 2022-04-12:
**roslaunch realsense2_camera rs_camera.launch** is a ROS1 wrapper launch file.  The equivalent launch command on ROS2 is: 

**ros2 launch realsense2_camera rs_launch.py**

miku54, 2022-04-12:
Sorry, this is the start command, but I made a typo
 
ros2 launch realsense2_camera rs_launch.py

MartyG-RealSense, 2022-04-12:
Thank you for the clarification.  Do you still experience problems if you launch with 'ros2 run' instead of rs_launch.py

**ros2 run realsense2_camera realsense2_camera_node**

MartyG-RealSense, 2022-04-18:
Hi @miku54  Do you require further assistance with this case, please?  Thanks!

miku54, 2022-04-19:
Thank you, I have solved it. Just don't need to set device_type to D435 in rs_camera.launch, run ros2 launch realsense2_camera rs_launch.py normally to open the camera, and the /camera/depth/image_rect_raw topic also has data. Thank you very much for your answer ! @MartyG-RealSense

closed 2022-04-19 by miku54

MartyG-RealSense, 2022-04-19:
That's great to hear, @miku54 - thanks very much for the update  :)